Hamlib/Hamlib #1401: Meaning of targeted VFO with \set_split_* commands

mikaelnousiainen, 2023-10-17:
@mdblack98 Again, maybe you know this best.

I'm trying to understand the *intended* function of the targeted VFO (in VFO mode) for the \set_splitfrewq

mikaelnousiainen, 2023-10-17:
Accidentally uploaded incomplete issue, finishing the text...

mikaelnousiainen, 2023-10-17:
@mdblack98 Ok, now finished the issue text with 2 questions :)

mdblack98, 2023-10-17:
The target VFO _should_ be irrelevant for set_split functions.
The problem is mainly with Icom as we don't know the active VFO of any Icom rig.  The current VFO is always RX and the "other" VFO is transmit.   So we have some choices....but we need to be able to support  reverse  split on everybody.
For Icom, we either have to assume current VFO is RX or we have to force what we think the RX VFO is before we set split things.  The 1st option let's the user choose (but would they deliberately change?) and the 2nd would force them to change it via software first if they want to.

This includes  Main/Sub rigs who can run split either way too but still can't figure out what VFO is active.  Icom rigs in sat mode can only transmit on Sub which is what you see in the Icom code if VFO_TX is the targeted mode or  VFO_SUB is the transmit based on the original split setup.
So before you do anything let's be clear about what you think ought to happen.

mdblack98, 2023-10-17:
Also current_vfo=RX is only true if not transmitting as I believe all Icom rigs show the split VFO active during transmit.
So, if we try to change frequency for Doppler control need allow that too.

Mike W9MDB

mikaelnousiainen, 2023-10-17:
So should it be instead so that `\set_split_freq` always alters the split (TX) VFO, which can be set by `\set_split_vfo`  (and completely ignore the target VFO)?

The fact is just that based on this code:

https://github.com/Hamlib/Hamlib/blob/680ce55ba26df1e3634356229bfe20491706f77e/src/rig.c#L4314

it is not so, but the target VFO **is** considered for the command.

mikaelnousiainen, 2023-10-17:
Just saying that there are two inconsistencies to be solved:

1) Whether the target VFO affects the result of `\set_split_*` and `\get_split_*` commands (all split commands should work the same way regarding target VFO, I think)

2) How to make (at least) Yeasu/Kenwood/Icom backends behave mostly the same way and avoid unnecessary VFO swapping (also on recent Icom rigs with 0x25+0x26 command support)

mikaelnousiainen, 2023-10-17:
Adding one more inconsistency:

3. Get/Set split commands should have the same logic to determine the VFO to get/set. Now (at least with Icom), the logic is different -- and does not correctly address all Main/Sub rigs.

mdblack98, 2023-10-17:
Any other thoughts?

I've been slowly migrating to the idea that clients should use VFO_RX and VFO_TX and then  either set_vfo or set_split_vfo for simplex or split operations.

That way clients don't need to know anything about the rig's modes.

We need to maintain backwards compatibility though.

The only exception to ignoring the VFO right now that I  know of is for VFO_TX to be specified for Icom.  That is/was a future-looking change but look like it may not be necessary.

#1 Simplex operations are no problem -- currVFO is all that is needed  -- if user changes VFOs then the other VFO becomes the currVFO.  Nothing we can do about that for Icom.
#2 Split is more complicated

Let's try to outline the desired behavior

set_split_vfo
    #1 Should use vfo_tx argument to determine TX vfo, vfo is ignored.  Should return an error if TX vfo cannot be done...not there are cases where  Main/Sub is restriced on which can transmit.
    #2 if RIG_VFO_RX/RIG_VFO_TX are passed in as vfo and vfo_tx it should defaults to VFOA/VFOB RX/TX respectively or Main/Sub RX/TX.
    #3 gpredict will ask for Main/Sub or Sub/Main or A/B or B/A.
    #4 turns split on
set_split_freq/set_split_mode/set_split_freq_mode
    #1 If vfo_rx == vfo_tx should also mean split==false and should either be an error or should default to setting up split automagically.  I favor the default split.
    #2 Should ignore vfo and should use set_split_vfo setup which defaults to A/B RX/TX or Main/Sub RX/TX

I can imagine a client just calling set_split_freq_mode and only having to call set_split_vfo when they want to turn split off.

Mike W9MDB

mikaelnousiainen, 2023-10-18:
Thanks for the comments. Now we are getting close to a good specification.

I also like the idea that the split (mode/freq) commands should use default split VFO settings if RX VFO == TX VFO, so they would never change the freq of the RX VFO. But these commands would still only change the freq/mode of the determined TX VFO (and set the rig state TX VFO accordingly maybe?) and NOT turn split on/off ever.

Instead, I think that only an explicit \set_split_vfo command should turn split on.

How does that sound?

Some additional questions if you happen to know:

- I've noticed that -- in addition to the rig state current_vfo and tx_vfo -- Icom code sometimes uses its own private state with current/tx VFO, which seems more or less like a duplicate of the rig state, but not consistently used. Should this be changed to use rig state, I guess?

- Do you know if popular Hamlib client apps like WSJT-X, JTDX, gpredict, fldigi and such actually ever use the VFO mode? How "popular" is the use of targeted VFO commands for split if we change them (to ignore the VFO) ?

mikaelnousiainen, 2023-10-18:
With the improvements I'm planning, I'd also try to make sure regular `\set_freq`, `\get_freq`, `\set_mode` and `\get_mode` work correctly with targeted VFOs -- especially on Icoms. I've experienced plenty of VFO swapping even on some newer rigs, so I thought it's time to fix it finally :D

mdblack98, 2023-10-18:
The VFO swapping will occur if one does not use the --vfo option.

VFO swapping should not occur on TARGETABLE_VFO rigs.

You have an example where it occurs?

mdblack98, 2023-10-18:
> The only time non-VFO commands are ever used is with rigctld when --vfo is not provided -- that's all for backwards compatiblity.  Most everybody has changed to use --vfo for rigctld including gpredict and Log4OM.   gpredict at least in my fork of it.
>
> Leaving set_split_vfo as-is is fine with me.  I've had a complaint or two from people who think they should be able to set split freq with having split on...the split functions should return an error then if split is off.
>
> Are you proposing split functions should still work if split is not on?

And yes -- the Icom vfo memory could likely be done at the top-level now.  That has been a migration.

mikaelnousiainen, 2023-10-18:
> The only time non-VFO commands are ever used is with rigctld when --vfo is not provided -- that's all for backwards compatiblity.  Most everybody has changed to use --vfo for rigctld including gpredict and Log4OM.   gpredict at least in my fork of it.

Ok, so VFO mode is widely in use. Good.

> And yes -- the Icom vfo memory could likely be done at the top-level now.  That has been a migration.

Great. I can work on that.

> The only time non-VFO commands are ever used is with rigctld when --vfo is not provided -- that's all for backwards compatiblity.  Most everybody has changed to use --vfo for rigctld including [...]

Not really. AFAIK the split functions have never returned an error if split is not on. This is why I'm confused. I'm all for making them easier to understand. Most of the confusion here is because you can call set_split_freq any time and it has (in my opinion) slightly unexpected results (like changing the RX frequency, or not doing anything at all).

I'm not sure how big of a backwards-compatibility issue it would be to change the split functions to return an error if split is not on. After all, frequency/mode of any VFO can be changed with set_freq/set_mode at any time.

-> So: should we possibly change the split functions (set_split_freq/get_split_freq/set_split_mode/get_split_mode/set_split_freq_mode/get_split_freq_mode) to return an error if split is not on (turned on via set_split_vfo)? Though I guess rig backends should auto-detect if split is already turned on when rigctl starts...

mdblack98, 2023-10-18:
I just tested WSJT-X and if you start WSJT-X with rig split=None but the rig does have split on then WSJT-X does not turn it off.
Seems it does nothing with the split setting if None is selected.   
From a Hamlib perspective we have no idea what the client wants.
So we end up with a choice.
#1 Rely on client to set what it wants -- WSJT-X would need to change and perhaps other clients.#2 Turn split off at startup and let client turn it back on #3 Wait for first freq/mode setting or split setting -- if freq/mode without split being set then turn it off.#4  Do #2 and then this gets into why setting split on when split functions are requested kind of makes sense to me.More of the path of least resistance.  If you are requesting split functions you must want split on.

mikaelnousiainen, 2023-10-18:
:+1: Yes, we come back to that set_split_freq/mode functions should probably set split on to preserve compatibility with as many clients as possible. This is also ok for me.

What I'm after here is that set_split_freq/mode commands should either return an error if split is not turned on or then *actively* turn split on -- the latter seems more compatible.

However, this still leaves the get_split_freq/mode commands a bit ambiguous, as I see that they should not alter the rig state. What should the result be if split is not turned on?

What I **would** like to change is the seemingly undefined behavior of set/get split freq/mode when split is not turned on.

mdblack98, 2023-10-18:
Agreed -- seems to me that's what any user or client would intend -- if you ask for a split function for freq or mode we force split on.  That works for me and I think all clients and users.
I'm going to ask WSJT-X to set split to off if None is selected.

mikaelnousiainen, 2023-10-18:
:+1: Repeating myself a bit, but how do you think the get_split_freq/mode functions should work in this case?

IMO they should not alter rig state (none of the get-functions do), but returning split freq/mode if split is not turned on is also incorrect.

mdblack98, 2023-10-18:
WSJT-X has to know the split state as is adjusts audio for it.And if the user has None or Fake  It then split needs to be off and then turned on if Rig is chosen.
This is unique to WSJT-X and why they need to do it.
gpredict also selects it's own split state at startup based on user settings.

mikaelnousiainen, 2023-10-18:
Sorry, I'm not sure what you mean.

What I'm thinking here that there are three possibilities for get_split_freq/mode when split is NOT enabled -- now talking **only** about the GET-functions:

1. Return the RX VFO frequency/mode or sometimes the "other" (or VFOB) frequency/mode (like they seem to do now, depending on rig model?) -- this is the undefined behavior I'm trying to alter here
2. Return an error (would be incompatible)
3. Something else, but somehow indicate that the values are not valid, because split is OFF? Any suggestions?

When split is ON, they should work the way we discussed above -- I guess that's what we've agreed on...

mdblack98, 2023-10-18:
I think split  get functions should turn on split too.
Otherwise they make no sense when split is off.

mikaelnousiainen, 2023-10-18:
I understand what you're after, but wouldn't it be a bit counter-intuitive to have get commands change the state of the rig? Currently (AFAIK), get commands only read the state of the rig and never alter it. I feel strongly there should always be the possibility in Hamlib to check/query for particular data without having it alter the device state...

Also, maybe some clients already poll get_split_freq?

I think the differentiation between get/set commands deserves a bit more thinking still.

Could the get values return some values that indicate they're not valid right now? E.g. frequency 0 and some "NONE" or "N/A" value for mode?

That kind of approach could still make the change mostly backwards-compatible.

mdblack98, 2023-10-18:
I don't know anybody polling split data that does not expect the rig to be in split mode.
So if you are running gpredict for example and you smoosh the split button it SHOULD turn split back on as that  is what gpredict expects.
WSJT-X calls get_split_vfo and get_split_mode and handles the freq with non-split functions.
In both those cases they expect the rig to be in split so split should be forced on.
If you want to start pressing buttons on the rig then shut down the app that's controlling the rig.

mikaelnousiainen, 2023-10-18:
Ok. I'm still hesitant to make such changes to get commands. That's more of a **conceptual** change to have get commands alter the rig state. I think there should be a well-defined set of commands that are "safe" to call without them making changes to the rig state. Currently, all the get commands follow this approach and I'd personally prefer to keep it that way.

That said, I've originally slightly misunderstood the use of get_split_freq/mode (vs. VFO mode and targeted get_freq/mode), since my remote control app Web Radio Control does use get_split_freq for getting the split frequency even if split is not on -- and certainly NOT expecting to turn split ON.

I think it's not the responsibility of Hamlib to try to "fix" the rig state if a person pushes the rig buttons (e.g. turns split off manually) or another app simultaneously controls the rig and changes it to an unwanted state. That's more a user error and Hamlib cannot anticipate all of these situations...

I'd personally go for some "split not enabled" return values for get_split_freq and get_split_mode

mdblack98, 2023-10-18:
I suppose return 0 frequency and RIG_MODE_NONE would be informative -- but I can already hear the questions..."Why am I seeing 0Hz?"On the other hand I doubt we would ever hear "why is hamlib putting my rig in split mode?"

mikaelnousiainen, 2023-10-19:
Heh. I see it more or less the other way around. And there are many rig models (such as the IC-756 and its variants) that allow putting the rig into split mode, but NOT reading the split status... making it even more difficult in remote use to figure out what's happening if get commands enable split.

Still thinking forcing split ON would be more problematic.

What if the get commands would output split state too (as the last value, to make it as compatible as possible):

e.g.

```
Rig command: i
TX Frequency: 0
Split: 0

Rig command: k
TX Frequency: 0
TX Mode: NONE
TX Passband: 0
Split: 0
```

Though maybe that will be another issue with backwards compatibility.

BTW it is true that some backends do already give out "dummy" values for split mode/filter if they can't read it or if split is not enabled. I've seen mode FM with 0 Hz filter in some cases. So the behavior is very much undefined even now.

Having *known* and well-specific "not available" values (without forcibly turning split ON or doing any VFO swapping, as split is not enabled) would IMO still be a better option.

I'd still propose we'd go that way.

mikaelnousiainen, 2023-10-24:
@mdblack98 While trying to work through some simplifications in the split management code, could you confirm one thing:

It seems the `rig_cache` struct (and `rig_state` with its `tx_vfo` and `rx_vfo`) here:

https://github.com/Hamlib/Hamlib/blob/master/include/hamlib/rig.h#L2492

contain a top-level cache for all of the things defined in `icom_priv_data` here:

https://github.com/Hamlib/Hamlib/blob/master/rigs/icom/icom.h#L261

Question: To me it looks like we could remove the Icom priv_data counterparts and simply use the top-level cache for everything, would you agree? Was it just that the `rig_cache` items were introduced later than the Icom ones, so the Icom priv_data is still in use (as a remnant of not migrating them to the top-level cache)?

The reason for this simplification is just to make sure the freq/mode/split operations always use the most recent cached data. Now there are two places where these exist and it's not always clear which one is up-to-date.

I can do some extended testing on multiple Icom rigs -- mainly asking if you see any reasons NOT to do this?

mikaelnousiainen, 2023-10-24:
@mdblack98 Thanks for the comments.

Yet another set of questions, as I work through the split / VFO code in Hamlib:

1. Isn't the whole point of the icom_priv_data struct to be **private**, used only within Hamlib AND within the Icom backend -- it is not exported by the dynamic link libraries, so we should be able to freely change its contents?

2. Regarding VFO detection, some more recent Icom rigs can tell the Main/Sub selection, e.g. IC-7610, IC-7800, IC-785x and IC-9700 (but 9700 doesn't still indicate VFO A/B, so it's not as useful). I'm seeing there is some code for this in `icom_get_vfo()`, but you've removed it in this commit: https://github.com/Hamlib/Hamlib/commit/ce052db3a290fd5f00050d79541f969ba9195d79 -- was there a specific issue you ran into with this?

mdblack98, 2023-10-24:
It's meant to be private but it is still accessible to clients through rig->caps->priv

We just have to be pedantic about maintaining structure offsets.

Mike

I tried doing a get_vfo 

#1 0x25 read both VFOs -- if both are equal can't tell
#2 Select VFOA -- read freq/mode should match one
#3 Select VFOB -- read freq/mode should match the other

Tested it for a short while...problem is the VFO flashing that occurs so backed it out plus if user changes things we can't detect it.

I don't know why Icom has refused to make such a command....other then they think we (i.e. they) don't need it.

Does like we could do get_vfo for these rigs that only have MAIN/SUB

ic7600.c:#define IC7600_VFOS (RIG_VFO_MAIN|RIG_VFO_SUB|RIG_VFO_MEM)
ic7610.c:#define IC7610_VFOS (RIG_VFO_MAIN|RIG_VFO_SUB|RIG_VFO_MEM)
ic7800.c:#define IC7800_VFOS (RIG_VFO_MAIN|RIG_VFO_SUB|RIG_VFO_MEM)
ic785x.c:#define IC785x_VFOS (RIG_VFO_MAIN|RIG_VFO_SUB|RIG_VFO_MEM)

Mike

@mdblack98 Thanks for the comments.

Yet another set of questions, as I work through the split / VFO code in Hamlib:

     
    1.  
Isn't the whole point of the icom_priv_data struct to be private, used only within Hamlib AND within the Icom backend -- it is not exported by the dynamic link libraries, so we should be able to freely change its contents?

    
    2.  
Regarding VFO detection, some more recent Icom rigs can tell the Main/Sub selection, e.g. IC-7610, IC-7800, IC-785x and IC-9700 (but 9700 doesn't still indicate VFO A/B, so it's not as useful). I'm seeing there is some code for this in icom_get_vfo(), but you've removed it in this commit: ce052db -- was there a specific issue you ran into with this?

mikaelnousiainen, 2023-10-24:
> It's meant to be private but it is still accessible to clients through rig->caps->priv

But the struct is not even available in hamlib/rig.h, so accessing it would not ever be safe. All of the library data is accessible, of course, but we have to draw a line somewhere. If it's called "priv" and it's a void pointer, it has to be assumed to be private, right?

> We just have to be pedantic about maintaining structure offsets.

Fully understand this for ANY of the public/exported structs.

For the Icom priv_data, I would not worry. It's also backend-specific.

> Tested it for a short while...problem is the VFO flashing that occurs so backed it out plus if user changes things we can't detect it.
> I don't know why Icom has refused to make such a [...]

Yeah, I understand also the problem with 0x25 and 0x26 commands for VFO A/B rigs.

But there is another command, 0x07 0xD2 to get Main/Sub and it looks like that was implemented and possibly in use for a short while. Maybe it was not tested. I can do some testing on IC-7610 and IC-9700. The "regular" VFO A/B rigs do not support this and would not have get_vfo.

mdblack98, 2023-10-24:
I would prefer to just deprecate things like that with a comment after the field and not eliminate them from the structure.
I've been bitten too many times with edge cases.
The priv structure is available to a client via rig->caps.priv

The 9700 Main/Sub is only available in dual watch mode.  I've got one here for testing.

get_vfo is pretty useless if only partially implemented so I would not recommend doing it for the 9700.
The 9700 does have the ability to set freq/mode for MainA, MainB, SubA, and SubB

Mike W9MDB

>  
> It's meant to be private but it is still accessible to clients through rig->caps->priv

But the struct is not even available in hamlib/rig.h, so accessing it would not ever be safe. All of the library data is accessible, of course, but we have to draw a line somewhere. If it's called "priv" and it's a void pointer, it has to be assumed to be private, right?

>  
> We just have to be pedantic about maintaining structure offsets.

Fully understand this for ANY of the public/exported structs.

For the Icom priv_data, I would not worry. It's also backend-specific.

> Tested it for a short while...problem is the VFO flashing that occurs so backed it out plus if user changes things we can't detect it.
> I don't know why Icom has refused to make such a [...]

Yeah, I understand also the problem with 0x25 and 0x26 commands for VFO A/B rigs.

But there is another command, 0x07 0xD2 to get Main/Sub and it looks like that was implemented and possibly in use for a short while. Maybe it was not tested. I can do some testing on IC-7610 and IC-9700. The "regular" VFO A/B rigs do not support this and would not have get_vfo.

mikaelnousiainen, 2023-10-24:
Ok, let's go with deprecation.

And yeah, for IC-9700 the Main/Sub choices are not enough. For IC-7610, IC-7800 and IC-785x it would work, I guess.

mikaelnousiainen, 2023-10-29:
Just noting that I'm also working on the targeting/split changes too now. Getting soon to testing on real rigs. I'm attempting to simplify / unify certain things across the backends and commands related to targeting and split, so they will behave in a predictable way.

mikaelnousiainen, 2023-11-09:
@mdblack98 After the initial multicast work, I was able to resume working on these changes. Note that the change set for the VFO targeting + split commands will be quite large. The high-level (`rig.c`) changes are rather simple, the way we discussed earlier.

The Icom backend (`icom.c`) has gone through more changes to streamline all freq/mode commands (incl. split) to allow use of the targeted freq/mode Icom CI-V commands consistently. The goal is to have ZERO VFO swapping on Icoms that have the new 0x25/0x26 commands (over all Hamlib commands, including the SPLIT ones). I've attempted to clean up much of the code to make it easier to maintain!

The rest of the backends don't really need changes right now, as the Kenwood/Yaesu protocols are already handled more or less correctly regarding VFO targeting. Those will need testing too, because of the high-level changes

Next, I will start local testing and after a couple of days I will open a PR for this.

One question though: As these changes may introduce bugs, would you rather merge them now for immediate testing (before 4.6 release) or for a later release?

mdblack98, 2023-11-09:
gpredict would be the most likely bug reporters and I have very few testers there.
Doppler  and  and manual VFO tracking are the bugaboos.

Probably ought to create a doppler emulator -- something that bumps the frequency up for rx and tx appropriately every second or so and then can test VFO action with it.

So let's wait for 4.7 for that patch.

Mike W9MDB

mikaelnousiainen, 2023-11-10:
@mdblack98 Ok, makes sense to wait for 4.6 release first. I'll try to keep this PR up-to-date with Hamlib master branch for a couple of weeks until 4.6 is release -- it'll be happening this year anyway, right?

So far, initial testing on IC-7300 has been successful, with no VFO swapping -- now, when the rig advertises TARGETABLE_FREQ and/or TARGETABLE_MODE, apps like loggers can easily poll/read *AND change* both the VFOA/VFOB frequencies and also the split freq/mode without VFO swapping. Reverse split with VFOA as TX vfo seems to work too. I will test with older rigs too (have an IC-7600 and IC-756Pro3 available too). Obviously, the older rigs (without 0x25/0x26 commands) still need VFO swapping if polling actively or changing opposite VFO freq/mode.

The UDP multicast will of course make polling unnecessary in the future -- once apps adopt it!

A small question to you though: I saw that you had been working on improving IC-9700 support (mainly the VFO selection) this year. That is all working and clear to me how it works, but did you ever test whether the 0x25/0x26 commands could read the SUB receiver freq/mode? To me it looks like those commands only read the MAIN receiver VFO (when not in satellite mode). That is very confusing. Selecting the SUB receiver using the CI-V command or by tapping the screen doesn't affect the 0x25/0x26 results at all.

Maybe the "old-style" freq get/set commands 0x03 and 0x05 are the only ones that are able to set the SUB receiver freq?

mikaelnousiainen, 2024-01-21:
Closing this as the discussed changes are implemented in #1481

closed 2024-01-21 by mikaelnousiainen