Hamlib/Hamlib #1402: LEVEL_USB_AF and LEVEL_USB_AF_INPUT

mdblack98, 2023-10-18:
Add LEVEL_USB_AF_INPUT and make equivalent to TOK_LEVEL__USB_AUDIO_INPUT for backwards compatbility
Make LEVEL_USB_AF equivalent to TOK_LEVEL_USB_AUDIO_OUTPUT

mdblack98, 2023-10-18:
TS_590S/SG complete
Need to review other rigs for this capability

GaryKO3F, 2023-10-19:
When building Hamlib this morning in JTSDK, I got the following error:

  CC       ts590.lo
../../../src/rigs/kenwood/ts590.c: In function 'ts590_set_ext_level':
../../../src/rigs/kenwood/ts590.c:1278:5: warning: overflow in conversion from 'long long unsigned int' to 'token_t' {aka 'long int'} changes value from '2251799813685248' to '0' [-Woverflow]
 1278 |     case RIG_LEVEL_USB_AF_INPUT: // this one expect 0-1.0 levels
      |     ^~~~
../../../src/rigs/kenwood/ts590.c:1305:5: warning: overflow in conversion from 'long long unsigned int' to 'token_t' {aka 'long int'} changes value from '1125899906842624' to '0' [-Woverflow]
 1305 |     case RIG_LEVEL_USB_AF:
      |     ^~~~
../../../src/rigs/kenwood/ts590.c:1305:5: error: duplicate case value
../../../src/rigs/kenwood/ts590.c:1278:5: note: previously used here
 1278 |     case RIG_LEVEL_USB_AF_INPUT: // this one expect 0-1.0 levels
      |     ^~~~
make[1]: *** [Makefile:568: ts590.lo] Error 1
make[1]: Leaving directory '/home/charlesrogers/src/hamlib/build/rigs/kenwood'
make: *** [Makefile:635: all-recursive] Error 1

CLEAR SOURCE then CLOSE and RESTART MSYS2 IF YOU INTEND TO BUILD STATIC LIBRARIES

And Raspberry Pi Bullseye:

  CC       ts590.lo
ts590.c: In function 'ts590_set_ext_level':
ts590.c:1278:5: warning: overflow in conversion from 'long long unsigned int' to 'token_t' {aka 'long int'} changes value from '2251799813685248' to '0' [-Woverflow]
 1278 |     case RIG_LEVEL_USB_AF_INPUT: // this one expect 0-1.0 levels
      |     ^~~~
ts590.c:1305:5: warning: overflow in conversion from 'long long unsigned int' to 'token_t' {aka 'long int'} changes value from '1125899906842624' to '0' [-Woverflow]
 1305 |     case RIG_LEVEL_USB_AF:
      |     ^~~~
ts590.c:1305:5: error: duplicate case value
ts590.c:1278:5: note: previously used here
 1278 |     case RIG_LEVEL_USB_AF_INPUT: // this one expect 0-1.0 levels
      |     ^~~~
make[1]: *** [Makefile:566: ts590.lo] Error 1
make[1]: Leaving directory '/home/pi/Downloads/Hamlib-master/rigs/kenwood'
make: *** [Makefile:634: all-recursive] Error 1
***@***.***:~/Downloads/Hamlib-master $ 

Surprisingly, it built without errors on Mac Sonoma.

mdblack98, 2023-10-20:
Now builds OK for KO3F

N0NB, 2025-07-31:
Functionality appears to be available and working.

closed 2025-07-31 by N0NB