Hamlib/Hamlib #1409: multicast_publisher UDP packet: No error

mdblack98, 2023-10-25:
[2023-10-25 12:58:11.849609][00:00:01.639873][RIGCTRL:debug] 3:network.c(1068):network_multicast_publisher_start entered
[2023-10-25 12:58:11.849609][00:00:01.639884][RIGCTRL:debug] network.c(1072):multicast address=224.0.0.1, port=4532
[2023-10-25 12:58:11.849609][00:00:01.639893][RIGCTRL:debug] network_init: WSAStartup OK
[2023-10-25 12:58:11.849609][00:00:01.639981][RIGCTRL:debug] network.c(1108) MULTICAST_TRANSCEIVE enabled
[2023-10-25 12:58:11.849609][00:00:01.639990][RIGCTRL:error] network.c(1119) unknown MULTICAST item requested=0x1
[2023-10-25 12:58:11.849609][00:00:01.640133][RIGCTRL:debug] 3:network.c(1168):network_multicast_publisher_start returning(0)
[2023-10-25 12:58:11.849609][00:00:01.640147][RIGCTRL:debug] rig.c(1503):rig_open returning2(0)
[2023-10-25 12:58:11.849609][00:00:01.640158][RIGCTRL:debug] 3:rig.c(3132):rig_get_vfo entered
[2023-10-25 12:58:11.849609][00:00:01.640165][RIGCTRL:warning] rig_get_vfo: no get_vfo
[2023-10-25 12:58:11.849609][00:00:01.640174][RIGCTRL:debug] 3:rig_get_vfo: elapsed=0ms
[2023-10-25 12:58:11.849609][00:00:01.640186][RIGCTRL:debug] 3:rig.c(3141):rig_get_vfo returning(-11) Feature not available
[2023-10-25 12:58:11.849609][00:00:01.640194][RIGCTRL:debug] 3:rig.c(2161):rig_get_freq entered
[2023-10-25 12:58:11.849609][00:00:01.640200][RIGCTRL:debug] rig_get_freq called vfo=currVFO
[2023-10-25 12:58:11.849609][00:00:01.640208][RIGCTRL:debug] rig_get_freq(2180) called vfo=currVFO
[2023-10-25 12:58:11.849609][00:00:01.640221][RIGCTRL:trace] vfo_fixup:(from rig_get_freq:2186) vfo=currVFO, vfo_curr=VFOA, split=0
[2023-10-25 12:58:11.849609][00:00:01.640227][RIGCTRL:trace] vfo_fixup: Leaving currVFO alone
[2023-10-25 12:58:11.849609][00:00:01.640235][RIGCTRL:debug] rig.c(2189) vfo=currVFO, curr_vfo=VFOA
[2023-10-25 12:58:11.849609][00:00:01.640262][RIGCTRL:trace] rig_get_freq: VFOA cache hit age=137ms, freq=28074000, use_cached_freq=0
[2023-10-25 12:58:11.849609][00:00:01.640269][RIGCTRL:debug] 3:rig_get_freq: elapsed=0ms
[2023-10-25 12:58:11.849609][00:00:01.640277][RIGCTRL:debug] 3:rig.c(2274):rig_get_freq returning(0)
[2023-10-25 12:58:11.849609][00:00:01.640323][RIGCTRL:debug] network.c(915): Starting multicast publisher
[2023-10-25 12:58:11.849609][00:00:01.640561][RIGCTRL:error] multicast_publisher: error sending UDP packet: No error

mikaelnousiainen, 2023-10-29:
@mdblack98 Is this on Windows?

mdblack98, 2023-10-29:
Yes...a Windows 10 user.I can't reproduce on my virtualbox Windows 10.
What fixes it is running message_aggregator for some reason I have yet to figure out.
MyApp does not receive anything on his system either until message_aggregator is run and then the packets start flowing and the error message in WSJT-X stops.

mikaelnousiainen, 2023-10-29:
@mdblack98 Ok, interesting. What is message_aggregator?

mdblack98, 2023-10-29:
You'll find it in wsjtx\bin -- it just shows the messages being sent to by WSJT-X as a 1st step to ensure it's actually doing something.And for some odd reason it fixes the problem when it  is run.

mdblack98, 2023-12-09:
Behavior is better now.

closed 2023-12-09 by mdblack98