Hamlib/Hamlib #1412: 700ms  delay needed for ptt on transition

mdblack98, 2023-10-31:
My 10GHz set needs 700 ms to go from receive to transmit due to the waveguide relay. So I think the audio must be delayed at least 700 ms with respect to the PTT-signal.
I could not set the TX delay to a value higher than 0.5 s.
And I am worried that the delay is too long, so that the transmission is too late

GeoBaltz, 2023-11-01:
This commit causes me to lose PTT control from wsjt-x (on my TS-890S). Verified with git bisect.

mdblack98, 2023-11-01:
Try the latest and if that doesn't work please provide  debug.
Please place this file as described belowhttps://www.dropbox.com/s/t52ngcalsgnpm8m/wsjtx_log_config.ini?dl=0
 C:\Users\[username]\AppData\Local\WSJT-X The WSJT-X_Rigcontrol.log file will be in the same location Here's a youtube video showing Windows users howto -- https://www.youtube.com/watch?v=q9tAvhNmSvc
For Linux put it in   ~/.config The WSJT-X_Rigcontrol.log file will be here: ~/.local/share/WSJT-X
For MacOS put it in /Users/[username]/Library/Preferences Restart WSJT-X and duplicate the problem.Shut down WSJT-X
Then send me the WSJT-X_RigControl.log fileMike W9MDB

GeoBaltz, 2023-11-09:
Still broken with current git 
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/13309778/WSJT-X_RigControl.log)

Still looks like this is caused by the additional item added to hamlib_port_t by commit 6cb17e49dc "Add post_ptt_delay for situations..."

mdblack98, 2023-11-09:
Please try the latest.....

Should be fixed  now.

Still looks like this is caused by the additional item added to hamlib_port_t by commit 6cb17e4 "Add post_ptt_delay for situations..."

GeoBaltz, 2023-11-09:
That was with the latest, as of 2023-11-09 10:20EST

GeoBaltz, 2023-11-09:
Pulled, made, installed, ran again - still broken.

mdblack98, 2023-11-09:
No it wasn't -- check the log

 2023-10-25T18:08:07Z

GeoBaltz, 2023-11-09:
Sorry gotta leave for appointment. Which log was that timestamp from?  Today's log attached 5 comments above.

mdblack98, 2023-11-09:
The  one you just posted.

https://github.com/Hamlib/Hamlib/files/13309778/WSJT-X_RigControl.log

All you have to do is "rigctld --version"  to see what version you are running.

GeoBaltz, 2023-11-09:
OK, back again.  
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/13312460/WSJT-X_RigControl.log)

Problem still exists.

Aside:  bogus version info due to stale copy of hamlibdatetime.h in Hamlib/src/, which was then included instead of newly created version in Hamlib/build/src/.  Dunno how it got there; this version from latest `git pull;make clean;make;`

GeoBaltz, 2023-11-15:
I see my emails still don't show up here.

Anyway, still broken with latest git master - no PTT control with wsjtx(2.6.1)
Latest debug log: 
[WSJT-X_RigControl.log](https://github.com/Hamlib/Hamlib/files/13365486/WSJT-X_RigControl.log)

ISTM the problem is with the item added to struct hamlib_port, in spite of the comment/warning to not touch it until 5.0.  Binary or shared library incompatibility??

closed 2023-12-14 by mdblack98