Hamlib/Hamlib #1418: HAMLIB IO ERROR

gridbasedotnet, 2023-11-07:
[image]
[image]

Hello! I have successfully made hamlib work inside WSJT-X & JS8CALL on windows & mac operating systems. However, I cannot get hamlib working on Debian Bullseye aarch64 because of a constant error with hamlib (see attached). Doesn't matter what I do. I have tried doing a ferrite choke on the USB cable, I have tried every troubleshooting method I could find with no avail. Any help is appreciated. Or, if this is a known issue, that would be helpful to know as well.

mdblack98, 2023-11-07:
I assume this is from a recent Hamlib version?
And you don't say what rig you have.
But 311 is definitely not a valid rig model.
Do "rigctl -l | grep 2000" for example to find all modes with 2000 in the name

gridbasedotnet, 2023-11-10:
> I assume this is from a recent Hamlib version? And you don't say what rig you have. But 311 is definitely not a valid rig model. Do "d" for example to find all modes with 2000 in the name

Correct. It is from the most recent stable hamlib version. I have an ICOM-IC-706. After running your command to find that the rig was actually 3011, not 311, this is what I come up with. 
[image]

mdblack98, 2023-11-10:
IC706 is 1 stop-bit -- but that should not be the problem.
You can shorten up your line and just let it use the defaults and add -vvv to see errors/warnings

rigctld -m 3011 -r /dev/ttyUSB0 --set-conf=dtr_state=OFF,rts_state=ON -vvv
Could be your serial cable may want dtr_state ON but that would  typically be for an older USB cable.
Try this and see what it says...
rigctld -m 3011 -r /dev/ttyUSB0 -vvv
 Mike W9MDB

gridbasedotnet, 2023-11-10:
When i ran your command, 
"rigctld -m 3011 -r /dev/ttyUSB0 -vvv
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
serial_open: Unable to open /dev/ttyUSB0 - Device or resource busy
rig_open: error = IO error "

When I ran it with 5x v's
"sudo rigctld -m 3011 -r /dev/ttyUSB0 -vvvvv
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
rigctld Hamlib 4.0
Last commit was Fri Dec 25 02:30:30 2020 +0000 SHA=519f82
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
rig_register called
rig_register: rig_register (3085)
rig_register called
rig_register: rig_register (3009)
rig_register called
rig_register: rig_register (3010)
rig_register called
rig_register: rig_register (3011)
rig_register called
rig_register: rig_register (3013)
rig_register called
rig_register: rig_register (3014)
rig_register called
rig_register: rig_register (3015)
rig_register called
rig_register: rig_register (3019)
rig_register called
rig_register: rig_register (3020)
rig_register called
rig_register: rig_register (3021)
rig_register called
rig_register: rig_register (3022)
rig_register called
rig_register: rig_register (3067)
rig_register called
rig_register: rig_register (3023)
rig_register called
rig_register: rig_register (3046)
rig_register called
rig_register: rig_register (3024)
rig_register called
rig_register: rig_register (3028)
rig_register called
rig_register: rig_register (3030)
rig_register called
rig_register: rig_register (3026)
rig_register called
rig_register: rig_register (3027)
rig_register called
rig_register: rig_register (3047)
rig_register called
rig_register: rig_register (3057)
rig_register called
rig_register: rig_register (3063)
rig_register called
rig_register: rig_register (3029)
rig_register called
rig_register: rig_register (3062)
rig_register called
rig_register: rig_register (3045)
rig_register called
[... 146 lines cut ...]
write_block(): TX 7 bytes
0000    fe fe 58 e0 19 00 fd                                ..X....         
read_string called, rxmax=80
read_string(): RX 4 characters
0000    f8 f8 00 00                                         ....            
read_string called, rxmax=76
read_string(): Timed out 1.423 seconds after 0 chars
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 58 e0 19 00 fd                                ..X....         
read_string called, rxmax=80
read_string(): RX 4 characters
0000    f8 f8 00 00                                         ....            
read_string called, rxmax=76
read_string(): Timed out 1.1038 seconds after 0 chars
icom_transaction: failed: Communication timed out
rig_get_vfo called
rig_get_vfo: no get_vfo
rig_open: Icom rig so default vfo = currVFO
Opened rig model 3011, 'IC-706MkIIG'
Backend version: 20210107.0, Status: Stable
main: Using IPV4"

mdblack98, 2023-11-10:
What USB cable do you have?
Try adding 
--set-conf=dtr_state=ON--set-conf=rts_state=ON
One  or the other might provide power to your USB cable.
Mike

gridbasedotnet, 2023-11-10:
sudo rigctld -m 3011 --set-conf=rts_state=ON -r /dev/ttyUSB0 -vvvvv
Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
rigctld Hamlib 4.0
Last commit was Fri Dec 25 02:30:30 2020 +0000 SHA=519f82
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
rig_register called
rig_register: rig_register (3085)
rig_register called
rig_register: rig_register (3009)
rig_register called
rig_register: rig_register (3010)
rig_register called
rig_register: rig_register (3011)
rig_register called
rig_register: rig_register (3013)
rig_register called
rig_register: rig_register (3014)
rig_register called
rig_register: rig_register (3015)
rig_register called
rig_register: rig_register (3019)
rig_register called
rig_register: rig_register (3020)
rig_register called
rig_register: rig_register (3021)
rig_register called
rig_register: rig_register (3022)
rig_register called
rig_register: rig_register (3067)
rig_register called
rig_register: rig_register (3023)
rig_register called
rig_register: rig_register (3046)
rig_register called
rig_register: rig_register (3024)
rig_register called
rig_register: rig_register (3028)
rig_register called
rig_register: rig_register (3030)
rig_register called
rig_register: rig_register (3026)
rig_register called
rig_register: rig_register (3027)
rig_register called
rig_register: rig_register (3047)
rig_register called
rig_register: rig_register (3057)
rig_register called
rig_register: rig_register (3063)
rig_register called
rig_register: rig_register (3029)
rig_register called
rig_register: rig_register (3062)
rig_register called
rig_register: rig_register (3045)
rig_register called
rig_register: rig_register (3056)
rig_register called
rig_register: rig_register (3075)
rig_register called
rig_register: rig_register (3060)
rig_register called
rig_register: rig_register (3070)
rig_register called
rig_register: rig_register (3061)
rig_register called
rig_register: rig_register (3073)
rig_register called
rig_register: rig_register (3078)
rig_register called
rig_register: rig_register (3031)
rig_register called
rig_register: rig_register (3012)
rig_register called
rig_register: rig_register (3016)
rig_register called
rig_register: rig_register (3017)
rig_register called
rig_register: rig_register (3032)
rig_register called
rig_register: rig_register (3034)
rig_register called
rig_register: rig_register (3044)
rig_register called
rig_register: rig_register (3068)
rig_register called
rig_register: rig_register (3035)
rig_register called
rig_register: rig_register (3081)
rig_register called
rig_register: rig_register (3069)
rig_register called
rig_register: rig_register (3077)
rig_register called
rig_register: rig_register (3036)
rig_register called
rig_register: rig_register (3058)
rig_register called
rig_register: rig_register (3080)
rig_register called
rig_register: rig_register (3037)
rig_register called
rig_register: rig_register (3038)
rig_register called
rig_register: rig_register (3039)
rig_register called
rig_register: rig_register (3040)
rig_register called
rig_register: rig_register (3041)
rig_register called
rig_register: rig_register (3042)
rig_register called
rig_register: rig_register (3079)
rig_register called
rig_register: rig_register (3043)
rig_register called
rig_register: rig_register (3066)
rig_register called
rig_register: rig_register (3003)
rig_register called
rig_register: rig_register (3004)
rig_register called
rig_register: rig_register (3006)
rig_register called
rig_register: rig_register (3007)
rig_register called
rig_register: rig_register (3002)
rig_register called
rig_register: rig_register (3052)
rig_register called
rig_register: rig_register (3053)
rig_register called
rig_register: rig_register (3051)
rig_register called
rig_register: rig_register (3064)
rig_register called
rig_register: rig_register (3065)
rig_register called
rig_register: rig_register (3054)
rig_register called
rig_register: rig_register (3083)
rig_register called
rig_register: rig_register (3084)
rig_register called
rig_register: rig_register (3082)
rig_register called
rig_register: rig_register (3071)
rig_register called
rig_register: rig_register (3072)
rig_register called
rig_register: rig_register (3074)
rig_register called
rig_register: rig_register (3076)
icom_init called
icom_init: done
rig_token_lookup called
rig_confparam_lookup called
rig_set_conf called
rig_confparam_lookup called
rig_set_conf: rts_state='ON'
main: twiddle=0, uplink=0
rig_open called
port_open called
serial_open called
serial_setup called
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: tcsetattr TCSANOW
serial_flush called
tcflush
ser_set_rts called
ser_set_rts: RTS=1
icom_rig_open 758 
icom_rig_open: IC-706MkIIG v20210107.0
icom_get_usb_echo_off called
icom_get_usb_echo_off: retry temp set to 1
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0, data_len=80
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 6 bytes
0000    fe fe 58 e0 03 fd                                   ..X...          
read_string called, rxmax=80
read_string(): RX 6 characters
0000    fe fe 58 e0 03 fd                                   ..X...          
read_string called, rxmax=80
read_string(): RX 11 characters
0000    fe fe e0 58 03 00 58 74 24 00 fd                    ...X..Xt$..     
icom_one_transaction: data_len=6, frm_len=11
icom_get_usb_echo_off: ack_len=6
icom_get_usb_echo_off: USB echo on detected
rig_get_vfo called
rig_get_vfo: no get_vfo
rig_open: Icom rig so default vfo = currVFO
Opened rig model 3011, 'IC-706MkIIG'
Backend version: 20210107.0, Status: Stable
main: Using IPV4

mdblack98, 2023-11-10:
Turn off echo on the rig.

gridbasedotnet, 2023-11-11:
> Turn off echo on the rig.

I cannot find how to turn off echo rigctl --help. If you could provide a command, that would be helpful. 

Additionally, does it matter that I am using CI-V?

gridbasedotnet, 2023-11-11:
> Turn off echo on the rig.

rigctld -m 3011 --set-conf=rts_state=ON -c -r /dev/ttyUSB0 -vvvvv

Recommend using --vfo switch for rigctld if client supports it
rigctl and netrigctl will automatically detect vfo mode
rigctld Hamlib 4.0
Last commit was Fri Dec 25 02:30:30 2020 +0000 SHA=519f82
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_init called
initrigs4_icom: _init called
rig_register called
rig_register: rig_register (3055)
rig_register called
rig_register: rig_register (3085)
rig_register called
rig_register: rig_register (3009)
rig_register called
rig_register: rig_register (3010)
rig_register called
rig_register: rig_register (3011)
rig_register called
rig_register: rig_register (3013)
rig_register called
rig_register: rig_register (3014)
rig_register called
rig_register: rig_register (3015)
rig_register called
rig_register: rig_register (3019)
rig_register called
rig_register: rig_register (3020)
rig_register called
rig_register: rig_register (3021)
rig_register called
rig_register: rig_register (3022)
rig_register called
rig_register: rig_register (3067)
rig_register called
rig_register: rig_register (3023)
rig_register called
rig_register: rig_register (3046)
rig_register called
rig_register: rig_register (3024)
rig_register called
rig_register: rig_register (3028)
rig_register called
rig_register: rig_register (3030)
rig_register called
rig_register: rig_register (3026)
rig_register called
rig_register: rig_register (3027)
rig_register called
rig_register: rig_register (3047)
rig_register called
rig_register: rig_register (3057)
rig_register called
rig_register: rig_register (3063)
rig_register called
rig_register: rig_register (3029)
rig_register called
rig_register: rig_register (3062)
rig_register called
rig_register: rig_register (3045)
rig_register called
rig_register: rig_register (3056)
rig_register called
rig_register: rig_register (3075)
rig_register called
rig_register: rig_register (3060)
[... 147 lines cut ...]
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 00 e0 19 00 fd                                .......         
read_string called, rxmax=80
read_string(): Timed out 1.1069 seconds after 0 chars
rig_flush: called for serial device
serial_flush called
tcflush
write_block called
write_block(): TX 7 bytes
0000    fe fe 00 e0 19 00 fd                                .......         
read_string called, rxmax=80
read_string(): Timed out 1.1068 seconds after 0 chars
icom_transaction: failed: Communication timed out
rig_get_vfo called
rig_get_vfo: no get_vfo
rig_open: Icom rig so default vfo = currVFO
Opened rig model 3011, 'IC-706MkIIG'
Backend version: 20210107.0, Status: Stable
main: Using IPV4
binding failed (trying next interface): Network error 98: Address already in use
binding failed (trying next interface): Network error 98: Address already in use
main: bind error - no available interface

mdblack98, 2023-11-11:
Please upgrade to the latest  hamlib.....
https://n0nb.users.sourceforge.net/

gridbasedotnet, 2023-11-11:
After updating and rebooting getting this error. 

rigctl --version
rigctl: symbol lookup error: rigctl: undefined symbol: lock_mode

mdblack98, 2023-11-11:
Remove any existing hamlib package.
Default build goes in /usr/local/lib and you probably had libhamlib* in /usr/lib
Removing the package should get rid of it.
Mike W9MDB

gridbasedotnet, 2023-11-11:
I have removed the existing hamlib package from /usr/local/bin/ and confirmed there was no "libhamlib" in /usr/lib. I am still getting the same error, "rigctl: symbol lookup error: rigctl: undefined symbol: lock_mode".

mdblack98, 2023-11-11:
Do "ldd rigctld" | grep hamlib 
That  will show  you the path being referenced.
Mike

gridbasedotnet, 2023-11-11:
ldd: ./rigctl: No such file or directory

mdblack98, 2023-11-11:
You need the full path to the file
ldd /usr/local/bin/rigctld
Also make sure you don't have /usr/bin/rigctl or /usr/bin/rigctld
Mike W9MDB

N0NB, 2023-11-11:
* On 2023 11 Nov 11:15 -0600, Michael Black wrote:
> You need the full path to the file
> ldd /usr/local/bin/rigctld

Or:

ldd $(which rigctl)

which will exec the first rigctl in the $PATH.

Also:

which rigctl

will print the absolute path of the first rigctl in the $PATH.  This is
useful for making sure that the executable found first in $PATH is
actually the one intended to be run.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

gridbasedotnet, 2023-11-12:
I saved a back-up before I updated Hamlib. So, I have gone back to the back-up to perform ldd $(which rigctl). I wanted to be sure that it was in the right place to begin with. Here is the output from that. 

ldd $(which rigctl)
	linux-vdso.so.1 (0x0000007f8e18d000)
	libreadline.so.8 => /lib/aarch64-linux-gnu/libreadline.so.8 (0x0000007f8e0a0000)
	libhamlib.so.4 => /lib/aarch64-linux-gnu/libhamlib.so.4 (0x0000007f8d75f000)
	libpthread.so.0 => /lib/aarch64-linux-gnu/libpthread.so.0 (0x0000007f8d72e000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000007f8d5ba000)
	/lib/ld-linux-aarch64.so.1 (0x0000007f8e15d000)
	libtinfo.so.6 => /lib/aarch64-linux-gnu/libtinfo.so.6 (0x0000007f8d57c000)
	libdl.so.2 => /lib/aarch64-linux-gnu/libdl.so.2 (0x0000007f8d568000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000007f8d4bd000)
	libusb-1.0.so.0 => /lib/aarch64-linux-gnu/libusb-1.0.so.0 (0x0000007f8d491000)
	libudev.so.1 => /lib/aarch64-linux-gnu/libudev.so.1 (0x0000007f8d45a000)

Now, I am about to perform the update to get 4.6 installed. After I install it, "ldconfig" should update the path, correct?

mdblack98, 2023-11-12:
Also just do
which rigctl
The /lib/aarch.... path looks wrong.
What Linux OS is this?
Also what date does this show....I'm betting it's an older date.
ls -l /lib/aarch64-linux-gnu/libhamlib.so.4
Mike W9MDB

gridbasedotnet, 2023-11-13:
which rigctl
/usr/bin/rigctl

ls -l /lib/aarch64-linux-gnu/libhamlib.so.4
lrwxrwxrwx 1 root root 18 Jul 15  2021 /lib/aarch64-linux-gnu/libhamlib.so.4 -> libhamlib.so.4.0.0

Up-to-Date OS: Debian GNU/Linux 11 (bullseye) aarch64

mdblack98, 2023-11-13:
That will be the wrong rigctl if you are compiling Hamlib with the default switches.

You should have  one in /usr/local/bin

If you do have /usr/local/bin/rigctl then remove /usr/bin/rigctl* /usr/bin/rotctl*

Mike W9MDB

gridbasedotnet, 2023-11-13:
I do _not have /usr/local/bin/rigctl that might be the problem?

mdblack98, 2023-11-13:
You did compile from the latest source code?

Did  you do "make install" ?

Mike W9MDB

N0NB, 2023-11-13:
* On 2023 13 Nov 07:28 -0600, GRIDBASE wrote:
> which rigctl
> /usr/bin/rigctl
> 
> ls -l /lib/aarch64-linux-gnu/libhamlib.so.4
> lrwxrwxrwx 1 root root 18 Jul 15  2021 /lib/aarch64-linux-gnu/libhamlib.so.4 -> libhamlib.so.4.0.0
> 
> Up-to-Date OS: Debian GNU/Linux 11 (bullseye) aarch64

This indicates to me that you have the libhamlib-utils package installed
from the Debian repository.  Unfortunately, at least the cqrlog and
wstjx packages depend on libhamlib-utils being installed so there is no
easy way to remove it.

However, when compiling from source, the hamlib build scripts will
default to installing its files under the /user/local hierarchy which
would put rigctl under /usr/local/bin.  Also, by default, Debian will
place /usr/local/bin in your PATH variable before /usr/bin so that
locally compiled packages will have precedence.  You can verify this
with:

echo $PATH

This way, calling 'rigctl' will result in the locally compiled version
being called and the package management system will not complain.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

gridbasedotnet, 2023-11-14:
"You should have one in /usr/local/bin

If you do have /usr/local/bin/rigctl then remove /usr/bin/rigctl*"

You stated what to do if I do have an exec. rigctl in /usr/local/bin/. I appreciate that. 

But, what if I don't have one? In this case. I do not have a rigctl in /usr/local/bin. 

I only have on in /usr/bin/

mdblack98, 2023-11-14:
Then you didn't compile/install Hamlib correctly.
Show me the steps you are using to compile/install Hamlib.
Mike

mdblack98, 2023-11-17:
Menu 36 CI-V TRNTurn it off.

mdblack98, 2023-11-27:
> After our session the other day, I went to run WSJT-X and received this error. I ensured to killall gpsd before opening the program
> Hamlib error: 2:icom.c(5039):icom_set_conf entered
> 2:icom.c(5069):icom_set_conf returning(0)
> rig.c(891):rig_open entered
> rig_settings_get_path: path=/home/gridbase/.config/hamlib_settings
> rig_settings_load_all: settings_file (/home/gridbase/.config/hamlib_settings): No such file or directory
> rig_open: cwd=/home/gridbase
> rig_open: /home/gridbase/.config//home/gridbase/hamlib_settings does not exist
> rig_open: async_data_enable=0, async_data_supported=1
> serial_open: /dev/ttyUSB0
> serial_open(230): open failed#1
> serial_open(230): open failed#2
> serial_open(230): open failed#3
> serial_open(230): open failed#4

sudo lsof /dev/ttyUSB0

port_open: serial_open(/dev/ttyUSB0) status=-6, err=Device or resource busy
rig_open: rs->comm_state==0?=0rig.c(1101):rig_open returning2(-6) IO error

IO error
IO error
while opening connection to rig

closed 2023-12-25 by mdblack98