Hamlib/Hamlib #1420: rig_cache structure change to pointer

mdblack98, 2023-11-08:
We need to deprecate the rig_cache struct reference in rig_state and make a  new pointer reference  like we did for rig_state.

GeoBaltz, 2025-05-07:
I have gotten back to working on this, and even have a dynamically allocated cache version that works with WSJT-X.  However, I made a few changes while deprecating the definitions in rig.h that may need to be discussed.

The definition in rig.h was changed from 'struct rig_cache' to 'struct rig_cache_deprecated', and the references in rig_state changed likewise.   However, the new definition went into src/cache.h, not include/hamlib/rig.h - out of scope for any application compiled with Hamlib.

I did this for several reasons -
- My general dislike for having active internal data shared between library and application - how the cached operations are really done is not germane to the app, only to Hamlib.
- Current programming practice - let the object manage its own data, without outside meddling.
- Having another albatross around the necks of current and future Hamlib developers.  See how much code was needed to handle previous deprecations, and even then compatibility is only in one direction.
- And the big technical reason - if Hamlib is ever going to handle rigs sending state changes automatically (Auto Info in Kenwood-speak), then the whole idea of cache needs to change.  STATE(rig) should always be up to date, commands sent to the rig are confirmed automatically, inquiries should have the requested data already available; cache may not even be involved.

Since I don't have a feel for what applications are doing with current cache data, I'm really looking for feedback on how to proceed.  Do I need to provide some sort of bridge to/from the old cache structure; are changes to the API necessary; or do I start over?  Should I start a discussion on the ML?

N0NB, 2025-05-08:
Yes, please start a ML discussion as these updates aren't forwarded to the list so far as I know and unless one is "subscribed" to an issue or actively looks for an issue, this update won't bee seen.

GeoBaltz, 2025-05-11:
On Thursday, May 8, 2025 7:33:20 AM Eastern Daylight Time Nate Bargmann wrote:
> N0NB left a comment (Hamlib/Hamlib#1420)
> 
> Yes, please start a ML discussion as these updates aren't forwarded to the
> list so far as I know and unless one is "subscribed" to an issue or
> actively looks for an issue, this update won't bee seen.
Will do.  As I got further into writing the comment on the issue, I started 
thinking it might deserve a wider audience.

N0NB, 2025-06-14:
Closing per request of George.

closed 2025-06-14 by N0NB