Hamlib/Hamlib #1435: Killing rigctld on Windows leaves protocol in funky state

georges, 2023-11-29:
**Background**
I'm creating a GoLang VARA [launcher](https://github.com/islandmagic/varanny) utility that triggers rigctld as needed. Because Windows lacks a graceful process shutdown mechanism, rigctld gets force closed when the session ends.

**Issue**
The problem sequence is:

1. Rigctld launches and functions normally at first.
2. The launcher force closes rigctld when the session is over.
3. Upon the second launch, rigctld usually encounters errors.

Rigctld is invoked with this command:
`C:\Program Files\hamlib-w64-4.5.5\bin\rigctld.exe -m 3085 -c 0xA4 -r COM7 -s 9600 -vvv`

The forced shutdown seems to leave the COM port or rig in a peculiar state. After a while, it appears to reset itself, but most commands return `RPRT -8` or `RPRT -9` codes in the meantime.

**Questions**

1. Can rigctld be shut down from the connection? Currently, I use a q command, but that only disconnects.
2. Is there a way to reset rigctld at startup to ensure it functions correctly?
3. Other suggestions?

**Error** Log
(When the connection is in an unusual state)

```
port_read_sync_data(): WaitForMultipleObjects() error: 0
read_string_generic(1333): I/O error after 0 chars, direct=0: -7
read_string_generic(): read failed, direct=0 - No such file or directory
icom_transaction: retry=1: write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
iofunc.c(501) trace
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
iofunc.c(514) trace
icom_one_transaction: DEBUG retval=6, frm_len=6, cmd=0x18
read_string_generic called, rxmax=200 direct=0, expected_len=1
iofunc.c(501) trace
iofunc.c(514) trace
iofunc.c(501) trace
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
iofunc.c(514) trace
read_string_generic(): read failed, direct=0 - No such file or directory
     5:frame.c(294):icom_one_transaction returning(-6) IO error

IO error
IO error

icom_transaction: retry=0: iofunc.c(514) trace
iofunc.c(501) trace
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
iofunc.c(514) trace
read_string_generic(): read failed, direct=0 - No such file or directory
     5:frame.c(294):icom_one_transaction returning(-6) IO error

IO error

read_string_generic called, rxmax=200 direct=1, expected_len=1
     5:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
icom_one_transaction: DEBUG retval=5, frm_len=6, cmd=0x18
icom_one_transaction: removing 0xfe power up echo, len=6     5:frame.c(242):icom_one_transaction returning(-8) Protocol error

Protocol error
Protocol error

icom_transaction: retry=0: rig_get_freq(2100): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: cache miss age=1000000ms, cached_vfo=None, asked_vfo=None, use_cached_freq=0
rig_get_freq(2130): vfo_opt=0, model=3085
icom_get_freq called for None, curr_vfo=None
icom_get_freq: using vfo=None
   3:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x25, subcmd=0x00, payload_len=0
    4:frame.c(138):icom_one_transaction entered
write_block(): TX 7 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
iofunc.c(501) trace
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
read_string_generic called, rxmax=200 direct=1, expected_len=1
iofunc.c(514) trace
icom_one_transaction: DEBUG retval=6, frm_len=7, cmd=0x25
    4:frame.c(242):icom_one_transaction returning(-8) Protocol error

Protocol error
Protocol error

icom_get_freq: rig probe shows 0x25 CI-V cmd not available
icom_transaction: retry=0: read_string_generic called, rxmax=200 direct=1, expected_len=1
Protocol error

   3:frame.c(455):icom_transaction returning(-8) Protocol error

icom_get_freq: rig probe shows 0x25 CI-V cmd not available
icom.c(1671) trace
set_vfo_curr: vfo=None, curr_vfo=None
set_vfo_curr: curr_vfo now=None
icom.c(9456):set_vfo_curr returning2(0)
   3:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
    4:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
icom_one_transaction: DEBUG retval=11, frm_len=6, cmd=0x03
    4:frame.c(242):icom_one_transaction returning(-8) Protocol error

Protocol error
Protocol error

rig_get_cache(303): unknown vfo=otherVFO
icom_set_vfo: unknown vfo 'otherVFO'
rig_set_vfo: set_vfo otherVFO failed with 'rig_get_freq: cache miss age=1000000ms, cached_vfo=otherVFO, asked_vfo=otherVFO, use_cached_freq=0
rig_get_freq(2130): vfo_opt=0, model=3085
icom_get_freq called for otherVFO, curr_vfo=None
icom_get_freq: using vfo=otherVFO
icom.c(1671) trace
set_vfo_curr: vfo=otherVFO, curr_vfo=None
set_vfo_curr: setting new vfo=otherVFO
icom.c(9441) trace
  2:rig.c(2771):rig_set_vfo entered
rig_set_vfo called vfo=otherVFO
vfo_fixup:(from rig_set_vfo:2791) vfo=otherVFO, vfo_curr=None, split=0
vfo_fixup:(from rig_set_vfo:2833) vfo=otherVFO, vfo_curr=None, split=0
rig.c(2852) trace
icom_set_vfo called vfo=otherVFO
icom_set_vfo: line#2878
icom_set_vfo: unknown vfo 'otherVFO'
icom.c(3078):icom_set_vfo returning2(-1) Invalid parameter

Invalid parameter
Invalid parameter
'
rig_get_vfo: no get_vfo
rig_get_cache(303): unknown vfo=otherVFO
rig_set_cache_freq(209): unknown vfo?, vfo=otherVFO
icom_transaction: retry=0: rig_get_freq(2100): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2100): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: cache miss age=10000ms, cached_vfo=None, asked_vfo=None, use_cached_freq=0
rig_get_freq(2130): vfo_opt=0, model=3085
icom_get_freq called for None, curr_vfo=None
icom_get_freq: using vfo=None
icom.c(1671) trace
set_vfo_curr: vfo=None, curr_vfo=None
set_vfo_curr: curr_vfo now=None
icom.c(9456):set_vfo_curr returning2(0)
 1:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
  2:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
icom_one_transaction: DEBUG retval=7, frm_len=6, cmd=0x03
  2:frame.c(242):icom_one_transaction returning(-8) Protocol error

Protocol error
Protocol error

icom_transaction: retry=0:                      -1:rig.c(1858):rig_set_freq returning(-8) Protocol error

                     -1:rig.c(2771):rig_set_vfo entered
rig_set_vfo called vfo=VFOA
vfo_fixup:(from rig_set_vfo:2791) vfo=VFOA, vfo_curr=None, split=0
vfo_fixup:(from rig_set_vfo:2833) vfo=VFOA, vfo_curr=None, split=0
rig.c(2852) trace
icom_set_vfo called vfo=VFOA
icom_set_vfo: line#2878
icom_set_vfo: line#3081
0:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x07, subcmd=0x00, payload_len=0
 1:frame.c(138):icom_one_transaction entered
write_block(): TX 7 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
icom_one_transaction: DEBUG retval=12, frm_len=7, cmd=0x07
 1:frame.c(242):icom_one_transaction returning(-8) Protocol error
```

georges, 2023-11-30:
Oddly enough, when running with `-vvvvv` I can't cause the problem to occur. Could it be some timing issue?

I got it to flake out running `-vvvv`. Log file attached.

[log-vvvv.txt](https://github.com/Hamlib/Hamlib/files/13508488/log-vvvv.txt)

mdblack98, 2023-11-30:
Can you send one with the 5 v's so I can the difference?

georges, 2023-11-30:
Log when working with `-vvvvv` 
[log-working-vvvvv.txt](https://github.com/Hamlib/Hamlib/files/13508674/log-working-vvvvv.txt)

georges, 2023-12-10:
Did some more investigating. Looks like a faulty USB cable was causing intermittent connection errors.

closed 2023-12-10 by georges