Hamlib/Hamlib #1436: rig_list_foreach API breakage in master

tmiw, 2023-11-29:
As reported in https://github.com/drowe67/freedv-gui/pull/613#issuecomment-1830778054, it appears that 4.6~git removed `const` from `rig_list_foreach` (specifically in commit https://github.com/Hamlib/Hamlib/commit/9841e500b94f642a0b4511991faccb0143c32475). Unfortunately, this breaks the current version of FreeDV:

```
/home/bdm/rpmbuild/BUILD/freedv-gui-1.9.6/src/rig_control/HamlibRigController.cpp: In static member function ‘static void HamlibRigController::InitializeHamlibLibrary()’:
/home/bdm/rpmbuild/BUILD/freedv-gui-1.9.6/src/rig_control/HamlibRigController.cpp:127:26: error: invalid conversion from ‘int ()(const rig_caps, void*)’ to ‘int ()(rig_caps, void*)’ [-fpermissive]
127 | rig_list_foreach(&HamlibRigController::BuildRigList_, &RigList_);
| ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| |
| int ()(const rig_caps, void*)
In file included from /home/bdm/rpmbuild/BUILD/freedv-gui-1.9.6/src/rig_control/HamlibRigController.h:36,
from /home/bdm/rpmbuild/BUILD/freedv-gui-1.9.6/src/rig_control/HamlibRigController.cpp:31:
/usr/include/hamlib/rig.h:3686:18: note: initializing argument 1 of ‘int rig_list_foreach(int ()(rig_caps, void*), void*)’
3686 | rig_list_foreach HAMLIB_PARAMS((int (*cfunc)(struct rig_caps *, rig_ptr_t),
| ^~~~~~~~~~~~~
gmake[2]: *** [src/rig_control/CMakeFiles/fdv_rig_control.dir/build.make:79: src/rig_control/CMakeFiles/fdv_rig_control.dir/HamlibRigController.cpp.o] Error 1
gmake[2]: Leaving directory '/home/bdm/rpmbuild/BUILD/freedv-gui-1.9.6/redhat-linux-build'
gmake[1]: *** [CMakeFiles/Makefile2:514: src/rig_control/CMakeFiles/fdv_rig_control.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
```

For reference, `BuildRigList_` is defined as follows:

```
static int BuildRigList_(const struct rig_caps *rig, rig_ptr_t);
```

Additionally, since both 4.5.5 and 4.6~git use `ABI_VERSION` and `ABI_REVISION` of 4 and 6, respectively, I can't conditionally compile in a defintiion of `BuildRigList_` without `const` for the latter. If this change is intentional, is there a way I can determine whether a user is running <= 4.5 vs. 4.6+? Thanks!

mdblack98, 2023-11-29:
> As reported in drowe67/freedv-gui#613 (comment), it appears that 4.6~git removed const from rig_list_foreach (specifically in commit 9841e50). Unfortunately, this breaks the current version of FreeDV:

If you just remove the "const" you may see a warning but hopefully still compile.I had to make this change to fix another bug.

tmiw, 2023-11-29:
Unfortunately removing `const` also causes FreeDV to stop compiling when using Hamlib 4.5.5:

```
/Users/mooneer/devel/freedv-gui/src/rig_control/HamlibRigController.cpp:127:9: error: no matching function for call to 'rig_list_foreach'
        rig_list_foreach(&HamlibRigController::BuildRigList_, &RigList_);
        ^~~~~~~~~~~~~~~~
/Users/mooneer/devel/freedv-gui/hamlib/include/hamlib/rig.h:3332:1: note: candidate function not viable: no known conversion from 'int (*)(struct rig_caps *, void *)' to 'int (*)(const struct rig_caps *, void *)' for 1st argument
rig_list_foreach HAMLIB_PARAMS((int (*cfunc)(const struct rig_caps *, rig_ptr_t),
^
1 error generated.
```

(This is with Clang 15 on macOS, but I suspect similar would happen on Linux.)

mdblack98, 2023-11-29:
Changing the ABI doesn't seem like the best solution....but in the future should be able to do that by adopting a slightly different ABI format.

Meanwhile I can add a #define for something like RIGCAPS_CONST so this should work.

#ifdef RIGCAPS_NOT_CONST
// use non-const definition
#else
// use const definition
#endif

Sound  OK?

There will be a couple weeks of Hamlib code that won't work but that's a small window.

Mike W9MDB

tmiw, 2023-11-29:
That will work. I created a bug in FreeDV for now to track the needed changes on my side once things are updated here. Thanks!

closed 2023-11-29 by mdblack98

mdblack98, 2023-12-14:
Please try the latest.