Hamlib/Hamlib #1445: rigstate hamlib_port_t deprecation

mdblack98, 2023-12-04:
Need to deprecate hamlib_port_t entries,  add pointer references to them, and change all internal references in all Hamlib code to the pointers.
This goal is to remove any dependency on rig_caps or rig_structure alignment by only accessing via function calls which will be internal to the shared libary instead of depending on rig.h
This will maintain backwards compatibility from 4.7 on and also allow older apps to update their access methods too for maximum compatibility if applicable.
WJST-X already did this change in Feb 2023.

GeoBaltz, 2023-12-19:
I'd like to get the ball rolling on this one, as it's going to take a long time....

What I'd like to do:

1. Define macros to access all the hamlib_port structures, replicating the current methods.  Make the macro calls generic enough so they cover both the current and final internal and external usages.
2. Port a few files to use the new macros with the current structures, to show proof of concept.
3. Start the long process of porting the entire code base.  Should be possible without breaking ABI compatibility with current S/W, and keeping the ongoing conversion transparent to anyone using current git or snapshots.
4. Make up external definitions for the macros to call functions within the library to retrieve pointers to the structures, as needed.  Write the functions, possibly with some sort of access control; should be flexible enough for other data access, if needed.
5. Release it.  Big warning messages - This is the last ABI compatible version, next version will require use of new methods access to port data (and anything else we throw in.)
6. Be ready to help 3rd parties port to new interface.
7. Cut over.  Redefine macros.  Rearrange structures. calloc() ports. See if anything else can be dropped. Clean up.

Timing:  I should have a PR of 1 & 2 ready for discussion in a day or so.  With the holidays and 4.6 coming up, I don't expect much for a while.  Then maybe 3-4 months for 3 & 4.

By a quick grep -R | wc -l, there are over 1500 references to the 3 port tables.  Gonna take a some time.

mdblack98, 2023-12-19:
Your idea has merit but we don't need to break the ABI.  

#1 The old structure stays around but gets deprecated.
#2 During open the new structure gets copied to the old one (just the sizeof for the deprecated entry -- that way old accesses work...new items aren't needed for old code.
#3 The macros to access make sense -- anything that updates it must update the deprecated one too if applicable.

GeoBaltz, 2023-12-20:
OK, I guess RAM is cheap enough...

The fun part of 2 is which deprecated items do you copy into the new structure.  I'm assuming anything updated by rig_set_conf() is copied to all 3, but what about items put directly into one of the deprecated structures(before or after rig_open)?

mdblack98, 2023-12-20:
You don't need to update anything in the deprecated structures.  The old version binaries will be updating those.The only problem we need to worry about is old software compiled against the old rig.h.  As long as the structures are still there they can utilize those instead of the new ones.
struct foo{    int a;
}struct foo2{    int a_deprecated;
    int a;}struct foo3{    int a_deprecated;
    int a_deprecated2;    int a;
}
Anything compiled against foo will always use a or a_deprecated (a_deprecated name is actually invisible but the offset is the same).Same works for foo3

GeoBaltz, 2023-12-20:
What I mean is, what happens if somebody sets a value into one of the deprecated structures but we're reading from the new one?  If a program compied against hamlib 4.5.5 is changing, say, rig->state->rigport.blah.blah, and the new hamlib is using the new pointer-based rigport->blah.blah, the value the program stored will never be used.  That's easier to track/copy at the time of rig_open(), but if the program does it at any time after that, we have no way of knowing which value is current/valid.

mdblack98, 2023-12-20:
Any binary compiled against the old header will always refer to the deprecated structure.The new binaries will refer to the new structure.  There's no conflict.  I've done that elsewhere.

GeoBaltz, 2023-12-20:
That's not the conflict I'm worried about.  It's the fact that actions of the program on the old structures will be ignored by hamlib.  Unless something copies that value from old, deprecated storage to the one that hamlib references, the change has no effect.  This problem happens today if a program sets something in the deprecated structure after hamlib populates the new one - nobody looks at the old one for newer data.

mdblack98, 2023-12-20:
You don't seem to understand how this works.....it's not like the structure is shared across programs....just the dll is.  Each program has it's own copy of the structure.

No need to update the old structure with the new code.

GeoBaltz, 2023-12-21:
I understand very well how it works - I've been working with shared code/unique data since the early/mid 1970s[1].  It's not the multiple programs that doesn't work - it is one program with different hamlibs.

Let's see if I can make up an example.  Suppose we have a program (compiled against 4.5.5 or 4.6) that is happily connected to a rig via hamlib (4.5.5 or 4.6).  The program decides to change mode to one where the timeouts and retries are taking too long.  So it sets rig->state.pttport.retry, and rig->state.pttport.timeout to new, smaller values, and gets faster notice of problems so it can adapt.

Now that same program (same binary) runs again with a new hamlib (4.7/5.0) that has the pointerized pttport.  The program sets the same values into the same (now deprecated) memory, but the hamlib in use isn't looking at those locations for its operating parameters.  So that same old program no longer functions as it did.

Granted, this is not the normal use case.  Most changes to pttport are essentially config changes, done before rig_open(), or are done via rig_set_conf(), and they can be copied into the new struct. So this may or may not be a problem in the real world.

And that's why I think that at some point in the (maybe far distant) future we should do a full reset of the ABI, and clear away the accumulated cruft of the last 20+ years. 

[1] Whenever Sperry-Univac implemented ER LINK$|ER EXLINK$ on Exec 8