Hamlib/Hamlib #1446: Python missing doc.

ve2tmq, 2023-12-06:
I'm trying to use cm108 with python.  Reading code are not helpfull. Do  you have doc about Hamlib class python or some sample code using cm108?

Thanks for your help.

echo-gravitas, 2024-12-01:
Same here. I just compiled Hamlib 4.5.5 for my macOS to use with Python. I also have successfully connected my IC-705 with `rigctl` directly.

Now I wonder how to work with the `Rig` class from Hamlib in Python. I'm quite new in Python but as a web developer I'm good in understanding docs. Is there any proper API documentation?

Vy 73 de Ralph HB3XCO

mdblack98, 2024-12-01:
You'll find some python examples in the "bindings" directory.
Other than that to get you started there is no other documentation.

Mike W9MDB

echo-gravitas, 2024-12-01:
Roger that. Thank you!

mdblack98, 2024-12-01:
If that's good enough can we close this issue?

echo-gravitas, 2024-12-01:
Yes, it's good enough for me. Already working with it.
Close the issue if you like.

closed 2024-12-01 by mdblack98

N0NB, 2024-12-01:
* On 2024 01 Dec 12:57 -0600, Ralph Bolliger wrote:
> Same here. I just compiled Hamlib 4.5.5 for my macOS to use with
> Python. I also have successfully connected my IC-705 with `rigctl`
> directly.
>
> Now I wonder how to work with the `Rig` class from [...]

As Mike suggests, there is a script, py3test.py, that may prove helpful
or not.  It is likely not a very Pythonic example just being a script
that steps through various calls.

There are likely better ways to create the bindings than Swig, but I
leave that to someone who knows what they're doing.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819