Hamlib/Hamlib #1449: Icom PCR-100 Shutdown

magellan-13016, 2023-12-10:
Hi,
I'm making a software with Python to control my Icom PCR-100 Radio. Actually, the program works very well and I use it to switch the radio on and let Gpredict control the frequency for Doppler correction.

To connect my rig I use : 
**subprocess.run("rigctld -v -m 4002 -r /dev/ttyUSB.ICOM-PCR100 -s 38400 -t 4532 -C stop_bits=1 &", shell=True, text=True)**

To change Volume I use : 
 **host = 'localhost'
port = 4532                   # The same port as used by the server
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((host, port))
s.send(b'L AF 0.3')
s.close()**

But, I didn't succeed to find what I have to do to switch off the radio. 
killing the Rigctld process does not work since my radio is still in receive mode after the commands below :
**rigctld_id=get_pid("rigctld")
os.kill(rigctld_id,9)**

When I want to connnect the radio again it is refused since it is always on.
The documentation said : 
 **0x87, set_powerstat 'Power Status'
              Set power On/Off/Standby 'Power Status'.
              0 = Power Off, 1 = Power On, 2 = Power Standby.  Defined as a bitmask in rig.h.**

So I've tried this code below but it did not work : 

**host = 'localhost'
port = 4532   
s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((host, port))
s.send(b'\x87\x02') 
s.close()**

Can someone help me please ?

magellan-13016, 2023-12-11:
Problem solved by myself with Chat GPT help. Running the command line **rigctld -v -m 4002 -r /dev/ttyUSB.ICOM-PCR100 -s 38400 -t 4532 -C stop_bits=1 &", shell=True, text=True)** in a terminal switch on the radio and typing Ctrl + C on the same terminal kill the rigctld process and shutdown the radio. But, killing the rigctld in python does not work because the process is killed but the radio stay powered on. I solved the problem by simulate Ctrl + C with this command : **os.system("kill -INT " + str(rigctld_process.pid))**

if this can help other people...

mcurtiscisco, 2024-11-30:
> Problem solved by myself with Chat GPT help.

How did ChatGTP help solve this?  
I am interested to learn new techniques. 
Thanks!

N0NB, 2025-04-27:
Original author marked this as "Solved".

closed 2025-04-27 by N0NB