Hamlib/Hamlib #1451: WSJT-X fails IC-9700 in satellite mode

mdblack98, 2023-12-13:
x25 command fails in sat mode followed by other failures
[2023-12-13 20:25:15.964017][00:00:02.015918][RIGCTRL:trace] write_block(): TX 7 bytes
[2023-12-13 20:25:15.964017][00:00:02.015931][RIGCTRL:trace] 0000    fe fe a2 e0 25 00 fd                                ....%..
[2023-12-13 20:25:15.968021][00:00:02.020097][RIGCTRL:trace] read_string_generic(): RX 6 characters, direct=1
[2023-12-13 20:25:15.968021][00:00:02.020107][RIGCTRL:trace] 0000    fe fe e0 a2 fa fd                                   ......
[2023-12-13 20:25:15.968021][00:00:02.020119][RIGCTRL:debug] 6:frame.c(380):icom_one_transaction returning(-9) Command rejected by the rig
[2023-12-13 20:25:15.968021][00:00:02.020140][RIGCTRL:debug] icom_transaction: failed: 0000    fe fe e0 a2 03 00 00 54 96 12 fd                    .......T...     
icom_one_transaction: frm_len=11, frm_len-1=fd, frm_len-2=12
      7:frame.c(431):icom_one_transaction returning(0) 
     6:frame.c(484):icom_transaction returning(0) 
icom_get_usb_echo_off: ack_len=6
icom_get_usb_echo_off: USB echo off detected
    5:icom.c(800):icom_get_usb_echo_off returning(1) 
icom_get_freq: using vfo=VFOA
    5:frame.c(453):icom_transaction entered
icom_transaction: cmd=0x25, subcmd=0x00, payload_len=0
     6:frame.c(143):icom_one_transaction entered
write_block(): TX 7 bytes
0000    fe fe a2 e0 25 00 fd                                ....%..         
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 6 characters, direct=1
0000    fe fe e0 a2 fa fd                                   ......          
     6:frame.c(380):icom_one_transaction returning(-9) Command rejected by the rig

Command rejected by the rig

.......

closed 2023-12-14 by mdblack98

florianrossmeier-svg, 2026-10-31:
**IC-9700 SATELLITE mode: issue #1451 still reproducible with Hamlib 4.7.2 and 4.7.3-rc**

73 from DG3RI.

I can confirm that the IC-9700 SATELLITE mode problem reported in issue #1451 is still reproducible on October 31, 2026.

**Test environment:**
- Icom IC-9700, CI-V address A2h
- Windows 11 x64, COM4, 19200 baud
- WSJT-X 3.0.0-beta1 / Hamlib 4.7.2 (SHA 40f6348)
- Hamlib 4.7.3-rc (SHA 78f5b78f2)
- IC-9700 backend 20250517.20
- SATELLITE mode ON, MAIN 433.540 MHz, SUB 144.040 MHz

**Confirmed failure in both Hamlib versions:**

Confirmed failure in both Hamlib versions:

TX: FE FE A2 E0 25 00 FD
RX: FE FE E0 A2 FA FD
rig_get_freq: -9 Command rejected by the rig

**Successful direct CI-V tests on the physical radio:**

[DG3RI-Hamlib-IC9700-Satellite-Report-20261031.zip](https://github.com/user-attachments/files/34216189/DG3RI-Hamlib-IC9700-Satellite-Report-20261031.zip)

16 5A correctly reports SATELLITE ON.

07 D0 selects MAIN; 03 reads 433.540 MHz.

07 D1 selects SUB; 03 reads 144.040 MHz.

05 is acknowledged on both sides when writing their existing frequencies.

FT8 reception and decoding work normally in SATELLITE mode when WSJT-X CAT control is disabled.

Hamlib appears to issue the unsupported `25 00` frequency query during initialization, before SATELLITE mode handling can avoid it.

Could the IC-9700 backend detect SATELLITE mode before its first frequency query and use the generic CI-V commands with appropriate MAIN/SUB selection?

I have full diagnostic traces from both versions and can test a patched Windows x64 build on the physical IC-9700. The transmitter was kept inhibited during all diagnostic tests.

73,
DG3RI

florianrossmeier-svg, 2026-10-31:
I have reproduced the original IC-9700 SATELLITE-mode CAT failure from #1451 with both the Hamlib 4.7.2 library bundled with WSJT-X and the official 4.7.3~rc snapshot (`78f5b78f2`, 2026-10-28). I also developed and tested a small source patch against that 4.7.3~rc snapshot on a physical IC-9700.

**Root cause:** When the radio is already in SATELLITE mode, Hamlib's initial `rig_get_powerstat()` calls `rig_get_freq()` before `icom_rig_open()` has detected SATELLITE mode. Consequently, the IC-9700 is sent a VFO-targeted CI-V `0x25 0x00` frequency query, which it rejects with `FA` / Hamlib error `-9`.

**Patch:** For the IC-9700 only, query SATELLITE state (`0x16 0x5A`) once before the first frequency read, so the existing `icom_satmode_fix()` logic disables unsupported VFO-targeted frequency commands. The check is reset on rig close. Files changed: `rigs/icom/icom.c` and `rigs/icom/icom.h`.

**Verified on real hardware (Windows 11 x64; CI-V COM4, 19200 baud, rig model 3081):**

- The unmodified 4.7.3~rc build reproduces `25 00 -> FA` / `-9`; the patched build reports `SATELLITE mode=1` and reads frequency using CI-V `0x03` instead.
- MAIN 433.540000 MHz and SUB 144.040000 MHz: both read correctly after selecting `07 D0` / `07 D1`.
- Both VFOs accept `0x05` set-frequency with their **unchanged** existing values and return ACK `FB`; readback matches.
- A separate MSYS2 MINGW64 (MSVCRT) build of the patched `libhamlib-4.dll` loads in **WSJT-X 3.0.0-beta1**. `Test CAT` succeeds; live frequency reporting and FT8 receive/decoding work.
- WSJT-X CAT PTT correctly selects the IC-9700's 144 MHz SUB transmit path. In WSJT-X Tune (USB-D audio, 50-ohm dummy load) the radio generates approximately 0.5 W forward RF power with VSWR around 1.00:1 and negligible ALC indication.

**Scope/limitations:** No over-the-air QSO or full FT8 transmit slot has been tested. The test above confirms PTT and USB-D Tune, not complete FT8 transmission. When the MAIN frequency is changed through WSJT-X CAT, the SUB frequency remains unchanged on this setup; physical tuning of MAIN does couple SUB. Automatic CAT-driven MAIN/SUB tracking is **not** implemented in this patch and should be treated as a separate enhancement. Non-IC-9700 radios and normal-mode regression coverage have not yet been tested.

I have attached the minimal `.patch` plus short before/after evidence and test notes. Would a maintainer be willing to review this regression fix, even though #1451 is closed? I can open a focused PR against the current development branch and perform additional tests on the actual IC-9700 if requested.

73, DG3RI

[DG3RI-Hamlib-IC9700-SAT-Fix-V1-GitHub.zip](https://github.com/user-attachments/files/34222599/DG3RI-Hamlib-IC9700-SAT-Fix-V1-GitHub.zip)