Hamlib/Hamlib #1454: ICOM IC-705 rigctl

marcinoza, 2023-12-19:
Hay,

My rig ICOM IC-705

If I`m use 

marcinoza@marcinoza:~$ rigctl -m 3073 -r /dev/ttyACM0 -s 115200 -vv -c 0xA4
rig_get_cache(302): unknown vfo=otherVFO
rig_set_cache_freq(208): unknown vfo?, vfo=otherVFO
rig_set_cache_freq(208): unknown vfo?, vfo=otherVFO
Opened rig model 3073, 'IC-7300'

Rig command: f
Frequency: 436392822

Rig command: q

Its oki working

more verbose:

> rig_check_rig_caps: p1=0x7f650fac1260, p2=0x7f650fac77e0, rig_model=0x7f650fac1260, macro_name=0x7f650fac77e0
> initrigs4_icom: _init called
> rig_init: rig_model=Icom IC-7300
> rig_init: rig has VFO_A
> rig_init: rig has VFO_B
> rig_init: rig has VFO_MEM
>  1:icom.c(623):icom_init entered
> icom_init: done
>  1:icom.c(724):icom_init returning(0) 
> rig_token_lookup called for civaddr
> rig_confparam_lookup called for civaddr
> rig_set_conf called
> rig_confparam_lookup called for 1
> rig_set_conf: civaddr='0xA4'
>  1:icom.c(4958):icom_set_conf entered
>  1:icom.c(4988):icom_set_conf returning(0) 
>  1:rig.c(817):rig_open entered
> rig_settings_get_path: path=/home/marcinoza/.config/hamlib_settings
> rig_settings_load_all: settings_file (/home/marcinoza/.config/hamlib_settings): No such file or directory
> rig_open: cwd=/home/marcinoza
> rig_open: /home/marcinoza/hamlib_settings does not exist
> rig_open: async_data_enable=1, async_data_supported=1
> create_sync_data_pipe: created data pipe for synchronous transactions
> serial_open: **/dev/ttyACM0**
> serial_setup: tcgetattr
> serial_setup: cfmakeraw
> serial_setup: cfsetispeed=115200,0x1002
> serial_setup: cfsetospeed=115200,0x1002
> serial_setup: data_bits=8
> serial_setup: parity=0
> serial_setup: Handshake=None
> serial_setup: tcsetattr TCSANOW

serial device its right, " serial_open: **/dev/ttyACM0** "

But we use:

> marcinoza@marcinoza:~$ rigctld -m 3085 ‐r /dev/ttyACM0 ‐s 115200 -vvvvvv
> ../../tests/rigctld.c(664) Startup: rigctld -m 3085 -vvvvvv ‐r /dev/ttyACM0 ‐s 115200
> rigctld Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
> Max# of rigctld client services=32
> rig_check_rig_caps: p1=0x7f33c38c1260, p2=0x7f33c38c77e0, rig_model=0x7f33c38c1260, macro_name=0x7f33c38c77e0
> initrigs4_icom: _init called
> rig_init: rig_model=Icom IC-705
> rig_init: rig has VFO_A
> rig_init: rig has VFO_B
> rig_init: rig has VFO_MEM
>  1:icom.c(623):icom_init entered
> icom_init: done
>  1:icom.c(724):icom_init returning(0) 
> main: twiddle=0, uplink=0, twiddle_rit=0
>  1:rig.c(817):rig_open entered
> rig_settings_get_path: path=/home/marcinoza/.config/hamlib_settings
> rig_settings_load_all: settings_file (/home/marcinoza/.config/hamlib_settings): No such file or directory
> rig_open: cwd=/home/marcinoza
> rig_open: /home/marcinoza/hamlib_settings does not exist
> rig_open: async_data_enable=1, async_data_supported=1
> create_sync_data_pipe: created data pipe for synchronous transactions
> serial_open: **/dev/ttyS0**
> serial_setup: tcgetattr
> serial_setup: cfmakeraw
> serial_setup: cfsetispeed=19200,0x000e
> serial_setup: cfsetospeed=19200,0x000e
> serial_setup: data_bits=8
> serial_setup: parity=0
> serial_setup: Handshake=None
> serial_setup: tcsetattr TCSANOW
> 

What we see - serial_open: **/dev/ttyS0**

Rig not working.

I'm thinking its bug ? , maybe other option ?

mdblack98, 2023-12-20:
Please upgrade to the latest hamlib.

https://n0nb.users.sourceforge.net/

Mike W9MDB

> rig_check_rig_caps: p1=0x7f650fac1260, p2=0x7f650fac77e0, rig_model=0x7f650fac1260, macro_name=0x7f650fac77e0
> initrigs4_icom: _init called
> rig_init: rig_model=Icom IC-7300
> rig_init: rig has VFO_A [...]

serial device its right, " serial_open: /dev/ttyACM0 "

But we use:

> ../../tests/rigctld.c(664) Startup: rigctld -m 3085 -vvvvvv ‐r /dev/ttyACM0 ‐s 115200
> rigctld Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
> Max# of rigctld client services=32
> rig_check_rig_caps: [...]

What we see - serial_open: /dev/ttyS0

Rig not working.

I'm thinking its bug ? , maybe other option ?

marcinoza, 2023-12-20:
Upgarde to latest version 4.5.5-3 , I don't feel up to compiling versions 4.6, I'll wait for the .deb packages

https://launchpad.net/ubuntu/+source/hamlib

i  libhamlib-dev:amd64                                         4.5.5-3                                     amd64        Development library to control radio transceivers and receivers
ii  libhamlib-utils                                             4.5.5-3                                     amd64        Utilities to support the hamlib radio control library
ii  libhamlib4:amd64                                            4.5.5-3                                     amd64        Run-time library to control radio 

seems like oki, rig work.

root@marcinoza:~# rigctld -m 3085 -r /dev/ttyACM0 -s 115200 -vvvvv -t 4533
../../tests/rigctld.c(664) Startup: rigctld -m 3085 -r /dev/ttyACM0 -s 115200 -vvvvv -t 4533
rigctld Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
Report bugs to <hamlib-developer@lists.sourceforge.net>

Max# of rigctld client services=32
rig_check_rig_caps: p1=0x7efc4ecc1260, p2=0x7efc4ecc77e0, rig_model=0x7efc4ecc1260, macro_name=0x7efc4ecc77e0
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-705
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
 1:icom.c(623):icom_init entered
icom_init: done
 1:icom.c(724):icom_init returning(0) 
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(817):rig_open entered
rig_settings_get_path: path=/root/.config/hamlib_settings
rig_settings_load_all: settings_file (/root/.config/hamlib_settings): No such file or directory
rig_open: cwd=/root
rig_open: /root/hamlib_settings does not exist
rig_open: async_data_enable=1, async_data_supported=1
create_sync_data_pipe: created data pipe for synchronous transactions
serial_open: /dev/ttyACM0
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=115200,0x1002
serial_setup: cfsetospeed=115200,0x1002
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=0, expected_len=1
tcflush

any way thanks. 

i

closed 2023-12-20 by mdblack98