Hamlib/Hamlib #1456: IC-9700 set/get Sub VFO data

phl0, 2023-12-20:
Just compiled the latest git version of Hamlib an cannot get rigctl to read/write the frequency/mode of the sub VFO. It instead return the frequency of the main VFO. Setting the VFO to SubA switches the VFO but the returned frequency is wrong:

```
~/Hamlib$ ./tests/rigctl -m 3081 -r /dev/ic-9700a -s 57600 

Rig command: V SubA

Rig command: f
Frequency: 144040010

Rig command: V MainA

Rig command: f
Frequency: 144040010
```

[image]

phl0, 2023-12-20:
Seems related to https://github.com/Hamlib/Hamlib/issues/730. But issue still exists?!

phl0, 2023-12-20:
Did some research and it seems there are various options to control the sub band VFO. You could use `G XCHG` to exchange main and sub then change or read the frequency. But there are also commands that can do that without changing the VFOs. If needed I can provide some demo script here that sets the sub VFO without exchanging them first.

mdblack98, 2023-12-20:
The 0x25 and 0x26 command only work on Main.

So if you have a sequence that can set SubA and SubB without swapping to either I'm all ears.
Same goes for setting MainA MainB when sitting on the Sub VFO.

Mike W9MDB

phl0, 2023-12-20:
I made myself a small script that sets the sub band VFO to a specific frequency and mode. See here: https://www.florian-wolters.de/posts/ic9700-hamlib-setup/. Let me know if I can help further.

mdblack98, 2023-12-21:
Just to let you know it's more complicated than that although your sequence is correct for what it does.

I have to allow for the starting point to be any of Main/Sub VFOs and come back to that same VFO when done (when in --vfo mode).

I should have this done tomorrow.

Mike W9MDB

mdblack98, 2024-06-14:
This should be working now in the latest Hamlib both in non-vfo and vfo mode
Rig command: V MainA
Rig command: f
Frequency: 146980500
Rig command: V SubA
Rig command: f
Frequency: 1294000000
Rig command: V MainB
Rig command: f
Frequency: 144561000
Rig command: V SubB
Rig command: f
Frequency: 1293895500

--vfo mode
Rig command: f MainA
Frequency: 146980500
Rig command: f MainB
Frequency: 144561000
Rig command: f SubA
Frequency: 1294000000
Rig command: f SubB
Frequency: 1293895500

closed 2024-08-18 by mdblack98