Hamlib/Hamlib #1463: is_networked undefined for Apple

markjfine, 2023-12-26:
Error in src/network.c

network.c:1265:10: error: call to undeclared function 'is_networked'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    if (!is_networked(ip4, sizeof(ip4)))
         ^

Noticed that is_networked() is defined for ifdef __MINGW32__ or ifndef __APPLE__.

It's possible that there's an extra (or misplaced) #endif on line 1247 but I got lost in all the different defines. Moving one or both to after multicast_receiver() (where the undefined is_networked() resides)  creates other issues.

GaryKO3F, 2023-12-26:
I got the same error this morning:

CC       event.lo
  CC       cal.lo
  CC       conf.lo
  CC       tones.lo
  CC       rotator.lo
  CC       locator.lo
  CC       rot_reg.lo
  CC       rot_conf.lo
  CC       rot_settings.lo
  CC       rot_ext.lo
  CC       iofunc.lo
  CC       ext.lo
  CC       mem.lo
  CC       settings.lo
  CC       parallel.lo
  CC       usb_port.lo
  CC       debug.lo
  CC       network.lo
network.c:1265:10: error: call to undeclared function 'is_networked'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
    if (!is_networked(ip4, sizeof(ip4)))
         ^
1 error generated.
make[2]: *** [network.lo] Error 1
make[1]: *** [all] Error 2
make: *** [all-recursive] Error 1
MacBook-Pro:Hamlib-master charlesrogers$

markjfine, 2023-12-26:
Works now... cheers.

closed 2023-12-26 by mdblack98

GaryKO3F, 2025-01-08:
For the last couple of days, I have been getting the error below after using the make command: 

 CC       fifo.lo
  CCLD     libhamlib.la
libtool:   error: only absolute run-paths are allowed
make[2]: *** [libhamlib.la] Error 1
make[1]: *** [all] Error 2
make: *** [all-recursive] Error 1

A few days ago, it compiled without error. This is on a Macbook Pro Apple chip running MacOS 15.2

Any idea how to resolve? Thanks

markjfine, 2025-01-08:
Might want to start a new issue instead of tacking on to this one (1463). This one was closed over a year ago and I'm fairly certain what you've raised is a different problem.

mdblack98, 2025-01-08:
Yes...please open a new issue...meanwhile what does your "configure" line look like?

markjfine, 2025-01-08:
That said, I just successfully rebuilt here on a similar system.

@GaryKO3F: I make it a habit to first run ./bootstrap and then ./configure after pulling from the repository. Recent changes to the repository included updates to .am and .in files that may not have been correctly configured prior to running make. You may find this fixes your build error.