Hamlib/Hamlib #1468: morse on cat error

leonardinux, 2023-12-31:
**Rig command: b v**
send_morse: error =     4:rig.c(2586):rig_get_mode returning(0) 
   3:rig.c(1403):rig_open returning(0) 
main: rig_open retcode=0
rigctl_parse: called, interactive=1
rigctl_parse: cmd=b(62) handle=0
rigctl_parse: vfo_opt=0
   3:rigctl_parse.c(4634):rigctl_send_morse entered
    4:rig.c(6601):rig_send_morse entered
netrigctl_send_morse called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 9 characters, direct=1
    4:rig.c(6624):rig_send_morse returning(-14) Communication bus collision

   3:rigctl_parse.c(4636):rigctl_send_morse returning(-14) Communication bus collision

Communication bus collision

Talking with rigctld -m 1035 -r /dev/ttyUSB2 -s 38400
FT991A via /dev/ttyUSB2 3800 baud - other commands works fine, CW via CAT is not working.
Have I missed some config option?

73 de Leo IZ5FSA

mdblack98, 2023-12-31:
What version of Hamlib?

leonardinux, 2023-12-31:
$ rigctl -V
rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3

$ uname -a 
Linux debian 6.5.0-5-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.5.13-1 (2023-11-29) x86_64 GNU/Linux

mdblack98, 2023-12-31:
Please update to the latest.  I believe things work a lot better now.

https://n0nb.users.sourceforge.net/

You'll need to uninstall your old Hamlib.

leonardinux, 2023-12-31:
Uninstalling hamlib is not possible without removing all dependant package like CQRLOG and others... if I make a /usr/local version can be stay together 4.5 official deb package version????

mdblack98, 2023-12-31:
You can replace the installed system.

./configure --prefix=/usr

It will overwrite the installed system.  Normally not recommended but should work.

Could also do the /usr/local/bin install and just rename /usr/bin/rigctl and /usr/bin/rigctld and any libhamlib.so you have already.

Mike W9MDB

N0NB, 2023-12-31:
* On 2023 31 Dec 10:15 -0600, Michael Black wrote:
> You can replace the installed system.
> 
> ./configure --prefix=/usr
> 
> It will overwrite the installed system.  Normally not recommended but should work.

Before doing any of that, because it will break the distribution's
packaging, do a default build  that will install under the /usr/local
hierarchy.  If that has never been done before then `ldconfig` will need
to be run as root.  Once that is complete, restarting dependent packages
should pick up the Hamlib installed under /usr/local.

Confirmation can be made with `ldd` with some dependent packages.  Those
that simply run `rigctld` will not show a library dependency on
`libhamlib`, however.

I do that here and have done so for years:

$ ldd $(which fldigi) | grep hamlib
	libhamlib.so.4 => /usr/local/lib/libhamlib.so.4 (0x00007ff19f400000)

Packages that run `rigctld` directly such as cqrlog may need to be
configured to call `/usr/local/bin/rigctld` or ensure that
`/usr/local/bin` is in the PATH variable before `/usr/bin`.

I do have the Debian libhamlib4 package installed, but it is unused and
is only there to satisfy apt dependencies.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

leonardinux, 2024-01-01:
Ok Nate.

I've extracted source code in /usr/local/src and installed as default ./configure with no options. As I run rigctld as a service I don't need to work around all other ham apps (qlog, cqrlog, wsjtx, fldigi, vara, varafm) and all works fine... and CW VIA CAT now works !!!

Thanks for tips.

Please release hamlib-utils-4.6.0 ASAP for "hamradio enthusiast people"!!!

73 HNY de Leo IZ5FSA (aka leonardinux)

leonardinux, 2024-01-01:
For me, thread is closed.

closed 2024-01-01 by mdblack98