Hamlib/Hamlib #1469: Change token_t to hamlib_token_t to avoid collision with libbsm

mdblack98, 2024-01-03:
/*

 * Contents of token_t are opaque outside of libbsm.

 */

typedef struct au_token token_t;

It is part of libbsm, a security library originally from FreeBSD. The problem on macOS is that if hamlib is used in a Xcode project, pretty much all system header files are included automatically though Cocoa.h or Framework.h.

It is present in macOS Sonoma and has been for at least the last 8-9 years, or longer.

Best Regards,
Chris

danupp, 2024-01-22:
When building according to my Docker receipt for WSJT-X, https://gist.github.com/danupp/57d8eb0f60837e782617bacd1b4a069e , I now get:

wsjtx/wsjt-wsjtx/Transceiver/HamlibTransceiver.cpp: In member function 'void HamlibTransceiver::impl::set_conf(const char*, const char*)':
/wsjtx/wsjt-wsjtx/Transceiver/HamlibTransceiver.cpp:310:3: error: 'token_t' was not declared in this scope; did you mean 'tone_t'?
  310 |   token_t token = rig_token_lookup (rig_.data (), item);
      |   ^~~~~~~
      |   tone_t

Seems this might be related to this issue?

Tyrbiter, 2024-01-22:
Has been fixed by this:

https://github.com/Hamlib/Hamlib/commit/75cf467c1d2d314df747247efee28af5130d0b03

mdblack98, 2024-01-22:
> Change token_t to hamlib_token_t.
> I need to see if I can avoid this problem.
> Mike W9MDB

Should be fixed now.

closed 2024-01-30 by mdblack98