Hamlib/Hamlib #1472: TS590 set_mode does not get bandwidth correctly and does not set at all

mdblack98, 2024-01-09:
This changes mode but not bandwidth.
M USB 3000 

And this always returns 1000 for USB and 500 for CW
m

rz4ag, 2024-01-10:
Just tested
Strange answer for LSB mode:

Rig command: m
Mode: CW
Passband: 400

Rig command: M CW 250

Rig command: m
Mode: CW
Passband: 250

Rig command: M LSB 1800

Rig command: m
Mode: LSB
Passband: 1750

Rig command: M LSB 2000

Rig command: m
Mode: LSB
Passband: 1950

Rig command: M LSB 1500

Rig command: m
Mode: LSB
Passband: 1550

Rig command: M LSB 2200

Rig command: m
Mode: LSB
Passband: 2150

mdblack98, 2024-01-10:
On the read it is showing the total bandwidth low to high.

So you must have a 50Hz low setting.

The 1500 actually gets set to 1600 and subtracts the 50 offset for 1550.

There is no 1500 setting in the rig so it picks the next highest.

I had to make a decision to show the high setting or high minus low and I thought high-low was more informative,.

Mike W9MDB

mdblack98, 2024-01-28:
This is as close as we're going to get until a new API is created.
See https://github.com/Hamlib/Hamlib/issues/1491

closed 2024-01-28 by mdblack98