Hamlib/Hamlib #1473: Python bindings throws segmentation fault when using osx-arm64

giordyb, 2024-01-09:
I am running a osx-arm64 Anaconda python environment for which I am trying to use Hamlib with.

I created the environment with the following command:
`CONDA_SUBDIR=osx-arm64 conda create -n doppler_shifter_arm_11 python=3.12`

I then activated the environment with: `conda activate doppler_shifter_arm_11`
 
I cloned and compiled Hamlib from source using the following commands:

- `./bootstrap`
- `./configure --with-python-binding PYTHON=$(which python3)`
- `make -j4 && sudo make install`

the compilation process completes successfully and the python bindings are installed in "`/usr/local/lib/python3.12/site-packages`"

when I try to import Hamlib with the following code a segfault is thrown

```
import sys
import faulthandler

faulthandler.enable()
sys.path.append("/usr/local/lib/python3.12/site-packages")
import Hamlib

```

output is the following

```
Fatal Python error: Segmentation fault

Current thread 0x00000001e3ee9000 (most recent call first):
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1288 in create_module
  File "<frozen importlib._bootstrap>", line 813 in module_from_spec
  File "<frozen importlib._bootstrap>", line 915 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1325 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1354 in _find_and_load
  File "/usr/local/lib/python3.12/site-packages/Hamlib.py", line 15 in <module>
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 994 in exec_module
  File "<frozen importlib._bootstrap>", line 929 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1325 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1354 in _find_and_load
  File "/Users/giordy/Documents/python projects/doppler_shifter/test_hamlib.py", line 6 in <module>
[1]    57987 segmentation fault  python3 test_hamlib.py
```

on the other hand if I try the above steps using an x86_64 python environment created using `CONDA_SUBDIR=osx-64 conda create -n doppler_shifter_arm_64 python=3.12` I get 

```
Traceback (most recent call last):
  File "test_hamlib.py", line 6, in <module>
    import Hamlib
  File "/usr/local/lib/python3.12/site-packages/Hamlib.py", line 15, in <module>
    import _Hamlib
ImportError: dlopen(/usr/local/lib/python3.12/site-packages/_Hamlib.so, 0x0002): tried: '/usr/local/lib/python3.12/site-packages/_Hamlib.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64')), '/System/Volumes/Preboot/Cryptexes/OS/usr/local/lib/python3.12/site-packages/_Hamlib.so' (no such file), '/usr/local/lib/python3.12/site-packages/_Hamlib.so' (mach-o file, but is an incompatible architecture (have 'arm64', need 'x86_64'))
```

I had to then re-configure and re-compile Hamlib using
`arch -x86_64 ./configure --with-python-binding PYTHON=$(which python3)`
`arch -x86_64 make -j4 && sudo make install`

confirmed that the python version is x86_64 using 

```
file -b $(which python3)
Mach-O 64-bit executable x86_64
```

but still received a segfault
```
Fatal Python error: Segmentation fault

Current thread 0x0000000201412240 (most recent call first):
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 1288 in create_module
  File "<frozen importlib._bootstrap>", line 813 in module_from_spec
  File "<frozen importlib._bootstrap>", line 915 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1325 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1354 in _find_and_load
  File "/usr/local/lib/python3.12/site-packages/Hamlib.py", line 15 in <module>
  File "<frozen importlib._bootstrap>", line 488 in _call_with_frames_removed
  File "<frozen importlib._bootstrap_external>", line 994 in exec_module
  File "<frozen importlib._bootstrap>", line 929 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 1325 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 1354 in _find_and_load
  File "/Users/giordy/Documents/python projects/doppler_shifter/test_hamlib.py", line 6 in <module>
[1]    51554 segmentation fault  python test_hamlib.py
```

mdblack98, 2024-01-09:
Try adding CFLAGS="-march=native" to the hamlib configure
Another possiblity is "-march=x86_64"

Are you using gcc or clang?

I don't think this is a Hamlib problem but is a binary incompatibility somewhere.  Works fine here on Ubuntu x86_64 and Python 3.10.12

Mike W9MDB

giordyb, 2024-01-10:
Hi @mdblack98 thanks for the suggestion. The flags that you refer to do the same thing on MacOS as prepending the `arch -x86_64` to the commands.

I am able to compile Hamlib both with x86_64 **and** arm64 and successfully run the utilities (rigctl/rotctl) without any issues. 
I am also able to compile and run gpredict (which depends on Hamlib) successfully.

This issue seems to be affecting **only** when loading the python bindings, which is a bit puzzling...

mdblack98, 2024-01-10:
Doesn't surprise me that Hamlib utilities work.  They are all compiled at the same time.

Have you tried compiling python yourself instead of using a package?  Could be some way they compiled it.

What version of python?  Also possible they broke something.

Mike W9MDB

giordyb, 2024-01-30:
Just wanted to provide an update before closing this issue:

Tried the above steps on a brand new install of macOS Ventura on an old Intel MacBook Pro with Anaconda and even though it builds, it still throws a segmentation fault when loading the bindings.
It looks like there is some conflict with the Anaconda python distro. Not sure what changed since it used to work fine previously.

I was able to compile and load the module using both the Python install that comes with the OS and a later version downloaded from https://www.python.org/downloads/macos/ but with a caveat: with the default settings the configure script was not able to compile the binding because the PYTHON_EXTRA_LDFLAGS would be detected as `Python.framework/Versions/3.11/Python`, which is the output of `sysconfig.get_config_var('LINKFORSHARED')`
which returns a path that does not exist. 

The correct path should be `/Library/Frameworks/Python.framework/Versions/3.11/Python`

After changing the path it compiled and I was able to load the binding.

Hopefully this helps anyone else having the same issue.

@mdblack98 to work around this I modified the "configure" script and hardcoded the PYTHON_EXTRA_LDFLAGS. Is there a way to pass this as a parameter of the configure script?

closed 2024-01-30 by giordyb

mdblack98, 2024-01-30:
See if this gets the correct path.

python3-config --ldflags

Although I understand sysconfig is the preferred way to go.

You're not alone with this problem...

https://bugs.python.org/issue3588

You should be able to define PYTHON_EXTRA_LDFLAGS yourself during configure.

./configure -D PYTHON_EXTRA_LDFLAGS="xxxxxx"

Mike W9MDB