Hamlib/Hamlib #1474: rig_band_changed customizaton

mdblack98, 2024-01-10:
Add ability to perform actions using rigctld commands.
This may require building rigctld service into Hamlib in a thread or a customized call in the rigctl_parse.c routine.

GeoBaltz, 2024-01-11:
Whoa.  Is this really in scope for Hamlib?  Seems much more like something that should be in the application layer, calling on the API that hamlib already has.

Looks to me like classic Mission Creep, aka feeping creaturism.

mdblack98, 2024-01-11:
Anything that controls a rig/amp/rotator is in scope.
And yes feature creep is always a real thing.  People trying to do more.
In this case it was requested for tuner control where tuner is available on some bands but not others.
So having automatic control of it for any application using Hamlib is appropriate.
Eventually want to expand this to allow all sorts of things customizing your shack setup.
Expecting the dozens of apps to implement this kind of thing is never going to happen.