Hamlib/Hamlib #1475: gqrx WSJT-X compatibility

CocolinoFan, 2024-01-11:
Hello,

When trying to connect WSJT-X to gqrx's Remote Control via TCP, I get an odd error:
```
Rig failure
Hamlib error:   2:rig.c(7481):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
  2:rig.c(7488):async_data_handler_start returning(0) 
rig.c(254):add_opened_rig returning2(0) 
rig_open: 0x1bf4056c rs->comm_state==1?=1
  2:rig.c(6084):rig_get_powerstat entered
rig.c(6104) trace
netrigctl_get_powerstat called
netrigctl_transaction: called len=15
rig_flush: called for network device
network_flush called
write_block(): TX 15 bytes, method=2
0000    5c 67 65 74 5f 70 6f 77 65 72 73 74 61 74 0a        \get_powerstat. 
read_string_generic called, rxmax=1024 direct=1, expected_len=1
read_string_generic(): RX 7 characters, direct=1
0000    52 50 52 54 20 31 0a                                RPRT 1.         
  2:rig.c(6109):rig_get_powerstat returning(0) 
rig_open: rig power is off, use --set-conf=auto_power_on if power on is wanted
Rig is not powered on
Rig is not powered on
 while opening connection to rig
```

Steps to reproduce:

1. Open gqrx
2. Enable "Remote Control via TCP"
3. Open WSJT-X 
4. File -> Settings -> Radio
5. Select Rig: Hamlib NET rigctl
6. Network Server: 127.0.0.1:7356
7. Click "Test CAT"

CocolinoFan, 2024-01-11:
Gqrx 2.17.2
WSJT-X v2.6.1

mdblack98, 2024-01-11:
That should be fixed in the latest hamlib.  It is an example of somebody trying to emulate rigctld and not doing a good job of it.
https://n0nb.users.sourceforge.net/

CocolinoFan, 2024-01-11:
I see.
I am using my distribution's package manager for installation. So I should probably wait for an update.
Thank you sir. If this is known, should I close the issue?

mdblack98, 2024-01-11:
It has been fixed.  I'm trying to get 4.6 out the door by the end of the month -- have no idea when the package people will update though.

You can uninstall the package and build this to replace it.  Or I can send you a Linux library you could test perhaps to replace the packaged one.

If you do "ldd wsjxt | grep hamlib" you should see the libhamlib reference that would need replacing.

Let me know and I can send you a link to replace it.

CocolinoFan, 2024-01-11:
Yes please, I could try to replace the library.
```
ldd wsjtx | grep hamlib
libhamlib.so.4 => /usr/lib64/hamlib/libhamlib.so.4 (0x00007f984fe00000)
```

mdblack98, 2024-01-13:
Try this one

https://www.dropbox.com/scl/fi/m8020l1ybghwehklj2fe1/libhamlib.so.4.0.6?rlkey=z41my0dkhm0k4kq2b2kqob3kp&dl=0

Mike W9MDB

CocolinoFan, 2024-01-13:
Amazing! Worked like a charm. 
Thank you sir.

CocolinoFan, 2024-01-13:
I'm not sure if this is relevant at all, but I was looking a `dmesg` and found some interesting outputs:
```
[175711.207631] traps: QThread[19743] general protection fault ip:7f9b1f8e08f2 sp:7f9af69ffd40 error:0 in libhamlib.so.4.0.6[7f9b1f886000+22b000]
[175793.579596] QThread[20946]: segfault at 51 ip 00007f6560982199 sp 00007f65395faea8 error 4 in libc.so.6[7f6560864000+13c000] likely on CPU 2 (core 2, socket 0)
[175793.579609] Code: 77 c3 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 44 00 00 89 f8 48 89 fa c5 f9 ef c0 25 ff 0f 00 00 3d e0 0f 00 00 0f 87 37 01 00 00 <c5> fd 74 0f
c5 fd d7 c1 85 c0 74 5b f3 0f bc c0 c5 f8 77 c3 0f 1f
```

mdblack98, 2024-01-13:
Is that repeatable?

CocolinoFan, 2024-01-13:
I have `dmenu -w` running all the time. It happen at the beginning of the day when messing about with WSJT-X's settings. Since then even tho I had WSJT-X running all the time, messing about with it and clicking "Ok" on errors; no other messages showed in dmenu.
This might be my memory playing tricks on me but I think it happen when (in File->Settings->Radio) I clicked "Test CAT" and then "Test PTT" quickly without waiting for the tests to finish. 
I tried but I am unable to reproduce it. Now, even tho I spam "Test CAT" and "Test PTT" really quickly I don't even get a error in WSJT-X.

CocolinoFan, 2024-01-18:
Another one, only once, unable to reproduce:
```
[20445.530724] QThread[23932]: segfault at aa00000404 ip 00007f75928e08f2 sp 00007f756b3ffd40 error 4 in libhamlib.so.4.0.6[7f7592886000+22b000] likely on CPU 5 (core 1, socket 0)
[20445.530738] Code: ff 5b 5d 41 5c 41 5d 41 5e 41 5f c3 66 66 2e 0f 1f 84 00 00 00 00 00 90 f3 0f 1e fa 48 85 ff 74 6e 55 53 48 89 fb 48 83 ec 08 <8b> 87 04 04 00 00 85 c0 78 54 8b 8f 00 04 00 00 3d
ff 03 00 00 0f
```
I was in FT8 and navigated to File -> Settings

mdblack98, 2024-01-18:
Can you please send the output from this:
nm /usr/local/lib/libhamlib.so.4.0.6 | grep 00022b

CocolinoFan, 2024-01-18:
I use Gentoo OS, I can only find `libhamlib` in `/usr/lib64/hamlib`
```
/usr/lib64/hamlib $ nm libhamlib.so.4.0.6 | grep 00022b
000000000022b210 t ft920_cleanup
000000000022b130 t ft920_get_split_mode
000000000022b2a0 t ft920_init
000000000022b550 t ft920_send_dynamic_cmd.constprop.0
000000000022b340 t ft920_send_rit_freq.constprop.0
000000000022b740 t ft920_set_mode
000000000022bee0 t ft920_set_rit
000000000022bd00 t ft920_set_split_mode
```

mdblack98, 2024-01-27:
We need to chase down this problem.  Can you please contact me offline at mdblack98@yahoo.com?

closed 2024-01-30 by mdblack98