Hamlib/Hamlib #1478: 2038 check

mdblack98, 2024-01-19:
Add _TIMESIZE==32 check to compile and generate warning of 2038 compatibility.

mdblack98, 2024-01-24:
mingw64 failed TIMESIZE check but passes 2038 test.  
So we've put in a test during "make check" instead.
We don't force anybody to do a "make check" so might want to implement the 2038 test during a normal build too.

mdblack98, 2024-01-27:
Need to review all time_t usages.  Perhaps put a a 2038Reviewed comment as reviewed.

N0NB, 2024-02-03:
At the bottom of this page: https://wiki.debian.org/ReleaseGoals/64bit-time is a section titled "ABI Transition".  At the bottom is a bullet list and `libhamlib++-dev` and `libhamlib-dev` are in the list of 329 packages showing ABI change due to compiling with a 64 bit time_t.

Debian is beginning to rebuild its library packages with 64 time_t.  I'm not sure if they'll upstream any patches to us or not.

mdblack98, 2024-02-25:
2038 testing shows working now on 32-bit build -- but library has to have _TIME_BITS capability along with the 64-bit time functions.  Now need to fix the time() function in debug.c to use this logic.
rig_test_2038: enter _TIME_BITS=64 testing enabled for unknown libc
rig_test_2038: __MSVCRT_VERSION__=0x0700
rig_test_2038: MSVCRT 2038 test = 0xf0000000:Mon Aug 05 04:04:00 2097
rig_init: 2038 time test passed