Hamlib/Hamlib #1483: Support for Guohe Q900 transceiver

olewsaa, 2024-01-22:
While the Yaesu FT-817  commands are supported by the Q900, it would be nice to have full support for the Q900.

mdblack98, 2024-01-22:
Added as a copy of FT-817.
What isn't supported?
rigctl -l | grep Gu
  1051  Guohe                  Q900                    20240122.0      Stable      RIG_MODEL_Q900

olewsaa, 2024-01-23:
Excellent, thanks. The Q900 has a lot of parameters, I will look into the differences between commands for 817 and Q900.

olewsaa, 2024-01-23:
I have attached a pdf manual which give a list of commands. The list of commands start at page 40.  I have not verified the commands listed here. So far I have just set the rig to FT-817. The essensial commands work, PTT, mode settings etc.

[FullmodeSDRRadioQ900Usermanual2.pdf](https://github.com/Hamlib/Hamlib/files/14027052/FullmodeSDRRadioQ900Usermanual2.pdf)
The manual can be downloaded from Guohe ltd.  https://pmo41839d-pic1.ysjianzhan.cn/upload/FullmodeSDRRadioQ900Usermanual2.pdf

clzls, 2024-06-23:
I have a Guohe PMR-171, which uses almost same command set with Guohe Q900. The frame format is totally different with quite simple FT-817 style 5-byte command, but a frame with 4-byte long header, variable length and CRC. I haven't look into the whole repo yet, but I think there is some rig using similar frame structure and we may build upon it.
Guohe protocol makes use of several advanced feature. For example, we can set VFO A and B's freq in one frame and no need to switch VFO A/B which needs to adjust filters while making noises (especially satellite ops). I would like to see it as soon as possible...

liCN, 2024-07-20:
Tested in 4.6+opensuse，It work on Q900v1 and PMR-119

closed 2024-07-20 by mdblack98

clzls, 2024-07-20:
Since it is different from FT8*7 now, may we consider moving them (Q900 and PMR-171) to its own folder and entitling the maker `Guohe`? :)

mdblack98, 2024-07-20:
It's not that different and I'm avoiding making new manufacturers if I don' t have to.  We have  limit of 64 manufacturers.
I'm considering making an "Others" category for these one-off brands.  They seems to be occurring a lot lately.

clzls, 2025-01-25:
> I have a Guohe PMR-171...

Tested PMR-171 on v4.6.1 and observed CRC calculation errors. I would look into it in next few weeks and try to make a usable minimum version, if it is desired.

mdblack98, 2025-01-25:
Previously reported as working so surprised you are seeing CRC errors.
Message ID: ***@***.***>

clzls, 2025-01-25:
> Previously reported as working so surprised you are seeing CRC errors.
> Message ID: ***@***.***>

If I select model `Q900`, it will work, but `Q900` entry is defined in `ft817.c` and it is using Guohe's "817 compatible mode". Model `PMR-171`, along with CRC checks, is defined in `pmr171.c` using Guohe's native protocols. I wonder whether others were testing the right model and which firmware version...

I saw in `pmr171.c` line 459 (also line 812) that it took 3 instead of 2 (length byte of 0x03, and 1 command byte) bytes into CRC16 calculations. After this patch, it passed CRC check on my PMR-171 with latest firmware v3.5.0 released in November 2024. But there's still bugs (freq's are in plain hex, not BCD code like Yaesu rigs, etc.) and I'm working on it.

By the way, according to upgrade notes, PMR-171 firmware v3.5.0 supports controls via UDP. Maybe we can make a new entry for it, in a later time.

A quick glance of the handshake between my computer and my rig:
```
...
rig_get_freq: cache miss age=62392ms, cached_vfo=VFOA, asked_vfo=VFOA, use_cached_freq=0
rig_get_freq(2574): vfo_opt=0, model=1052
pmr171_send_cmd1: called
write_block(): TX 9 bytes
0000    a5 a5 a5 a5 03 0b f9 37 00                          .......7.       
read_block_generic called, direct=1
read_block_generic(): RX 5 bytes, direct=1
0000    a5 a5 a5 a5 1b                                      .....           
read_block_generic called, direct=1
read_block_generic(): RX 27 bytes, direct=1
0000    0b 00 00 00 01 41 a7 c0 01 41 90 50 00 00 3c 3c     .....A...A.P..<<
0010    32 00 a6 05 2c 0c 21 01 00 14 b0                    2...,.!....     
from_bcd_be called
from_bcd_be called
pmr171_get_freq: vfoa=1420820, vfob=1419050
...
```
I tuned to 21080000 and 21074000 on my rig.

mdblack98, 2025-01-25:
Thanks for catching that .  It's been fixed.
Mike W9MDB