Hamlib/Hamlib #1489: Rigorous specification for the RigCTL protocol

AlexandreRouma, 2024-01-27:
Hello,

I've been trying to properly re-implement the rigctl client and server code of SDR++, the issue I'm running into is there doesn't seem to be a clear specification for the command arguments. The man page does list the commands but is unclear about the use of VFOs. With the addition of VFO arguments, it's not clear how those are supposed to be used or which commands support them or not.

Although running a custom server is not the original purpose of the protocol, good documentation is still required to write client code that can make use of all available feature.

Regards,
Alexandre Rouma, ON5RYZ (SDR++ author)

mdblack98, 2024-01-27:
You would be a lot better off emulating a rig instead of rigctld.Pick a rig like TS2000 or whatever covers your capability.  Then you will work with all software.Mike W9MDBSent from my iPhone

AlexandreRouma, 2024-01-28:
> You would be a lot better off emulating a rig instead of rigctld.

I don't think you understand. SDR++ is software, it can't "emulate" the serial connection of a real rig. Even if it could, that would be extremely jank.

Additionally, supporting the TCP protocol is a must have to have feature parity with other existing SDR software. For years they have all supported the protocol to an extent.

mdblack98, 2024-01-28:
Sure it can.  All you are doing is standard rig functions like freq, mode, ptt, bandwidth (assuming) and whatever else you support.

That's what PowerSDR and Thetis do (TS-2000 emulation).  And fully supported over a TCP connection too.

You'll be chasing yourself forever trying to emulate rigctld as it evolves.

Mike W9MDB

I don't think you understand. SDR++ is software, it can't "emulate" the serial connection of a real rig. Even if it could, that would be extremely jank.

Additionally, supporting the TCP protocol is a must have to have feature parity with other existing SDR software. For years they have all supported the protocol to an extent.

AlexandreRouma, 2024-01-28:
> You'll be chasing yourself forever trying to emulate rigctld as it evolves.

Well if it was properly defined this would not be a problem. Unless you don't care about cross-version compatibility (which for a network protocol where different machine might run different versions of the software is a must), the protocol should be designed in such a way that unsupported features fail safe. All other SDR software emulate the protocol, there is no getting around this if feature parity is desired.

I hope this issue can at least allow for the man page of rigctld to be brought up to date with the current protocol. And if you decide that the protocol should not be used by external software, this should be clearly stated instead of being advertised in the man page:
> It is hoped that rigctld will be especially useful for client authors using languages such as Perl, Python, PHP, and others.

mdblack98, 2024-01-28:
All those languages have direct access to the hamlib API and don't need rigctld access (it's just an option on the almost 300 rig list).  That's the best way to write a client so you can connect to anything.

rigctld is useful from a client perspective but not from a build-your-own-rigctld-server perspective.
That's why it is documented.  Never intended for anybody to try and emulate rigctld and you are the only I know doing it.  Way too much work IMHO.

If you insist on staying on this path (I imagine you have a lot of time vested) how do you suggest the man pages reflect vfo usage to make it easily understandable that it is optional?

P.S. Hamlib supports TCP connections to any rig.  I use VSPE Serial/TCP server to connect to the rigs in my shack -- you just use IP:PORT instead of COM port.  For example 192.168.1.164:5503 connects to my IC7300.  And I can do that from any machine on my network.

I also have rigctlcom.c in Hamlib which emulates a TS-2000 so that software like N1MM (or most any software that doesn't know about Hamlib) can connect to FLRig for example.

Also...all efforts are made to maintain backwards compatibility but that most certainly cannot be guaranteed going forward.  A major version change may signal rigctld incompatibility.  On the other hand the TS-2000 will never lose backwards compatibility and is why I suggested you emulate a rig instead.

Mike W9MDB

>  
> You'll be chasing yourself forever trying to emulate rigctld as it evolves.

Well if it was properly defined this would not be a problem. Unless you don't care about cross-version compatibility (which for a network protocol where different machine might run different versions of the software is a must), the protocol should be designed in such a way that unsupported features fail safe. All other SDR software emulate the protocol, there is no getting around this if feature parity is desired.

I hope this issue can at least allow for the man page of rigctld to be brought up to date with the current protocol. And if you decide that the protocol should not be used by external software, this should be clearly stated instead of being advertised in the man page:

AlexandreRouma, 2024-01-28:
> rigctld is useful from a client perspective but not from a build-your-own-rigctld-server perspective.

I understand that, however it is by far the most general option. Pretenting to be some random rig only works up to one point.
And I need to write a client anyway. That means I'll have the protocol objects already in the code and writing the server part isn't much work at all.

> you are the only I know doing it.

That's not true:
GQRX: https://www.gqrx.dk/doc/remote-control#more-162
SDRAngel: https://github.com/f4exb/sdrangel/blob/master/plugins/feature/rigctlserver/readme.md

These are the two other most used applications on linux.

> If you insist on staying on this path (I imagine you have a lot of time vested)

I don't really have much time vested, I just don't have the choice to keep feature parity with other apps. People expect to be able to connect to it a certain way with existing tools like gpredict. Which by the way, adds commands not defined in the rigctld  man page for AOS/LOS of satellites (used to start/stop recordings).

mdblack98, 2024-01-28:
OK...so we're left with how to reflect VFO usage in the man pages.

Suggestion?

Mike W9MDB

AlexandreRouma, 2024-01-28:
Well that's the issue, I myself don't really understand how/when they're supposed to be used, so I'm not sure what I can help with on that side. 

Ideally, it'd be nice to clearly define which commands accept a VFO as an argument and how this argument should be given to it (I think it's positional? but not sure).

mdblack98, 2024-01-28:
I'll propose this where [] indicates an optional parameter and then add a note at the top about VFO mode.

For example

       f, get_freq
              Get 'Frequency', in Hz.

Change to
       f, get_freq [VFO]
              Get 'Frequency', in Hz of current VFO.  Or with VFO option the requested VFO.

mdblack98, 2024-01-28:
I'll add that it's easy to see when it's required with rigctl -- and the need for VFO has changed at times to add or remove the need.
If you send a command without VFO you'll get prompted for it.

Rig command: f
VFO: VFOA
Frequency: 145000000

Rig command: m
VFO: VFOA
Mode: FM
Passband: 15000

AlexandreRouma, 2024-01-28:
> I'll propose this where [] indicates an optional parameter and then add a note at the top about VFO mode.

Yup, that sounds good!

By the way, I was reminded by reading the bug reports I had gotten that hamlib often calls `\dump_state` upon connection. It used to cause issues because rejecting the command (RPRT -1) would cause the connection to just close. It was specifically with WSJT-X, not sure if this was something all software using hamlib would use.

Since that command is specifically for the internal state of rigctld, I agree that it should just not be implemented in third party clients/server. Does hamlib force the use of this command by always sending it out or does software using hamlib have control over calling it or not?

mdblack98, 2024-01-28:
dump_state is needed to find out the current state of the rig interface.
But in your case it probably isn't needed since I doubt you have any options which would change the state.
Things like PTT type and such can change and not match the rig_caps.

It will be called every time the rig is opened.

Mike W9MDB

Yup, that sounds good!

By the way, I was reminded by reading the bug reports I had gotten that hamlib often calls \dump_state upon connection. It used to cause issues because rejecting the command (RPRT -1) would cause the connection to just close. It was specifically with WSJT-X, not sure if this was something all software using hamlib would use.

Since that command is specifically for the internal state of rigctld, I agree that it should just not be implemented in third party clients/server. Does hamlib force the use of this command by always sending it out or does software using hamlib have control over calling it or not?

N0NB, 2025-07-31:
I'm not sure if this has been resolved to the OP's satisfaction, and perhaps it never can be.