Hamlib/Hamlib #1496: IC-705 Tone  CTCSS

marcinoza, 2024-01-30:
How to run rigctld so that it does not erase TONE CTCSS while transmitting.

Example:
ISS GPREDICT
145.990 MHz / CTCSS: 67 Hz – (Uplink VHF)
437.800 MHz – (Downlink UHF)

When I set the tone ccts to 67Hz on the radio and run rigctld, the tone setting disappears.

mdblack98, 2024-01-30:
When you say the tone setting disappears you mean the visual display?
That happens but it should not be turning off Tone...if you select the function again it should still be on.
It's just the display that disappears due to doing other things.

Mike W9MDB

marcinoza, 2024-01-31:
The tone disappears as soon as rigctld is enabled, I try to manually enable it, but when gpredict starts updating the frequency, the tone disappears, over and over again.

Maybe gpredict doesn't send the tone setting to the radio, so I thought that maybe there is an option in rigcrtld to not change this setting. or maybe there is another way.

mdblack98, 2024-01-31:
I need rigctld debug info.  Tone should not be touched at all.

For rigctld add this

-vvvvv >log.txt 2>&1

Duplicate the problem and send me the log.txt file

marcinoza, 2024-01-31:
Linux marcinoza 6.5.0-15-generic #15-Ubuntu SMP PREEMPT_DYNAMIC x86_64 GNU/Linux

i  hamradio-datamodes                                          0.8                                         all
ii  hamradio-files                                              20230630                                    all
ii  hamradio-tasks                                              0.8                                         all 
ii  libax25                                                     0.0.12-rc5+git20230513+5581005-1            amd64
ii  libhamlib-dev:amd64                                         4.5.5-3                                     amd64
ii  libhamlib-utils                                             4.5.5-3                                     amd64
ii  libhamlib4:amd64                                            4.5.5-3                    amd64      
ii  python3-hamcrest                                            2.0.3-2                                     all
ii  python3-hamlib                                              4.5.5-3            amd64radio transceivers and receivers

ICOM IC-705  1.32
[log.txt](https://github.com/Hamlib/Hamlib/files/14118469/log.txt)

TONE disappears when you press the Engage button in Gpredict.

mdblack98, 2024-02-01:
These are the only commands that I see that might affect tone

0000    fe fe a4 e0 03 fd                                   ......
0000    fe fe a4 e0 07 00 fd                                .......
0000    fe fe a4 e0 0f fd                                   ......
0000    fe fe a4 e0 18 fd                                   ......
0000    fe fe a4 e0 1a 03 fd                                .......
0000    fe fe a4 e0 1c 00 fd                                .......
0000    fe fe a4 e0 26 00 fd                                ....&..

To see if one of these is causing is run rigctl instead of rigctld

Turn on the tone then execute each command and see if it turns it off.

Do it this like this for each command...first one is the example here

w \0xfe\0xfe\0x94\0xe0\0x03\0xfd

The command will timeout and might give an answer but all we're interested in is the tone turning off.

Mike W9MDB

marcinoza, 2024-02-01:
I did a test, I hope it's OK, none of these commands turned off TONE

root@marcinoza:~# rigctl -m 3085 -r /dev/ttyACM0 -s 115200 -c 0xA4
Rig command: f
Frequency: 145500000
Rig command: c
CTCSS Tone: 670
Rig command: w
Cmd: \0xfe\0xfe\0x94\0xe0\0x03\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: w                                  
Cmd: \0xfe\0xfe\0xa4\0xe0\0x07\0x00\0xfd
Reply: 
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0xf\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0x18\0xfd
Reply: 
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0x18\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0x1a\0x03\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0x1c\0x00\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: w
Cmd: \0xfe\0xfe\0xa4\0xe0\0x26\0x00\0xfd
Reply: 
Rig command: c
CTCSS Tone: 670
Rig command: f
Frequency: 145500000
Rig command: _
Info: None

73 Marcin SP4WRI

marcinoza, 2024-02-01:
I think I found the reason for TONE deletion

If in VFO mode it is constant, e.g. 430.000 and you set TONE, it is ok, but just change the frequency, the radio turns off TONE.

Gpredict changes the frequency but does not enable TONE.

It does not send the turn on tone command.

marcinoza, 2024-02-01:
This is the Gpredict problem.

Thank you for your help.

marcinoza, 2024-02-01:
Do you know what command to send set_tone 67Hz?

mdblack98, 2024-02-01:
Assuming you can compile the latest version get it from the git master or wait until tonight's build if you want at https://n0nb.users.sourceforge.net/

I added a new option

--set-conf=tone_enable=1

This will re-enable tone after every frequency change.

Mike W9MDB

marcinoza, 2024-02-01:
it works, I recompiled the new version and it works :))

/opt/hamlib/bin/rigctld -m 3085 -r /dev/ttyACM0 -s 115200 -c 0xA4 -t 4533 --set-conf=tone_enable=1

Thank you very much for your help

closed 2025-04-05 by N0NB