Hamlib/Hamlib #1500: Enable both static and dynamic library builds

mdblack98, 2024-01-31:
See if we can create both libhamlib.a and libhamlib.so during the same build.
Utilities can be built against the shared library I think so it's really just building the static library along with shared.

hamlibdk, 2024-02-01:
Intent of this request is to better support the integration of the FL-toolkit into the JTSDK.

Initial request asked for tools etc. that were say dynamically linked at the end of the process. On reflection the endpoint tools, if produced at all by this pathway, should be static linked.

hamlibdk, 2024-04-29:
WIth Ubuntu 24.04 LTS released this is now possibly of greater need. Many Linux distros base off the Ubuntu repos. 

Maintainers will not release .deb / snap packages that will not support both Static and Dynamic Endpoints.

N0NB, 2025-04-08:
I'm not sure what Mike's reason for disabling concurrent static and shared libraries with commit 43159e55a.  His commit message doesn't reference an issue and I didn't find anything relevant in that time frame when searching the closed issues.

My quick testing shows that commenting out the lines added in that commit restore the default behavior of building shared and static libraries concurrently.  The compiled utilities are dynamically linked to the shared library so I don't see an issue there.  I will continue to research and I may drop a revert commit in a few days for wider testing.