Hamlib/Hamlib #1502: Adding USB-D1,USB-D2,USB-D3 modes to flrig for Kenwood TS-990S

bpuklich, 2024-02-02:
I wouldn't mind taking a crack at adding these supported modes to the flrig model so Kenwood TS-990S has a bit better support. I've been using rigctld to allow MacLoggerDX to talk flrig which in turn talks to my Kenwood TS-990S. fldigi can then use flrig, same time. Could also add all the others with D1-D3 suffixes once I figure out the pattern used. I've been studying the source code a little here so if there are things to know to get up to speed quickly let me know.

Thanks much.

mdblack98, 2024-02-02:
I'm more than a little reluctant to add that many modes for just one rig.We have a 64-bit limit on the mode list and we're up to 39 now and this would add 12 more taking up 50% of the remaining space.
Is there some big advantage to these?
I may be able to make an "exception" for the modes so these could be put in the backend of the ts950 instead of for all rigs.That would a new thing for mode but would allow some expansion beyond 64 bits and be applicable for unique rigs which we're starting to see more of.
Mike W9MDB

bpuklich, 2024-02-02:
Thanks for the quick reply.When you said 64 bit I had to go back and look at what you meant. You’re right, that’s pretty limited and that makes this a not awesome idea.Why was this coded as a bitmap? Usually those are used for other types of reasons, but not for discrete values like I would imagine rig modes are. If they’re combined in some way then it might make sense, but if they’re just flags then integers would be more than adequate.Enlighten me!So, where this was all coming from is that with the Hamlib I installed using brew, using rigctld to talk to flrig and having MacLoggerDX talk to rigctld somehow would cause MacLoggerDX to not connect to the radio when the mode of the radio in flrig was one that Hamlib didn’t have a mode for. I didn’t dig into that interaction any further than to see that the -D1,D2,D3 modes were not understood by Hamlib.Interestingly, when I compiled and used my own Hamlib the MacLoggerDX radio connection stayed put. The flrig USB-D1 mode though is seen by MacLoggerDX as USB-D, which smells like it’ll cause a problem at some point.Maybe it’ll be a problem again when using MacLoggerDX to try and set one of these modes that Hamlib understands but the radio through flrig doesn’t, or some other interaction like that.---Blayne Puklich             Minneapolis, MN       PGP Key ID: 0xC52CA6C1* mailto:blayne at puklich.com         I'd explain it to you, but your brain would explode