Hamlib/Hamlib #1514: CW Morse does not work with tlf

mdblack98, 2024-02-20:
Thread collisions are causing problem with tlf
https://github.com/Tlf/tlf
tlf is making these calls 
***2:rig.c(2333):rig_get_freq entered
****3:rig.c(2856):rig_get_mode entered
****3:rig.c(3315):rig_get_vfo entered
***2:rig.c(5757):rig_get_split_vfo entered

Perhaps using a MUTEX while send_morse is active and having these functions return cached info while send_morse is active?

closed 2024-08-18 by mdblack98

N0NB, 2024-08-18:
I didn't realize this needed testing.  Is this supported by the Elecraft K3 backend?

mdblack98, 2024-08-18:
Yes...should be supported.

N0NB, 2024-08-19:
* On 2024 18 Aug 17:46 -0500, Michael Black wrote:
> Yes...should be supported.

Now to figure out how to do it!

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819