Hamlib/Hamlib #1517: Make GUI for rigctlcom

mdblack98, 2024-02-23:
Make a GUI for rigctlcom.

N0NB, 2024-02-23:
I've been thinking that something like Python and Tkinter would be a good cross-platform choice for something like this.  Tkinter now has themed widgets which look much better, at least on the Linux side, than the old widgets.  Unlike GTK or Qt Tkinter is quite stable which should keep ongoing maintenance of the GUI much lower than with the former tool kits.

mdblack98, 2024-02-23:
Ain't gonna' do python as don't want users to have to install python.

What I've been testing is Avalonea which is cross platform Windows/MacOS/Android/IOS.

I've got the multicast receiver working on Windows/Linux but have to implement the control channel going back at a minimum before adding it to Hamlib.

Mike

GeoBaltz, 2024-02-24:
I'm with Nate on this one.  Much more likely to have Python already on a system that to have to get Avalonia/Visual Studio/.NET working - especially since i don't see Linux in your supported list.

mdblack98, 2024-02-24:
Except the binary will be distributed with Hamlib -- no need to build it...or install python.

Mike W9MDB

N0NB, 2024-02-24:
* On 2024 23 Feb 22:12 -0600, Michael Black wrote:
> Except the binary will be distributed with Hamlib -- no need to build
> it...or install python.

First, Python has a method for doing this as well so "installing" Python
would not be required.

Second, I am not in favor of distributing a GUI program within the
existing generated binary packages.  As a separate binary?  Sure, that
should be doable (note, I still have no facilities for building binaries
for MacOS).

Third, if development or even generating binaries for this GUI require
MS tools, then I am unable to assist with it.

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

mdblack98, 2024-02-24:
I didn't mean putting the binary in the GIT repository -- I meant with the installation packages.

I'd think you should be able to build at least on your 64-bit system.

And if python and Qt would quite breaking things I would have considered it.
So at this point I'm just forward-looking to future compatibility.

If you want to try and build here's my current testing version and will start receiving the multicast packets whenever rigctl or hamlib is running.

https://github.com/mdblack98/MyApp

Mike W9MDB

First, Python has a method for doing this as well so "installing" Python
would not be required.

Second, I am not in favor of distributing a GUI program within the
existing generated binary packages. As a separate binary? Sure, that
should be doable (note, I still have no facilities for building binaries
for MacOS).

Third, if development or even generating binaries for this GUI require
MS tools, then I am unable to assist with it.

73, Nate

N0NB, 2025-07-31:
Closely related to #1370.