Hamlib/Hamlib #1530: Rigctl timeout when sending raw (extended) command that gets a response.

wwortel, 2024-03-25:
Struggling with the use of some extended commands of an Anan G2 SDR.
Using the rigctl basic command set and model Kenwood TS2000 (which the Anan emulates), e.g. to ask for the transmit frequency, works fine.
But querying the Anan with an extended command, using the w option (send command), the command times out even though the Anan answers correctly.
Problem is that the routine in rigctl that does the communication is not given the proper expected length of an answer (it doesn't know the command) and once past the default 1 length value it misses the ';' that ends the answer that was given and keeps waiting for more, that does not come, until time out.
It has worked correctly in the past. Problem exists in current Hamlib.

mdblack98, 2024-03-25:
Use the upper case W.  Lower case always times out looking for ALL input from the rig.  Unknown how many packets may be received.

e.g.

W FA; ;

The 3rd argument is the terminator.

What extended command are you using?  Perhaps it should be added to hamlib?

Mike W9DMB

wwortel, 2024-03-25:
@w9dmb
Thanks for the suggestion! Will try it.

mdblack98, 2024-03-25:
Please close the ticket if it works OK.

wwortel, 2024-03-25:
Issue can be closed. Using W option, and with both the command and reply termination character between single or double quotation marks, the  reply is given correctly.

closed 2024-03-25 by wwortel