Hamlib/Hamlib #1531: Hamlib not working with Log4OM

mdblack98, 2024-03-25:
Log4OM is timing out trying to connect to rigctld

This does not work
rigctld Hamlib 4.6~git 2023-12-22T13:51:48Z SHA=70c357 64-bit

This works with Log4OM
rigctld Hamlib 4.6~git May 31 19:22:37Z 2023 SHA=b34f18 64-bit

mdblack98, 2024-03-25:
This is where it fails
003b50bc57a376556c96fbc8e60e6ed1c66a7bb1 is the first bad commit
commit 003b50bc57a376556c96fbc8e60e6ed1c66a7bb1
Author: Mike Black W9MDB <mdblack98@yahoo.com>
Date:   Sun Aug 13 23:23:23 2023 -0500

    Standardize extended response in rigctld for level, func, and parm
    ,l VFOA AGC
    get_level: VFOA,AGC,0,RPRT 0

    ,u VFOA RIT
    get_func: VFOA,RIT,0,RPRT 0

    ,p ANN
    get_parm: currVFO,ANN,0,RPRT 0

    Error example
    ,p NB
    get_parm: currVFO,NB,RPRT -1

    https://github.com/Hamlib/Hamlib/issues/1361

 tests/rigctl_parse.c | 28 ++++++++++++++--------------
 1 file changed, 14 insertions(+), 14 deletions(-)

PianetaRadio, 2024-03-26:
Oh, this is the reason why I cannot connect with log4OM to my radio!
Now it's working.

PianetaRadio, 2024-04-12:
I'm using for several days Log4OM after this commit and rigctld now is working well.
You can close this issue.

closed 2024-04-12 by mdblack98