Hamlib/Hamlib #1538: GPIO PTT broken as of recent Linux kernel updates

aschuhardt, 2024-04-10:
## Device

- Platform: Raspberry Pi 4B
- OS: Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

## Background

The [sysfs GPIO interface](https://docs.kernel.org/admin-guide/abi-obsolete.html#abi-sys-class-gpio) was removed in a recent Linux kernel update:

> This ABI is obsoleted by [testing/gpio-cdev](https://docs.kernel.org/admin-guide/abi-testing.html#abi-file-testing-gpio-cdev) and will be removed after 2020.

See this blog post for more background info: https://waldorf.waveform.org.uk/2022/the-one-where-dave-breaks-stuff.html

I found out about this while trying to troubleshoot the above error and stumbling across this forum response: https://forums.raspberrypi.com/viewtopic.php?t=359540#p2157021

## Hamlib Error

The effect of this is that [the Hamlib code](https://github.com/Hamlib/Hamlib/blob/master/src/gpio.c#L41) which relies on using `/sys/class/gpio` to manage GPIO ports will now throw an IO error.

```
$ sudo rigctl -P GPIO -p 4 -vv
GPIO4 direction (using /sys/class/gpio/gpio4/direction): No such file or directory
rig_open: cannot open PTT device "GPIO4"
rig_open: error = rigctl_parse.c(1974):set_conf return(0)
 1:rig.c(817):rig_open entered
rig_settings_get_path: path=/root/.hamlib_settings
rig_settings_load_all: settings_file (/root/.hamlib_settings): No such file or directory
rig_open: cwd=/home/aschuhardt
rig_open: /home/aschuhardt/hamlib_settings does not exist
rig_open: async_data_enable=0, async_data_supported=0
GPIO4 direction (using /sys/class/gpio/gpio4/direction): No such file or directory
rig_open: cannot open PTT device "GPIO4"
 1:rig.c(1237):rig_open returning(-6) IO error

IO error
 (null) No such file or directory
```

aschuhardt, 2024-04-10:
I'll start on a PR to integrate [libgpiod](https://github.com/brgl/libgpiod).

closed 2024-05-12 by mdblack98

mattbk, 2026-02-08:
Should this have been marked completed? It looks like #1539 was reverted in #1551.

craigerl, 2026-10-28:
I'm on hamlib 4.7.2, and rigctld is still trying to open the deprecated /sys/class/gpio stuff.   Is there libgpiod support coming at some point?  Raspberry Pi OS has long since abandoned the old rpigpio api.   Modem's like Xara have no native gpio support and rely on hamlib.  Direwolf is doing all the lifting otherwise.

craigerl, 2026-11-01:
the regressed commit uses libgpiod api v1.x which is deprecated... again.  compilation fails.

Need to update gpio.c to use the newer llibgpiod API v2.x function names.   AI opportunity?

craigerl, 2026-11-01:
https://github.com/Hamlib/Hamlib/issues/2293   new gpio.c fixes this, I don't have the cognitive capacity right  now for a pull request, but i'd be happy to test.

PR #1539 (2024-05-12): Refactoring GPIO push-to-talk logic to use libgpiod rather than the d…
…eprecated/broken sysfs method.

I've confirmed that this works on my Raspberry Pi 4B.

TODO: I'm not familiar with autoconf to set up linking to libgpiod correctly, so I'd like to ask someone else to do that.
TODO: Should we keep both implementations for the sake of backwards-compatibility?

Fixes #1538

I've included the verbose rigctld output below.

```
â  Hamlib git:(libgpiod) â sudo ./tests/rigctld -P GPIO -p 4 -vvvvv              
rigctld.c(640) Startup: /home/aschuhardt/Hamlib/tests/.libs/rigctld -P GPIO -p 4 -v [... cut]
commit cf91e963 (2024-05-12): Refactoring GPIO push-to-talk logic to use libgpiod rather than the deprecated/broken sysfs method.
TODO: I'm not familiar with autoconf to set up linking to libgpiod correctly, so I'd like to ask someone else to do that.
Fixes #1538