Hamlib/Hamlib #1540: Direct interface to routines removed in lieu of API?

markjfine, 2024-04-21:
Not a terrible issue, more of an FYI and a question:

Some older code (e.g., grig) used direct calls to things like `rig_list_foreach()`. Seems the only interface available to build from is the API call for this now. So when an application's code tries to build using something like:
`rig_list_foreach (grig_list_add, (void *) array);`
it gets a compile error like:
```
/usr/local/include/hamlib/rig.h:3764:39: note: passing argument to parameter 'cfunc' here
rig_list_foreach HAMLIB_PARAMS((int (*cfunc)(struct rig_caps *, rig_ptr_t),
```
Is there a reason why those interfaces were removed, or am I doing something completely wrong here?

mdblack98, 2024-04-21:
Remove the "const" in got the declaration and implementation of grid_list_add 

static gint        grig_list_add       (const struct rig_caps *, void *);

To
static gint        grig_list_add       (struct rig_caps *, void *);

Mike W9MDB

markjfine, 2024-04-21:
That fixed it... If I see it anywhere else I'll be sure to do the same thing.
Thanks Mike!

closed 2024-04-21 by markjfine