Hamlib/Hamlib #1549: Build fails on Fedora core 39

j4mbob, 2024-05-11:
Hi,

trying to buil hamlib from source on a FC39 machine and get the following when running make:

make[1]: Entering directory '/home/jamie/Hamlib/c++'
  CXX      rigclass.lo
  CXX      rotclass.lo
  CXX      ampclass.lo
  CXXLD    libhamlib++.la
make[1]: Leaving directory '/home/jamie/Hamlib/c++'
Making all in tests
make[1]: Entering directory '/home/jamie/Hamlib/tests'
  CC       rigctl-rigctl.o
rigctl.c: In function ‘main’:
rigctl.c:714:9: error: ‘rl_readline_name’ undeclared (first use in this function)
  714 |         rl_readline_name = "rigctl";
      |         ^~~~~~~~~~~~~~~~
rigctl.c:714:9: note: each undeclared identifier is reported only once for each function it appears in
rigctl.c:717:9: warning: implicit declaration of function ‘using_history’; did you mean ‘write_history’? [-Wimplicit-function-declaration]
  717 |         using_history();    /* Initialize Readline History */
      |         ^~~~~~~~~~~~~
      |         write_history
make[1]: *** [Makefile:1423: rigctl-rigctl.o] Error 1
make[1]: Leaving directory '/home/jamie/Hamlib/tests'
make: *** [Makefile:635: all-recursive] Error 1

can anyone suggest anything?

many thanks

mdblack98, 2024-05-11:
Try this:
sudo dnf makecache --refresh
sudo dnf -y install libreadline-dev

j4mbob, 2024-05-11:
ah yes that works, installing the readline-devel rpm. sorry, i should have thought to look if readline was an external libary that was needed

thank you

closed 2024-05-11 by mdblack98