Hamlib/Hamlib #1555: Incorrect binding swig definition for get_split_vfo

kyleboyle, 2024-05-27:
https://github.com/Hamlib/Hamlib/blob/18568986f85688a6a2d6ae89c950af94dea6be2f/bindings/rig.swg#L448

I don't think get_split_vfo is setup correctly in the bindings. It shouldn't take any arguments, as the `s, get_split_vfo` command in rigctl doesn't take any arguments.

N0NB, 2025-06-22:
@dforsi Daniele, is this related to anything you've worked on recently?  Is it possible you've resolved this?

dforsi, 2025-06-22:
No, it's one that I wasn't able to fix yet.

dforsi, 2025-06-25:
If you run `rigctl --vfo` then the VFO argument is mandatory. The binding will allow an optional vfo argument.
```
$ tests/rigctl \get_split_vfo
0
None
$ tests/rigctl --vfo \get_split_vfo
0
None
Invalid arg for command 'get_split_vfo'
$ tests/rigctl --vfo \get_split_vfo VFO_A
0
None
```

dforsi, 2025-08-11:
This issue can be closed.
Fixes are in:
* commit 453e68c6cd6d5b21286cc914773cfdb029ca86c7 (the bindings)
* commit bc25855f48edf631d7653345f0bb763759094ba5 (the test)

closed 2025-08-11 by N0NB