Hamlib/Hamlib #1556: python binding for power2mW

kyleboyle, 2024-05-27:
How can the power2mW function be invoked from python binding? I don't see it exposed.

mdblack98, 2024-05-27:
Are you using to the the transmit power level?
If so, take a look at RIG_LEVEL_RFPOWER_METER_WATTS

kyleboyle, 2024-05-28:
I'm testing with FT-710.. RFPOWER_METER_WATTS seems to always return 5.0. The RFPOWER level value seems to correspond with changing the RF Power setting on the radio (but it's not in a watt unit). Invoking power2mW via rigctl doesn't yield a matching value to the radios RF Power setting so this is probably a fruitless endeavour. The power2mW function is reporting the power level at 39.2 % of the radio's setting.

mdblack98, 2024-05-28:
What version of hamlib?

Send me a small python test script just for RFPOWER_METER and RFPOWER_METER_WATTS so I can see what you are doing.

Also run rigctl and generate debug so I can see what the rig is doing.

rigctl -m 1049 -r /dev/xxxxx -s 19200 -vvvvv -Z l RFPOWER_METER l RFPOWER_METER_WATTS >log.txt 2>&1

And send the log.txt file.

Mike W9MDB

kyleboyle, 2024-05-28:
I'm not doing anything in python with this right now, the _Hamlib module just doesn't have the power2mW function. But this function doesn't appear to be doing anything useful anyway (for 710 at least)

RFPOWER_METER l RFPOWER_METER_WATTS l RFPOWER
0.050000
5.000000
0.392157

[log.txt](https://github.com/Hamlib/Hamlib/files/15475895/log.txt)

dforsi, 2025-04-26:
@kyleboyle

the values for RFPOWER_METER and RFPOWER_METER_WATTS are consistent with what your radio is returning, eg. see this line:
> 2024-05-28T17:49:00.917309-0400: newcat_get_cmd: read count = 10, ret_data = RM5000000;

the first 000 is the power level, so the radio reported that it wasn't transmitting at that time (eg. in SSB when PTT is pushed but not speaking on the mike) and Hamlib was working as it is supposed to do, that is mapping 0 to the lowest possible value for a given radio which 0.05 for FT-710 and when converted to watts gives 5; we could argue that 0 means no output but this is a different issue.

For RFPOWER the output is clearly wrong, but you were using Hamlib 4.5.5 and there have been changes in version 4.6 for  FT-710 and now RFPOWER is converted in the same way as other Yaesus, so it would be useful if could test again and confirm if `get_level RFPOWER` is working for you and returning the same power setting in Watts as set and changed from the radio, when not transmitting.

N0NB, 2025-06-22:
@kyleboyle Does Daniele's comment help?  Have you tested with the 4.6.3 release?

closed 2025-06-22 by kyleboyle