Hamlib/Hamlib #1559: Macos & usb output problem

kyleboyle, 2024-05-31:
I have a FT-170 which provides a sound output over the usb connection which is also providing the com ports for cat control. On my system, the com ports are listed (by flrig) as:
```NONE
xml client
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.JabraEvolve75
/dev/cu.SLAB_USBtoUART
/dev/cu.SLAB USBtoUART3
/dev/tty.Bluetooth-Incoming-Port
/dev/tty.JabraEvolve75
/dev/tty.SLAB USBtoUART
/dev/tty.SLAB USBtoUART3
```

I can successfully use rigctl on /dev/cu.SLAB_USBtoUART and /dev/cu.SLAB_USBtoUART3, and flrig can connect there as well.

What is interesting is if flrig or rigctld is connected, I can no longer use the usb sound output device provided by the radio. If I run a program which plays sound on the device, it results in a few clicks on the sound monitor and seg faults or in the case of using it as the system output, the system output reverts to the previous. If I close out flrig or stop rigctld, running the same sound output test results in transmitting the sound on the radio with no issue as expected.

Doing these same tests in windows works fine.

Using flrig 2.0.05 and rigctl[d] via homebrew installed 4.5.5 `rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3`
```
Model Name:	MacBook Pro
  Model Identifier:	MacBookPro18,4
  Chip:	Apple M1 Max
  Total Number of Cores:	10 (8 performance and 2 efficiency)
  Memory:	64 GB
  System Firmware Version:	7459.141.1
  OS Loader Version:	7459.141.1
```

mdblack98, 2024-05-31:
This is not a Hamlib problem.

I assume in Windows you have a COM port and a Sound device showing.

In MacOS appears you need to use portaudio to talk to the sound device.

https://stackoverflow.com/questions/14867033/which-dev-devices-are-the-microphone-and-speaker-in-mac-os-x

Mike W9MDB

kyleboyle, 2024-05-31:
I think maybe you misread some of what I wrote. I can use the USB sound device fine when hamlib (rigctld or flrig) isn't running. I'm using the sound device correctly - I can simply choose it as my system output if I want to, or for example in python using the sounddevice library i can target the "USB Audio Device, Core Audio" device name. I listed the /dev/ items to show how the serial devices for cat control appear from my radio for context around how im using rigctl or flrig to connect to the radio.

mdblack98, 2024-05-31:
As you've noticed it works in Windows so there it's not an FLRig or Hamlib problem.

I have no idea why this doesn't work on MacOS.

I suggest you try portaudio to see if behaves better.

Mike W9MDB

kyleboyle, 2024-05-31:
Just to re-iterate the simplest reproduction scenario. If I set my system output to the rig's sound input channel and start playing an audio file, the rig correctly transmits the audio file. If I then start rigctld, the sound device stops working.
[image]

Maybe the macos drivers are buggy. I haven't confirmed on windows yet - I forgot that I was using omnirig integration last time I was testing on windows.

There seems to be a similar report here on cat & audio not working together for my rig: https://wsjtx.groups.io/g/main/topic/mac_m2_pro_yaesu_ft_710/100440419?p=,,,20,0,0,0::recentpostdate/sticky,,,20,2,40,100440419,previd%3D1690927941570392025,nextid%3D1690403926674990743&previd=1690927941570392025&nextid=1690403926674990743

dforsi, 2025-04-25:
@kyleboyle did you do tests on Windows?
Have you had a chance to test on you Mac Hamlib version 4.6.2 which was released on Feb 9, 2025?
You mentioned Python, do you have some code to share so that other people can try to reproduce your issue?