Hamlib/Hamlib #1560: Generic RTS PTT device

Dinsmoor, 2024-06-01:
Background: Linux users trying to use a Baofeng using a Digirig using software that depends on Hamlib

Firstly, I am aware of rigctl's invocation for what I am asking for: `rigctld --ptt-file=/dev/ttyUSB0 --ptt-type=RTS`

New users may have an issue with there not being a rigfile, as when/how do do a generic serial RTS ptt is not readily apparent (outside of writing your own software), and often not covered in any other manuals. The workflow for using rigctld is usually "find your rig" -> "set your rig and serial port."

It would make it much easier for users of generic RTS PTT devices (such as the Digirig: digirig.net) to have a "DigiRig (RTS Generic)" to have a rigfile that does the same thing as above. It would eliminate a lot of confusion and be much like the normal 'hamlib workflow'.

`rigctld -l` may show
```
 Rig #   Mfg                    Model                   Version         Status      Macro
12345    Digirig                Generic                 202405031.0     Stable      RIG_RTS_GENERIC

``` 

`rigctld -m 12345 -r /dev/ttyUSB0` would be equivalent to `rigctld --ptt-file=/dev/ttyUSB0 --ptt-type=RTS`

I have no experience contributing to Hamlib but if nobody experienced hops up, I'll give it a shot.

mdblack98, 2024-06-13:
Have you tested this?

Dinsmoor, 2024-06-14:
Not yet, the weather has been great and I've paused work on the computer. I should be able to compile and test in a day or two though. Thanks.

Dinsmoor, 2024-06-16:
I compiled the latest Master branch on [a462bc1](https://github.com/Hamlib/Hamlib/commit/a462bc19ae1f28a86bd8386ab2dceb3690592493)

Handheld is on /dev/ttyUSB2 connected via a digirig.

`rigctl --ptt-file=/dev/ttyUSB2` does not work (no crash, but does not do anything when I )
`rigctl --ptt-file=/dev/ttyUSB2 --ptt-type=RTS` yields segfault
here's the GDB output:

```
tyler@dinsmoor:~/ham/Hamlib$ gdb rigctl
GNU gdb (Ubuntu 12.1-0ubuntu1~22.04) 12.1
...
Reading symbols from rigctl...
(gdb) set args --ptt-file=/dev/ttyUSB2 --ptt-type=RTS
(gdb) run
Starting program: /usr/local/bin/rigctl --ptt-file=/dev/ttyUSB2 --ptt-type=RTS
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x000055555555d5d6 in main (argc=3, argv=0x7fffffffe038) at rigctl.c:334
334	                my_rig->caps->ptt_type = ptt_type = RIG_PTT_SERIAL_RTS;
(gdb)
```

I have never used gdb before, and I just hopped on the computer to test this real quick.

closed 2024-07-04 by mdblack98