Hamlib/Hamlib #1566: Icom 2730A -> Hamlib -> GPredict , Not making the doppler adjustments

curtiswilliams, 2024-06-13:
I am running a .bat : rigctld -vvvvv -m 3072 -r COM1

It's not working great with adjusting the frequency as GPredict makes doppler shift adjustments. Can you identify any issue or suggest a better .bat command that may assist in getting the frequency to consistently change with what GPredict is changing the frequency to?

In GPRedict I have Host localhost, Port 4532, Radio type I've tried all choices, PTT Status I've tried all settings, also VFO Up/Down same thing. No matter what I do with settings I can't resolve the issue. 
[image]

LOG SEGMENT

C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5~git\bin>rigctld -vvvvv -m 3072 -r COM1
rigctld.c(626) Startup: rigctld -vvvvv -m 3072 -r COM1
rigctld Hamlib 4.5~git Tue May 17 22:03:48 2022 +0000 SHA=468dd1
Report bugs to <hamlib-developer@lists.sourceforge.net>

Max# of rigctld client services=32
rig_check_rig_caps: p1=000000005C104E60, p2=000000005C10B3D8, rig_model=000000005C104E60, macro_name=000000005C10B3D8
initrigs4_icom: _init called
rig_init(496): 0000000000F2AFFC rs->comm_state==0?=0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
 1:icom.c(613):icom_init entered
icom_init: done
 1:icom.c(695):icom_init returning(0)
main: twiddle=0, uplink=0, twiddle_rit=0
 1:rig.c(743):rig_open entered
rig_open: cwd=C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5~git\bin
rig_open: C:\Users\garmi\OneDrive\Desktop\AntRunner-master\software\hamlib-w64-4.5~git\bin/.hamlib_settings does not exist
serial_open: COM1
serial_setup: tcgetattr
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1
  2:rig.c(7055):async_data_handler_start entered
async_data_handler_start: async data support disabled
  2:rig.c(7060):async_data_handler_start returning(0)
rig.c(241):add_opened_rig returning2(0)
rig_open: 0000000000F2AFFC rs->comm_state==1?=1
  2:icom.c(944):icom_rig_open entered
icom_rig_open: IC-2730 v20220326.0
   3:icom.c(748):icom_get_usb_echo_off entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 6 bytes, method=2
0000    fe fe 90 e0 03 fd                                   ......
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 6
read_string_generic(): RX 6 characters, direct=1
0000    fe fe 90 e0 03 fd                                   ......
icom_one_transaction: frm_len=6, frm_len-1=fd, frm_len-2=03
     5:frame.c(403):icom_one_transaction returning(0)
    4:frame.c(455):icom_transaction returning(0)
icom_get_usb_echo_off: ack_len=1
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 11
read_string_generic(): RX 11 characters, direct=1
0000    fe fe e0 90 03 00 00 62 45 01 fd                    .......bE..
icom_get_usb_echo_off: USB echo on detected, get freq retval=11
   3:icom.c(777):icom_get_usb_echo_off returning(0)
icom_rig_open: echo status result=0
icom_rig_open: echo status known, getting frequency
rig_get_freq(1837) called vfo=currVFO
vfo_fixup:(from rig_get_freq:1843) vfo=currVFO, vfo_curr=None, split=0
vfo_fixup: Leaving currVFO alone
rig.c(1846) vfo=currVFO, curr_vfo=None
rig_get_freq: cache miss age=1000000ms, cached_vfo=None, asked_vfo=None
rig_get_freq(1928): vfo_opt=0, model=3072
icom_get_freq called for None, curr_vfo=None
icom_get_freq: using vfo=None
   3:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x25, subcmd=0x00, payload_len=0
    4:frame.c(138):icom_one_transaction entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1
write_block(): TX 7 bytes, method=2
0000    fe fe 90 e0 25 00 fd                                ....%..
read_string_generic called, rxmax=200 direct=1
read_string_generic: minlen now 7
read_string_generic(): RX 7 characters, direct=1
0000    fe fe 90 e0 25 00 fd                                ....%..
icom_one_transaction: DEBUG retval=7, frm_len=7, cmd=0x25
read_string_generic called, rxmax=200 direct=1
read_string_generic(): RX 6 characters, direct=1
0000    fe fe e0 90 fa fd                                   ......
    4:frame.c(352):icom_one_transaction returning(-9) rigctld.c(626) Startup: rigctld -vvvvv -m 3072 -r COM1
rigctld Hamlib 4.5~git Tue May 17 22:03:48 2022 +0000 SHA=468dd1
Report bugs to <hamlib-developer@lists.sourceforge.net>

mdblack98, 2024-06-13:
Upgrade hamlib...you log shows the Hamlib version you have is using the 0x25 command which the 2730 does not support.
I don't recall fixing that but testing the latest Hamlib here doesn't use that command.

Download this https://n0nb.users.sourceforge.net/hamlib-w64-4.6~git-20240613-d160c6206.zip

Extract it and copy the bin directory to the AntRunner directory.

Mike W9DMB

Screenshot.2024-06-13.111712.jpg (view on web)

curtiswilliams, 2024-06-13:
After downloading, I am unable to access the radio. It looks like it is somehow still trying to use x25/x26 command

C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin>rigctld -vvvvvv -m 3072 -r COM1
rigctld.c(641) Startup: rigctld -vvvvvv -m 3072 -r COM1
rigctld Hamlib 4.6~git 2024-06-12T20:32:21Z SHA=d160c6 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

Max# of rigctld client services=32
rig_test_2038: enter _TIME_BITS=64 testing enabled for unknown libc
rig_test_2038: __MSVCRT_VERSION__=0x0700
rig_test_2038: MSVCRT 2038 test = 0xf0000000:Mon Aug 05 04:04:00 2097
rig_init: 2038 time test passed
rig_check_rig_caps: p1=0000000067fb1960, p2=0000000067fb7ee0, rig_model=0000000067fb1960, macro_name=0000000067fb7ee0
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-2730 20240511.0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
**2:icom.c(656):icom_init entered
icom_set_x25x26_ability: Hamlib thinks rig does not support x25/x26 command
icom_init: done
**2:icom.c(728):icom_init returning(0)
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='COM1'
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(946):rig_open entered
rig_settings_get_path: path=\Users\garmi/.hamlib_settings
rig_open: cwd=C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin
rig_open: C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin/hamlib_settings No such file or directory
rig_open: async_data_enable=0, async_data_supported=0
serial_open: COM1
serial_open(230): open failed#1 No such file or directory
serial_open: Unable to open COM1 - No such file or directory
port_open: serial_open(COM1) status=-6, err=No such file or directory
rig_open: rs->comm_state==0?=0
rig.c(1158):rig_open returning2(-6) IO error

Opened rig model 3072, 'IC-2730'
rig_open: error = icom_set_x25x26_ability: Hamlib thinks rig does not support x25/x26 command
icom_init: done
**2:icom.c(728):icom_init returning(0)
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='COM1'
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(946):rig_open entered
rig_settings_get_path: path=\Users\garmi/.hamlib_settings
rig_open: cwd=C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin
rig_open: C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin/hamlib_settings No such file or directory
rig_open: async_data_enable=0, async_data_supported=0
serial_open: COM1
serial_open(230): open failed#1 No such file or directory
serial_open: Unable to open COM1 - No such file or directory
port_open: serial_open(COM1) status=-6, err=No such file or directory
rig_open: rs->comm_state==0?=0
rig.c(1158):rig_open returning2(-6) IO error

IO error
 COM1 No such file or directory
Backend version: 20240511.0, Status: Beta
main: Using IPV6
main: rigctld listening on port 4532

curtiswilliams, 2024-06-13:
In GPRedict I have downlink set to 141.330.000 but if going down to .329 - .328 doesn't change the Icom 2730a but once it gets to .327 it changes to 141.325. If I change I change from 141.330.000 to 141.329.000 or 141.330.000 to 141.230.000 it changes to exactly the frequency it is changed to. So seems like I have an offset misconfigured, I'm a newbie to this radio and to Hamlib , GPRedict, any insight you might have would be greatly appreciated.

mdblack98, 2024-06-13:
I need a debug log.
Add this to rigctl
-vvvvv >log.txt 2>&1
Duplicate the problem and send me the log.
Mike W9MDB

curtiswilliams, 2024-06-14:
After spending time controlling the 2730a via Hamlib I figured out that it’s basically rounding up or down from its 5khz offset. It won’t go any lower than that, but which kinda makes it a struggle bus if you want to use for fine tuned Doppler adjustments.

So it’s the radio not the Hamlib.

But do you haven a link or advice on how I might control both VFOA and VFOB , particularly via GPredict? I can’t get my VFOB or Sub to change based on frequency changes using Gpredict radio control manual changes.

mdblack98, 2024-06-14:
Yes..seems like a common thing with UHF/VHF rigs.  If you use gpredict's A/B setup VFO B should change.  But perhaps not until transmit.

curtiswilliams, 2024-06-14:
So let’s say I have 10 satellites im interested in and the radio has enough memory slots to CHIRP program in multiple frequency for each satellite. Like ISS 1,2,3,5 to adjust for Doppler. Seems like I could right a python script to initiate a rolling command via hamlib that would just use memory slots vs the real time way GPredict does it. Like outside of any programs like GPredict, just have a drop down for ISS and click Enter and it runs a timer to roll through the memory for IsS?Yea I could turn a knob as it flies over but I’m a nerd and Hamlib is like a ball pit to jump into. Curtis Williams

mdblack98, 2024-06-14:
That' what gpredict does is to use Doppler to follow the satellite.
I don't see where memory slots make it any better unless your claiming memory slot have better then 5KHz resolution which I doubt.
You might be to able to set a frequency in the memory slot but once that radio reads it then freq should get rounded.

Mike W9MDB

curtiswilliams, 2024-06-14:
Your point has merit. So I did the A/B as you said but it won't change the uplink frequency.

Here is log closest I could get to not replicating, happy to send whole thing if needed:

rigctld.c(641) Startup: rigctld -vvvvvv -m 3072
rigctld Hamlib 4.6~git 2024-06-12T20:32:21Z SHA=d160c6 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

Max# of rigctld client services=32
rig_test_2038: enter _TIME_BITS=64 testing enabled for unknown libc
rig_test_2038: __MSVCRT_VERSION__=0x0700
rig_test_2038: MSVCRT 2038 test = 0xf0000000:Mon Aug 05 05:04:00 2097
rig_init: 2038 time test passed
rig_check_rig_caps: p1=0000000067fb1960, p2=0000000067fb7ee0, rig_model=0000000067fb1960, macro_name=0000000067fb7ee0
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-2730 20240511.0
rig_init: rig has VFO_SUB
rig_init: rig has VFO_MAIN
**2:icom.c(656):icom_init entered
icom_set_x25x26_ability: Hamlib thinks rig does not support x25/x26 command
icom_init: done
**2:icom.c(728):icom_init returning(0) 
main: twiddle=0, uplink=0, twiddle_rit=0
rig.c(946):rig_open entered
rig_settings_get_path: path=\Users\garmi/.hamlib_settings
rig_open: cwd=C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin
rig_open: C:\Users\garmi\OneDrive\Desktop\hamlib-w64-4.6~git-20240613-d160c6206\hamlib-w64-4.6~git\bin/hamlib_settings No such file or directory
rig_open: async_data_enable=0, async_data_supported=0
serial_open: \\.\COM1
serial_setup: tcgetattr
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: stopbits=1
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
**2:rig.c(8301):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
**2:rig.c(8308):async_data_handler_start returning(0) 
rig_open: 000000000260744c rs->comm_state==1?=1
**2:rig.c(6796):rig_get_powerstat entered
**rig.c(6810) trace
***3:icom.c(8107):icom_get_powerstat entered
***icom.c(8131) trace
***icom.c(8149) trace
****4:rig.c(2397):rig_get_freq entered
rig_get_freq called vfo=VFOA
rig_get_freq(2415) called vfo=VFOA
rig_get_freq(2417): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2417): freqMainB=0, modeMainB=, widthMainB=0
vfo_fixup:(from rig_get_freq:2422) vfo=VFOA, vfo_curr=currVFO, split=0
vfo_fixup: final vfo=Main
rig.c(2424) vfo=Main, curr_vfo=currVFO
rig_get_freq(2453): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2453): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: depth=4
cache.c(319):rig_get_cache entered
rig_get_cache(322):  vfo=Main, current_vfo=currVFO
rig_get_cache(517): vfo=Main, freq=0, mode=, width=0
cache.c(523):rig_get_cache returning2(0) 
rig_get_freq(2502): freqMainA=0, modeMainA=, widthMainA=0
rig_get_freq(2502): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq: cache miss age=1000000ms, cached_vfo=Main, asked_vfo=Main, use_cached_freq=0
rig_get_freq(2537): vfo_opt=0, model=3072
****rig.c(2585) trace
icom_get_freq called for Main, curr_vfo=currVFO
*****5:icom.c(776):icom_get_usb_echo_off entered
******6:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
[... 168 lines cut ...]
icom.c(2965):icom_set_vfo returning2(0) 
rig_get_freq(2620): freqMainA=437205000, modeMainA=, widthMainA=0
rig_get_freq(2620): freqMainB=0, modeMainB=, widthMainB=0
rig_set_cache_freq(163): freqMainA=437205000, modeMainA=, widthMainA=0
rig_set_cache_freq(163): freqMainB=0, modeMainB=, widthMainB=0
rig_set_cache_freq(166):  vfo=Main, current_vfo=Main
rig_set_cache_freq(186): set vfo=Main to freq=437205000
rig_set_cache_freq(278): freqMainA=437205000, modeMainA=, widthMainA=0
rig_set_cache_freq(278): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq(2622): freqMainA=437205000, modeMainA=, widthMainA=0
rig_get_freq(2622): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq(2647): freqMainA=437205000, modeMainA=, widthMainA=0
rig_get_freq(2647): freqMainB=0, modeMainB=, widthMainB=0
rig_set_cache_freq(163): freqMainA=437205000, modeMainA=, widthMainA=0
rig_set_cache_freq(163): freqMainB=0, modeMainB=, widthMainB=0
rig_set_cache_freq(166):  vfo=Main, current_vfo=Main
rig_set_cache_freq(186): set vfo=Main to freq=437205000
rig_set_cache_freq(278): freqMainA=437205000, modeMainA=, widthMainA=0
rig_set_cache_freq(278): freqMainB=0, modeMainB=, widthMainB=0
rig_get_freq(2654): freqMainA=437205000, modeMainA=, widthMainA=0
rig_get_freq(2654): freqMainB=0, modeMainB=, widthMainB=0
rig_get_band called
****4:rig_get_freq: elapsed=138ms

[image]

mdblack98, 2024-06-15:
Try the Main Sub options.  

Hamlib.jpg (view on web)

curtiswilliams, 2024-06-15:
The RX looks to be shifting every 5khz, as the limitation of the Icom2730a minimum restriction stipulates. But I can't get any of the options in GPRedict Interfaces to make changes to the TX. It is a dual band so I'd assume it would let you control Main/Sub or VFOA/VFOB , particularly since the .bat file is showing it is finding both frequencies.

Any ideas on what I could possibly be doing wrong or have misconfigured? I know some satellites don't even need doppler on the transmit, but would be nice to get GPredict to set the TX frequency for the current satellite.

My project has a DIY Rotator that works with GPredict/Hamlib/GrblTrk and I have built a battery box that holds the radio body, small PC , and all the electrical stuff. My goal would be to get this part automated and I'd have a pretty cool self-contained satellite box I could travel with. 

I'm attaching logs for all the different options available and screen shots. Hoping I've done something in error and I can get this radio to receive both the RX and TX.

Thanks for all the help!

[duplex_MainUp_Sub_Down_log.txt](https://github.com/user-attachments/files/15848521/duplex_MainUp_Sub_Down_log.txt)
[image]
[image]
[duplex_SubUp_MainDown_log.txt](https://github.com/user-attachments/files/15848522/duplex_SubUp_MainDown_log.txt)
[duplex_VFOADown_VFOBUP_log.txt](https://github.com/user-attachments/files/15848523/duplex_VFOADown_VFOBUP_log.txt)
[image]
[duplex_VFOAUp_VFODown_log.txt](https://github.com/user-attachments/files/15848524/duplex_VFOAUp_VFODown_log.txt)
[image]
[rx_only_log.txt](https://github.com/user-attachments/files/15848525/rx_only_log.txt)
[simplex_log.txt](https://github.com/user-attachments/files/15848526/simplex_log.txt)
[image]
[trx_only_log.txt](https://github.com/user-attachments/files/15848527/trx_only_log.txt)

mdblack98, 2024-06-15:
Which satellite are you trying?

Duplex_MainUp_SubDown.jpg (view on web)
Duplex_SubUp_MainDown.jpg (view on web)
Duplex_VFOAUP_VFODOWN_log.jpg (view on web)
Duplex_VFODOWN_VFOUP.jpg (view on web)
SimplexTRX_ReadPTT_otapplicable.jpg (view on web)

mdblack98, 2024-06-15:
Are you enabling "Track" in the Radio Control?  I just tested with my 2730 simulator here and it's handing Main & Sub.
You can see this output like this:
rigctld -m 3072 -r /dev/ttyS0 -vvvvv 2>&1 | grep "set_freq called vfo"

rig_set_freq called vfo=Sub, freq=145887542
rig_set_freq called vfo=currVFO, freq=145892455
rig_set_freq called vfo=Sub, freq=145887545
rig_set_freq called vfo=currVFO, freq=145892452
rig_set_freq called vfo=Sub, freq=145887548
rig_set_freq called vfo=currVFO, freq=145892450
rig_set_freq called vfo=Sub, freq=145887550
rig_set_freq called vfo=currVFO, freq=145892444
rig_set_freq called vfo=Sub, freq=145887556
rig_set_freq called vfo=currVFO, freq=145892442
rig_set_freq called vfo=Sub, freq=145887558
rig_set_freq called vfo=currVFO, freq=145892439
rig_set_freq called vfo=Sub, freq=145887561
rig_set_freq called vfo=currVFO, freq=145892436
rig_set_freq called vfo=Sub, freq=145887564
rig_set_freq called vfo=currVFO, freq=145892433
rig_set_freq called vfo=Sub, freq=145887567

rig_set_freq called vfo=currVFO, freq=145892431

rig_set_freq called vfo=Sub, freq=145887569

rig_set_freq called vfo=currVFO, freq=145892428

com_set_freq called Main=145892728
icom_set_freq: set_vfo_curr=Main
icom.c(1673):icom_set_freq returning2(0) 
rig_set_freq: Asked freq=145892728, got freq=145892736
*****5:rig_set_freq: elapsed=1ms
*****5:rig.c(2353):rig_set_freq returning(0) 
rigctl_parse.c(2150):rigctl_set_freq returning2(0) 
*****5:rig.c(2072):rig_set_freq entered
rig_set_freq called vfo=Sub, freq=145887272
vfo_fixup:(from rig_set_freq:2149) vfo=Sub, vfo_curr=Sub, split=1
skip_freq: not skipping set_freq on vfo Sub
rig_set_freq: TARGETABLE_FREQ vfo=Sub
icom.c(1511):icom_set_freq entered
icom_set_freq called Sub=145887272
icom_set_freq: set_vfo_curr=Sub
icom.c(1673):icom_set_freq returning2(0) 

rig_set_freq: Asked freq=145887272, got freq=145887264

*****5:rig_set_freq: elapsed=2ms

curtiswilliams, 2024-06-15:
Those logs were from a pass of Tevel sats. Want to have one actively passing.

mdblack98, 2024-06-15:
I don't see Tevel sats in the list -- but I just tested with one that had Doppler going on and it works as long as "Track" is enabled in Radio Control.

Mike W9MDB

curtiswilliams, 2024-06-15:
It def works with RX , even with the Tevel. But with tx nothing. I was able to change both with rigctl and it does show the two frequencies in my .bat file when it runs. 

Let me replicate the rigctl, copy the data from ricctld .bat and see what you think. 

I truly appreciate the help. I’m definitely wrapped around the axle at this point but too stubborn to give up.

curtiswilliams, 2024-06-18:
So here is a rigtctl attempt. Both the VFOA and VFOB or rather SubA and MainA. So how is it chaning the frequencies and not rigctld for Gpredict?

The only way to get Sub/Main or VFOA/VFOB is to set the radio type to Duplex but it isn't a Duplex, it's a Dual Band. 

Any thoughts?

rigctl -m 3072

Rig command: V SubA

Rig command: f
Frequency: 143145000

Rig command: V MainA
set_vfo: error = icom.c(3248):icom_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig
'
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=SubA
*****5:rig.c(3394):rig_get_vfo entered
rig_get_vfo: no get_vfo
*****5:rig_get_vfo: elapsed=0ms
*****5:rig.c(3404):rig_get_vfo returning(-11) Feature not available

rig_set_vfo: Expiring all cache due to VFO change and no get_vfo
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=SubA
rig_set_vfo: return -9, vfo=MainA, curr_vfo=SubA
****4:rig_set_vfo: elapsed=37ms
****4:rig.c(3361):rig_set_vfo returning(-9) Command rejected by the rig

rigctl_set_vfo: set_vfo(MainA) failed, requested MainA
rigctl_parse.c(2380):rigctl_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig

Rig command: f
Frequency: 142290000

Rig command: F 144000000

Rig command: V SubA

Rig command: F 144144000

mdblack98, 2024-06-18:
Were you in duplex when you did that test?
Please send me log.txt from this:

rigctl -m 3072 V SubA f V MainA >log.txt 2>&1

curtiswilliams, 2024-06-18:
[Uploading log.txt…]()

curtiswilliams, 2024-06-18:
145143750
set_vfo: error = icom.c(3248):icom_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig
'
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=SubA
*****5:rig.c(3394):rig_get_vfo entered
rig_get_vfo: no get_vfo
*****5:rig_get_vfo: elapsed=0ms
*****5:rig.c(3404):rig_get_vfo returning(-11) Feature not available

rig_set_vfo: Expiring all cache due to VFO change and no get_vfo
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=SubA
rig_set_vfo: return -9, vfo=MainA, curr_vfo=SubA
****4:rig_set_vfo: elapsed=41ms
****4:rig.c(3361):rig_set_vfo returning(-9) Command rejected by the rig

rigctl_set_vfo: set_vfo(MainA) failed, requested MainA
rigctl_parse.c(2380):rigctl_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig

curtiswilliams, 2024-06-18:
set_vfo: error = icom.c(3248):icom_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig
'
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=Main
*****5:rig.c(3394):rig_get_vfo entered
rig_get_vfo: no get_vfo
*****5:rig_get_vfo: elapsed=0ms
*****5:rig.c(3404):rig_get_vfo returning(-11) Feature not available

rig_set_vfo: Expiring all cache due to VFO change and no get_vfo
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=Main
rig_set_vfo: return -9, vfo=SubB, curr_vfo=Main
****4:rig_set_vfo: elapsed=37ms
****4:rig.c(3361):rig_set_vfo returning(-9) Command rejected by the rig

rigctl_set_vfo: set_vfo(SubB) failed, requested SubB
rigctl_parse.c(2380):rigctl_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig

145143750
set_vfo: error = icom.c(3248):icom_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig
'
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=Main
*****5:rig.c(3394):rig_get_vfo entered
rig_get_vfo: no get_vfo
*****5:rig_get_vfo: elapsed=0ms
*****5:rig.c(3404):rig_get_vfo returning(-11) Feature not available

rig_set_vfo: Expiring all cache due to VFO change and no get_vfo
rig_set_cache_freq(167):  vfo=AllVFOs, current_vfo=Main
rig_set_vfo: return -9, vfo=MainA, curr_vfo=Main
****4:rig_set_vfo: elapsed=37ms
****4:rig.c(3361):rig_set_vfo returning(-9) Command rejected by the rig

rigctl_set_vfo: set_vfo(MainA) failed, requested MainA
rigctl_parse.c(2380):rigctl_set_vfo returning2(-9) Command rejected by the rig

Command rejected by the rig

mdblack98, 2024-06-18:
Where's the log.txt file?

rigctl -m 3072 V VFOB f V VFOA >log.txt 2>&1

145143750

curtiswilliams, 2024-06-19:
That was all the log data that came from it. You want RIGCTL or RIGCTLD?

curtiswilliams, 2024-06-19:
Just an update. Using the RIGCTL below I can make changes to the Main and Sub VFO's. So the functionality is there , the RIGCTLD just isn't performing correctly.

@echo off
rigctl -vvvvv -m 3072 -r COM1 -s 9600 --vfo
pause

These all work:

F Sub 435000000

F Main 433000000

f Sub

f Main

mdblack98, 2024-06-21:
The available Windows versions of gpredict do not support the --vfo option.

I've pretty much given up on anybody compiling the new version of gpredict for Windows.  No issues have been fixed in years.

Mike W9MDB

curtiswilliams, 2024-06-22:
You just need someone to compile the executable with the updated changes?

mdblack98, 2024-06-22:
Yes.

mdblack98, 2024-06-28:
Here is a 64-bit 2.3.38 version that 
Run rigctld with the "--vfo" option and see how it behaves.
https://www.dropbox.com/scl/fi/m9fnryi5whblj5e5p7h7u/gpredict-2.3.38_64.zip?rlkey=10tl9g8gux302iotf4kdlkvpc&dl=1

curtiswilliams, 2024-06-28:
Thanks for the builds. How do new radios get added? Does a developer with a radio tweak an existing radio model code and see if they can manage to get it to work or when you get a new radio and have time to mess with it?

https://www.dropbox.com/scl/fi/m9fnryi5whblj5e5p7h7u/gpredict-2.3.38_64.zip?rlkey=10tl9g8gux302iotf4kdlkvpc&dl=1

mdblack98, 2024-06-28:
Depends -- some rigs are only slightly different than others.  Some completely new.So it depends on how much code is common.I've added some new rigs in a matter of minutes -- others that are unique can take days.Some rigs are added blind but I always ask for testers so they can get promoted from Alpha->Beta->Stable.The alpha versions are ones written and not tested....Beta is in testing...Stable at this point means basic functions work (might still need more features added).  So frequency, mode, ptt are expected to be stable.
Mike W9MDB

 
 

https://www.dropbox.com/scl/fi/m9fnryi5whblj5e5p7h7u/gpredict-2.3.38_64.zip?rlkey=10tl9g8gux302iotf4kdlkvpc&dl=1

curtiswilliams, 2024-06-28:
So how do I help with this icom 2730a I have or the new Alinco DR735 I picked up (testing with a kenwood model number)? Just take notes and submit to issues or something more proper?Curtis Williams

 

 

 

https://www.dropbox.com/scl/fi/m9fnryi5whblj5e5p7h7u/gpredict-2.3.38_64.zip?rlkey=10tl9g8gux302iotf4kdlkvpc&dl=1

mdblack98, 2024-06-28:
Have you tested the new build with and used --vfo for rigctld?
I need to see a debug log from rigctld and for you to describe the behavior you see.

rigctl -vvvvv -m 3072 -r COM1 -s 9600 --vfo >log.txt 2>&1

mdblack98, 2024-06-28:
Or contact me offline and let's hook up using AnyDesk and debug this.
mdblack98@yahoo.colm

closed 2024-08-18 by mdblack98