Hamlib/Hamlib #1569: Upgraded IC-705 to firmware 1.32 getting communication timeout

jonlundy, 2024-06-23:
Using Hamlib 4.5.5-6 with Fedora 40.
I tried reverting firmware to 1.26 but get the same errors. I have been successful in using WFview which does not use hamlib to communicate via C-IV.

```
 rigctl -m 3085 -r /dev/ic-705a -s 115200 -vvvv
rigctl.c(445) Startup: rigctl -m 3085 -r /dev/ic-705a -s 115200 -vvvv
rigctl Hamlib 4.5.5 Apr 05 11:43:08Z 2023 SHA=6eecd3
Report bugs to <hamlib-developer@lists.sourceforge.net>

initrigs4_icom: _init called
rig_init: rig_model=Icom IC-705
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
 1:icom.c(623):icom_init entered
 1:icom.c(724):icom_init returning(0)
 1:rig.c(817):rig_open entered
rig_settings_load_all: settings_file (/home/xuu/.config/hamlib_settings): No such file or directory
rig_open: cwd=/home/xuu/Downloads
rig_open: /home/xuu/Downloads/hamlib_settings does not exist
create_sync_data_pipe: created data pipe for synchronous transactions
serial_open: /dev/ic-705a
tcflush
  2:rig.c(7481):async_data_handler_start entered
  2:rig.c(7517):async_data_handler_start returning(0)
rig.c(254):add_opened_rig returning2(0)
rig_open: 0x556f0a7f00dc rs->comm_state==1?=1
  2:rig.c(6084):rig_get_powerstat entered
   3:icom.c(8103):icom_get_powerstat entered
async_data_handler: Starting async data handler thread
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x18, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
async_data_handler: received frame: len=6 async=0
read_string_generic(): Timed out 1.000 seconds after 0 chars, direct=1
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=0
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

icom_transaction: retry=1:    3:icom.c(8103):icom_get_powerstat entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x18, subcmd=0xffffffff, payload_len=0
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
     5:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
read_string_generic called, rxmax=200 direct=1, expected_len=1
icom_one_transaction: DEBUG retval=6, frm_len=6, cmd=0x18
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): Timed out 1.000 seconds after 0 chars, direct=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=0
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out
Communication timed out

     5:frame.c(138):icom_one_transaction entered
async_data_handler: received frame: len=6 async=0
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=1
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=0
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

icom_transaction: retry=0: read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=0
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out

     5:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
read_string_generic called, rxmax=200 direct=1, expected_len=1
icom_one_transaction: DEBUG retval=6, frm_len=6, cmd=0x18
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out
Communication timed out

icom_transaction: failed:
Communication timed out

     5:frame.c(138):icom_one_transaction entered
write_block(): TX 6 bytes, method=2
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): RX 6 characters, direct=1
async_data_handler: received frame: len=6 async=0
read_string_generic called, rxmax=200 direct=1, expected_len=1
icom_one_transaction: DEBUG retval=6, frm_len=6, cmd=0x18
read_string_generic called, rxmax=200 direct=0, expected_len=1
read_string_generic(): Timed out 1.001 seconds after 0 chars, direct=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
     5:frame.c(294):icom_one_transaction returning(-5) Communication timed out

Communication timed out

Communication timed out
Communication timed out

    4:frame.c(455):icom_transaction returning(-5) Communication timed out

   3:icom.c(8144):icom_get_powerstat returning(-5) Communication timed out

  2:rig.c(6109):rig_get_powerstat returning(-5) Communication timed out

rig_open: Some rigs cannot get_powerstat while off
rig_open: Known rigs: K3, K3S
rig_open: Rigs that should but don't work: TS480
  2:icom.c(987):icom_rig_open entered
icom_rig_open: IC-705 v20230109.8
   3:icom.c(777):icom_get_usb_echo_off entered
    4:frame.c(425):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
     5:frame.c(138):icom_one_transaction entered
async_data_handler: received frame: len=6 async=0
     5:frame.c(403):icom_one_transaction returning(0)
    4:frame.c(455):icom_transaction returning(0)
icom_get_usb_echo_off: ack_len=1

```

jonlundy, 2024-06-23:
[image]
[image]

mdblack98, 2024-06-23:
Upgrade to the latest Hamlib -- pretty sure this has been fixed.

#1 Remove any Hamlib package you have installed
#2 Download the latest tar.gz file from here: https://n0nb.users.sourceforge.net/
#3 tar xzf hamlib*tar.gz
#4 cd hamlib-4.6~git
#5 ./configure
#6 make install

This will install in /usr/local by default.

Mike W9MDB

jonlundy, 2024-06-24:
I tried the latest 4.6 hamlib. i seem to get a different error. 

```
 rigctl -m 3085 -r /dev/ic-705a -vvvv
rigctl.c(531) Startup: rigctl -m 3085 -r /dev/ic-705a -vvvv
rigctl Hamlib 4.6~git 2024-06-21T21:58:19Z SHA=097f2b 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_test_2038: time_t 2038 test = 0xf0000000:Mon Aug  5 03:04:00 2097
rig_init: 2038 time test passed
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-705 20240511.11
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
**2:icom.c(656):icom_init entered
**2:icom.c(728):icom_init returning(0) 
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='/dev/ic-705a'
rig.c(932):rig_open entered
rig_open: cwd=/home/xuu
rig_open: /home/xuu/hamlib_settings No such file or directory
serial_open: /dev/ic-705a
**2:rig.c(8287):async_data_handler_start entered
**2:rig.c(8294):async_data_handler_start returning(0) 
rig_open: 0x1c944ed4 rs->comm_state==1?=1
**2:icom.c(1105):icom_rig_open entered
icom_rig_open: IC-705 v20240511.11
***3:icom.c(776):icom_get_usb_echo_off entered
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
*****5:frame.c(455):icom_one_transaction returning(0) 
****4:frame.c(508):icom_transaction returning(0) 
icom_get_usb_echo_off: ack_len=1
icom_get_usb_echo_off: USB echo on detected, get freq retval=-5
***3:icom.c(798):icom_get_usb_echo_off returning(-5) Communication timed out

***3:icom.c(776):icom_get_usb_echo_off entered
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
*****5:frame.c(455):icom_one_transaction returning(0) 
****4:frame.c(508):icom_transaction returning(0) 
icom_get_usb_echo_off: ack_len=1
icom_get_usb_echo_off: USB echo on detected, get freq retval=-5
***3:icom.c(798):icom_get_usb_echo_off returning(-5) Communication timed out

**2:icom.c(1240):icom_rig_open returning(0) 
**2:rig.c(8331):morse_data_handler_start entered
***3:icom.c(3991):icom_get_level entered
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x14, subcmd=0x0c, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
icom_transaction: retry=0: icom_rig_open: echo status result=-5
icom_rig_open: echo status unknown
**2:icom.c(1240):icom_rig_open returning(0) 
rig_open: vfo_curr=currVFO, tx_vfo=TX
rig_open: default vfo = currVFO
**2:rig.c(8331):morse_data_handler_start entered
***3:icom.c(3991):icom_get_level entered
icom_get_level: no extcmd found
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x14, subcmd=0x0c, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
write_block(): TX 7 bytes
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 7 characters, direct=1
icom_one_transaction: DEBUG retval=7, frm_len=7, cmd=0x14
[... 119 lines cut ...]
scanfc: errno=9, Bad file descriptor
Command '' not found!
***3:rig.c(1639):rig_close entered
****4:rig.c(8413):morse_data_handler_stop entered
****4:rig.c(8459):morse_data_handler_stop returning(0) 
****4:rig.c(8372):async_data_handler_stop entered
****4:rig.c(8403):async_data_handler_stop returning(0) 
****4:event.c(327):rig_poll_routine_stop entered
event.c(253): Stopping rig poll routine thread
****4:event.c(362):rig_poll_routine_stop returning(0) 
****4:network.c(1903):network_multicast_receiver_stop entered
multicast_receiver(1495): pselect signal
network.c(1556): Stopped multicast receiver
****4:network.c(1948):network_multicast_receiver_stop returning(0) 
****4:network.c(1735):network_multicast_publisher_stop entered
network.c(1103): Stopped multicast publisher
****4:network.c(1774):network_multicast_publisher_stop returning(0) 
****4:icom.c(1252):icom_rig_close entered
****4:icom.c(1254):icom_rig_close returning(0) 
ser_close: restoring options
rig_close(1783): 0x1c944ed4 rs->comm_state==0?=0
***3:rig.c(1787):rig_close returning(0) 
***3:icom.c(740):icom_cleanup entered
***3:icom.c(760):icom_cleanup returning(0) 
```

mdblack98, 2024-06-24:
And one more 'v' please.

jonlundy, 2024-06-25:
```
 rigctl -m 3085 -r /dev/ic-705a -vvvvv
rigctl.c(531) Startup: rigctl -m 3085 -r /dev/ic-705a -vvvvv
rigctl Hamlib 4.6~git 2024-06-21T21:58:19Z SHA=097f2b 64-bit
Report bugs to <hamlib-developer@lists.sourceforge.net>

rig_test_2038: enter _TIME_BITS=64, __TIMESIZE=64 testing enabled for GLIBC 2.39
rig_test_2038: time_t 2038 test = 0xf0000000:Mon Aug  5 03:04:00 2097
rig_init: 2038 time test passed
rig_check_rig_caps: p1=0x7f3953f1dc80, p2=0x7f3953f24208, rig_model=0x7f3953f1dc80, macro_name=0x7f3953f24208
initrigs4_icom: _init called
rig_init: rig_model=Icom IC-705 20240511.11
rig_init: rig has VFO_A
rig_init: rig has VFO_B
rig_init: rig has VFO_MEM
**2:icom.c(656):icom_init entered
icom_init: done
**2:icom.c(728):icom_init returning(0) 
rig_set_conf called
rig_confparam_lookup called for 1073741834
rig_set_conf: rig_pathname='/dev/ic-705a'
rig.c(932):rig_open entered
rig_settings_get_path: path=/home/xuu/.config/hamlib_settings
rig_open: cwd=/home/xuu
rig_open: /home/xuu/hamlib_settings No such file or directory
rig_open: async_data_enable=0, async_data_supported=1
serial_open: /dev/ic-705a
serial_setup: tcgetattr
serial_setup: cfmakeraw
serial_setup: cfsetispeed=19200,0x000e
serial_setup: cfsetospeed=19200,0x000e
serial_setup: data_bits=8
serial_setup: stopbits=1
serial_setup: parity=0
serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
**2:rig.c(8287):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
**2:rig.c(8294):async_data_handler_start returning(0) 
rig_open: 0x302cbed4 rs->comm_state==1?=1
**2:icom.c(1105):icom_rig_open entered
icom_rig_open: IC-705 v20240511.11
***3:icom.c(776):icom_get_usb_echo_off entered
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
write_block(): TX 6 bytes
0000    fe fe a4 e0 03 fd                                   ......          
icom_transaction: retry=0: serial_setup: Handshake=None
serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1, expected_len=1
**2:rig.c(8287):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
**2:rig.c(8294):async_data_handler_start returning(0) 
rig_open: 0x302cbed4 rs->comm_state==1?=1
**2:icom.c(1105):icom_rig_open entered
icom_rig_open: IC-705 v20240511.11
***3:icom.c(776):icom_get_usb_echo_off entered
****4:frame.c(477):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
write_block(): TX 6 bytes
0000    fe fe a4 e0 03 fd                                   ......          
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(1356): retrying read timeout 1/1 timeout=1000
read_string_generic(): Timed out 2.011 seconds after 0 chars, direct=1
Communication timed out
Communication timed out

icom_transaction: failed: read_string_generic called, rxmax=4095 direct=1, expected_len=1
[... 234 lines cut ...]
rigctl_parse: cmd= (00) handle=-1
Command '' not found!
*****5:rig.c(1639):rig_close entered
******6:rig.c(8413):morse_data_handler_stop entered
******6:rig.c(8459):morse_data_handler_stop returning(0) 
******6:rig.c(8372):async_data_handler_stop entered
******6:rig.c(8403):async_data_handler_stop returning(0) 
******6:event.c(327):rig_poll_routine_stop entered
event.c(253): Stopping rig poll routine thread
******6:event.c(362):rig_poll_routine_stop returning(0) 
******6:network.c(1903):network_multicast_receiver_stop entered
multicast_receiver(1495): pselect signal
network.c(1556): Stopped multicast receiver
******6:network.c(1948):network_multicast_receiver_stop returning(0) 
******6:network.c(1735):network_multicast_publisher_stop entered
network.c(1103): Stopped multicast publisher
******6:network.c(1774):network_multicast_publisher_stop returning(0) 
******6:icom.c(1252):icom_rig_close entered
******6:icom.c(1254):icom_rig_close returning(0) 
ser_close: restoring options
rig_close(1783): 0x302cbed4 rs->comm_state==0?=0
*****5:rig.c(1787):rig_close returning(0) 
*****5:icom.c(740):icom_cleanup entered
*****5:icom.c(760):icom_cleanup returning(0) 
```

mdblack98, 2024-06-25:
I'm guessing the baud rate is wrong.

You need to check the baud rate on the rig since you updated the firmware.

Hamlib is defaulting to 19,200

Mike W9MDB

jonlundy, 2024-06-27:
So i figured out the issue with connecting. After trying every baud rate i tried loading a profile from the menu and it seems to be working now. I cant even find anything about that button in the manuals.

closed 2024-06-27 by jonlundy