Hamlib/Hamlib #1574: Hamlib error for "TX" command TS2000

scottlbaker, 2024-07-04:
I have a home brew radio that emulates TS2000 cat commands.
I am testing it with wsjtx
The "Test CAT" button passes (turns green)
The "Test PTT" fails with the following error message:

Hamlib error: read_string_generic called, rxmax=4095 direct=1, expected_len=1
write_block(): TX 3 bytes, method=2
0000    54 58 3b                                            TX;             
write_block(): TX 3 bytes, method=2
0000    49 44 3b                                            ID;             
read_string_generic called, rxmax=51 direct=1, expected_len=1
read_string_generic(): RX 3 characters, direct=1
0000    54 58 3b                                            TX;             
kenwood_transaction: read_string(expected=51, len=3)='TX;'
kenwood_transaction: No data expected, checking ID; in TX;
kenwood_transaction: wrong reply TX for command verification ID
kenwood.c(624):kenwood_transaction returning2(-8) Protocol error
 2:kenwood.c(4670):kenwood_set_ptt returning(-8) Protocol error
 1:rig_set_ptt: elapsed=30ms
 1:rig.c(3013):rig_set_ptt returning(-8) Protocol error
Protocol error while setting PTT on

What is Hamlib expecting?

Thanks and 73,
Scott

mdblack98, 2024-07-04:
It's expecting an "ID019;" response to the "ID;" command.
It's a way to confirm the rig is still talking for all the commands that have no response.

Mike W9MDB

scottlbaker, 2024-07-04:
Thanks. It's working now. This issue can be closed.

73,
Scott