Hamlib/Hamlib #1576: FT-950 VFOB toggling RX VFO for every command

nicheath, 2024-07-06:
I am executing rigctld 4.5.5 with command:
rigctld.exe --vfo -vvvvv -m 1028 -r COM9 -s 19200

When I send **any** command to my rig e.g. "rigctl.exe -m 2 -r 127.0.0.1:4532 f" when i am using VFO B it causes VFO B RX to toggle to VFO A and back to VFO B.

I discovered this because I was trying out Log4OM and I use VFO B for digital modes but connecting to hamlib from log4om while in vfo B caused my rig to toggle RX between VFO B and VFO A constantly.

I have attached a -vvvv capture of rigctld and this command being sent to it.
[trace.txt](https://github.com/user-attachments/files/16116311/trace.txt)

I see in the trace.txt file there is a command sent
FT1;
Looking at the CAT documentation for the FT-950 it looks like that could cause the toggle.

mdblack98, 2024-07-06:
Try this instead -- with the --vfo option many commands use a VFO argument to avoid VFO swapping.

rigctl.exe -m 2 -r 127.0.0.1:4532 f VFOB

And I would also recommend you use the rigctld that comes with Log4OM -- it's a more recent version of Hamlib.

\Users\[username]\AppData\Roaming\Log4OM2\hamlib\rigctld.exe

Mike W9MDB

nicheath, 2024-07-06:
Thanks for the reply. I have tried hamlib that is shipped with log4om and it has the same behavior. I edited my comment to note that this will vfo swap will happen when rigctrld starts up too.

nicheath, 2024-07-06:
I just verified compiling latest on master branch also has this issue even when just starting rigctld.

mdblack98, 2024-07-06:
You need the VFO argument to avoid the VFO swapping.

There might be a problem with mode though since mode is not targetable for the FT-950.

Give me a debug log for each

rigctld.exe --vfo -vvvvv -m 1028 -r COM9 -s 19200 >log1.txt 2>&1
rigctl.exe -m 2 -r 127.0.0.1:4532 f VFOB >log2.txt 2>&1

Send me both log files.

Mike W9MDB

nicheath, 2024-07-06:
Here are log1.txt and log2.txt as specified.
[log1.txt](https://github.com/user-attachments/files/16116474/log1.txt)
[log2.txt](https://github.com/user-attachments/files/16116476/log2.txt)

Note this toggle also occurs when rigctld starts up.

nicheath, 2024-07-06:
hamlib does report the correct frequency for both

rigctl.exe -m 2 -r 127.0.0.1:4532 f VFOB

and

rigctl.exe -m 2 -r 127.0.0.1:4532 f

when VFOB is selected.

nicheath, 2024-07-06:
I get unknown option for both rigctld and rigctl.

I see in the master branch skipinit in the options but maybe I need to compile a version specially for that?

mdblack98, 2024-07-06:
That was added back in Jan 2024 -- only for rigctl

What version rigctl are you running?
rigctl --version

commit 9b817f325df21e52416416aad431f50c6e9d770a
Author: Mike Black W9MDB ***@***.***>
Date:   Sat Jan 13 08:26:44 2024 -0600

    Add --skipinit to rigctl.c -- can speed up single command-line invocations by quite a bit

nicheath, 2024-07-06:
I am running 4.5.5 but I have bleeding edge from master branch compiled in a linux VM and I am testing stuff. I am trying to track down which cat command is actually doing the switch and trace that back to where it is getting sent.

mdblack98, 2024-07-06:
Normal startup of rigtctl queries mode from both VFOs.  It's the mode request that has to change VFOs to get the "other" VFO mode.  It's a deficiency in the FT-950 commands that the "MD" command is not VFO-specific.

The --skipinit is available in the latest packages here
https://n0nb.users.sourceforge.net/
https://n0nb.users.sourceforge.net/hamlib-w32-4.6~git-20240706-6ed7cfe73.zip

Mike W9MDB

nicheath, 2024-07-06:
newcat_get_vfo_mode: offset=21, width_frequency=8
newcat_get_vfo_mode: vfo mode = VFO
    4:newcat.c(10438):newcat_get_vfo_mode returning(0) 
cmd_str = VS0;
    4:newcat.c(10911):newcat_set_cmd entered
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
tcflush
cmd_str = VS0;
     5:newcat.c(10728):newcat_set_cmd_validate entered
newcat_set_cmd_validate: priv->cmd_str=VS0;
rig_flush: called for serial device
read_string_generic called, rxmax=4095 direct=1, expected_len=1
tcflush
write_block(): TX 4 bytes, method=2
0000    56 53 30 3b                                         VS0;

If i send VS0 CAT command to my radio while it is in VFOB mode.. that toggles RX to be VFOA

so I guess just trying to figure out where that happens in hamlib to see what is going on.

mdblack98, 2024-07-06:
Yes VS0; selects VFOA.

If you just use the more current version of hamlib the --skipinit should solve your problem.

nicheath, 2025-06-16:
I've done quite a bit of testing with this trying to narrow it down and I think I have pinpointed the issue to being with Log4OM and not hamlib. I have tried several other tools that use rigctl and use VFOB with zero problems. I even scripted out my own control utility for making changes quickly to my rig and it all works fine.