Hamlib/Hamlib #1582: tried compile, got thru all the rigs then choked

wb2dyb, 2024-07-12:
Did I get something wrong?? Here's where it blows up:

_In file included from misc.c:48:0:
misc.c: In function 'rig_test_2038':
misc.c:3054:37: error: '__TIMESIZE' undeclared (first use in this function)
               __func__, _TIME_BITS, __TIMESIZE, __GLIBC__, __GLIBC_MINOR__);
                                     ^
../include/hamlib/rig.h:3760:94: note: in definition of macro 'rig_debug'
 #define rig_debug(debug_level,fmt,...) do { snprintf(debugmsgsave2,sizeof(debugmsgsave2),fmt,__VA_ARGS__);rig_debug(debug_level,fmt
                                                                                              ^
misc.c:3054:37: note: each undeclared identifier is reported only once for each function it appears in
               __func__, _TIME_BITS, __TIMESIZE, __GLIBC__, __GLIBC_MINOR__);
                                     ^
../include/hamlib/rig.h:3760:94: note: in definition of macro 'rig_debug'
 #define rig_debug(debug_level,fmt,...) do { snprintf(debugmsgsave2,sizeof(debugmsgsave2),fmt,__VA_ARGS__);rig_debug(debug_level,fmt
                                                                                              ^
Makefile:604: recipe for target 'misc.lo' failed
make[2]: *** [misc.lo] Error 1
make[2]: Leaving directory '/usr/src/hamlib4r5/hamlib455-mot/Hamlib-master/src'
Makefile:473: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/hamlib4r5/hamlib455-mot/Hamlib-master/src'
Makefile:626: recipe for target 'all-recursive' failed
make: *** [all-recursive] Error 1_

GeorgeC W2DB

mdblack98, 2024-07-12:
What OS is this on?

Mike W9MDB

wb2dyb, 2024-07-12:
It’s on hamvoip which is Arch LinuxGeorgeC Sent from my iPhone

mdblack98, 2024-07-12:
Hopefully the patch I put in let's it compile for you now.

Mike W9MDB

wb2dyb, 2024-07-12:
Thanks Mike, I’ll give it a shotGCSent from my iPhone

wb2dyb, 2024-07-12:
It worked, thanks MikeGeorgeC Sent from my iPhone

closed 2024-07-12 by mdblack98