Hamlib/Hamlib #1593: IC-751 not working

mac1-131, 2024-08-09:
Maybe I should post this issue to WSJT-X first, but I am pretty sure this is hamlib related because just one rig is not working and works fine on N1MM, and WJST-X works fine via Omnirig, and works fine with hamlib on a K3.

I am having a terrible time getting the IC-751 to work with WSJT-X, all versions. CAT takes about 5 seconds and several retries (watching USB dongle LEDs) before it finally gets it.  Seems to poll the radio, radio answers, but is missed by WSJT-X (hamlib) and after several attempts it gets it.

IC-751 is funny, out of the box it has no CAT so you have to add the interface. In this case, it is the PIEXX RS-232 board. Works fine on N1MM, and works fine on Omnirig.

I think it may be a timing issue because I see the radio answer the poll (watching the dongle LEDs).

Does not work on hard com port or USB com port, and tried two different laptops.

I tried a couple different versions of hamlib including the latest. Also tried both 4800 and 9600 baud.

OS is Windows 11 x64.

thanks

Brian N3OC

mdblack98, 2024-08-09:
What version of WSJT-X?  

Please place this file as described below
https://www.dropbox.com/s/t52ngcalsgnpm8m/wsjtx_log_config.ini?dl=1

    C:\Users\[username]\AppData\Local\WSJT-X
    The WSJT-X_Rigcontrol.log file will be in the same location
    Here's a youtube video showing Windows users howto -- https://www.youtube.com/watch?v=q9tAvhNmSvc

For Linux put it in  
    ~/.config
    The WSJT-X_Rigcontrol.log file will be here:
    ~/.local/share/WSJT-X

For MacOS put it in
    /Users/[username]/Library/Preferences
 
Restart WSJT-X and duplicate the problem.
Shut down WSJT-X

Then send me the WSJT-X_RigControl.log file
Mike W9MDB

mac1-131, 2024-08-09:
Hi Mike, thanks. The wsjt versions were 2.7.0-rc6 (latest rc) and 2.6.1
(latest ga) both with same results, although 2.6.1 was slightly better time
wise, but still unusable.

Attached is the log you requested, hopefully the .ini file comes through
gmail. Log was taken with latest hamlog installed in WSJT.

thanks for looking at this

Brian N3OC

mdblack98, 2024-08-09:
Log didn't come through -- send it direct to me.

***@***.***

mac1-131, 2024-08-09:
re-sent

N0NB, 2025-04-05:
Brian, has this been resolved?

mac1-131, 2025-04-06:
Hi Nate, unfortunately that radio is on Antigua now and I am away from it
in September. I can check it then. If this has been addressed fixed, feel
free to close it and I can re-open it then.

Brian N3OC

N0NB, 2025-04-06:
No hurry.  I'll just mark it as needed testing and it can remain open until you get back to the radio.