Hamlib/Hamlib #1597: Windows 32 & 64 binaries fail to build nightlies

N0NB, 2024-08-18:
Hi Mike.

I see the following failures from MinGW when attempting to build the current nightly snapshot:

``` make
Making install in rigs/kenwood
make[1]: Entering directory '/home/nate/builds/hamlib-4.6~git/rigs/kenwood'
  CC       ts850.lo
  CC       ts870s.lo
  CC       ts570.lo
  CC       ts450s.lo
  CC       ts950.lo
  CC       ts50s.lo
  CC       ts790.lo
  CC       ts2000.lo
  CC       k2.lo
  CC       k3.lo
  CC       xg3.lo
  CC       ts930.lo
  CC       ts680.lo
  CC       ts690.lo
  CC       ts140.lo
  CC       ts480.lo
  CC       trc80.lo
  CC       ts590.lo
  CC       ts890s.lo
  CC       ts990s.lo
  CC       flex6xxx.lo
  CC       pihpsdr.lo
  CC       tx500.lo
  CC       thd7.lo
  CC       thf7.lo
  CC       thg71.lo
  CC       tmd700.lo
  CC       tmv7.lo
  CC       thf6a.lo
  CC       thd72.lo
  CC       tmd710.lo
  CC       thd74.lo
  CC       ts440.lo
  CC       ts940.lo
  CC       ts711.lo
  CC       ts811.lo
  CC       r5000.lo
  CC       kenwood.lo
  CC       th.lo
  CC       ic10.lo
  CC       elecraft.lo
kenwood.c: In function ‘kenwood_get_level’:
kenwood.c:3907:9: error: a label can only be part of a statement and a declaration is not a statement
 3907 |         pbwidth_t twidth;
      |         ^~~~~~~~~
kenwood.c:3908:9: error: expected expression before ‘int’
 3908 |         int err = rig_get_mode(rig, vfo, &priv->curr_mode, &twidth);
      |         ^~~
kenwood.c:3910:14: error: ‘err’ undeclared (first use in this function)
 3910 |         if (!err && priv->last_mode_pc != priv->curr_mode)  // only need to check when mode changes
      |              ^~~
kenwood.c:3910:14: note: each undeclared identifier is reported only once for each function it appears in
kenwood.c:3907:19: warning: unused variable ‘twidth’ [-Wunused-variable]
 3907 |         pbwidth_t twidth;
      |                   ^~~~~~
kenwood.c: In function ‘kenwood_set_level’:
kenwood.c:3317:9: warning: statement will never be executed [-Wswitch-unreachable]
 3317 |     int retval = RIG_OK;
      |         ^~~~~~
make[1]: *** [Makefile:566: kenwood.lo] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/nate/builds/hamlib-4.6~git/rigs/kenwood'
make: *** [Makefile:634: install-recursive] Error 1
```

Building on current Debian 12 doesn't result in this error, so it looks to be MinGW specific at the moment.

73, Nate

mdblack98, 2024-08-18:
Should be fixed now.

N0NB, 2024-08-18:
That particular error appears to be quelled.

For W32 only I see the following warnings:

``` make
Making install in rotators/flir
make[1]: Entering directory '/home/nate/builds/hamlib-4.6~git/rotators/flir'
  CC       flir.lo
flir.c: In function ‘flir_open’:
flir.c:209:32: warning: format ‘%f’ expects argument of type ‘float *’, but argument 3 has type ‘float_t *’ {aka ‘long double *’} [-Wformat=]
  209 |         sscanf(return_str, "* %f", &resolution_pp);
      |                               ~^   ~~~~~~~~~~~~~~
      |                                |   |
      |                                |   float_t * {aka long double *}
      |                                float *
      |                               %Lf
In file included from ../../include/hamlib/rotator.h:26,
                 from flir.c:29:
flir.c:210:38: warning: format ‘%f’ expects argument of type ‘double’, but argument 4 has type ‘float_t’ {aka ‘long double’} [-Wformat=]
  210 |         rig_debug(RIG_DEBUG_VERBOSE, "PAN resolution: %f arcsecs per position\n",
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  211 |                   resolution_pp);
      |                   ~~~~~~~~~~~~~       
      |                   |
      |                   float_t {aka long double}
../../include/hamlib/rig.h:3760:90: note: in definition of macro ‘rig_debug’
 3760 | #define rig_debug(debug_level,fmt,...) do { snprintf(debugmsgsave2,sizeof(debugmsgsave2),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); add2debugmsgsave(debugmsgsave2); } while(0)
      |                                                                                          ^~~
flir.c:210:56: note: format string is defined here
  210 |         rig_debug(RIG_DEBUG_VERBOSE, "PAN resolution: %f arcsecs per position\n",
      |                                                       ~^
      |                                                        |
      |                                                        double
      |                                                       %Lf
flir.c:222:32: warning: format ‘%f’ expects argument of type ‘float *’, but argument 3 has type ‘float_t *’ {aka ‘long double *’} [-Wformat=]
  222 |         sscanf(return_str, "* %f", &resolution_tp);
      |                               ~^   ~~~~~~~~~~~~~~
      |                                |   |
      |                                |   float_t * {aka long double *}
      |                                float *
      |                               %Lf
In file included from ../../include/hamlib/rotator.h:26,
                 from flir.c:29:
flir.c:223:38: warning: format ‘%f’ expects argument of type ‘double’, but argument 4 has type ‘float_t’ {aka ‘long double’} [-Wformat=]
  223 |         rig_debug(RIG_DEBUG_VERBOSE, "TILT resolution: %f arcsecs per position\n",
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
  224 |                   resolution_tp);
      |                   ~~~~~~~~~~~~~       
      |                   |
      |                   float_t {aka long double}
../../include/hamlib/rig.h:3760:90: note: in definition of macro ‘rig_debug’
 3760 | #define rig_debug(debug_level,fmt,...) do { snprintf(debugmsgsave2,sizeof(debugmsgsave2),fmt,__VA_ARGS__);rig_debug(debug_level,fmt,##__VA_ARGS__); add2debugmsgsave(debugmsgsave2); } while(0)
      |                                                                                          ^~~
flir.c:223:57: note: format string is defined here
  223 |         rig_debug(RIG_DEBUG_VERBOSE, "TILT resolution: %f arcsecs per position\n",
      |                                                        ~^
      |                                                         |
      |                                                         double
      |                                                        %Lf
  CCLD     libhamlib-flir.la
make[2]: Entering directory '/home/nate/builds/hamlib-4.6~git/rotators/flir'
make[2]: Nothing to be done for 'install-exec-am'.
make[2]: Nothing to be done for 'install-data-am'.
make[2]: Leaving directory '/home/nate/builds/hamlib-4.6~git/rotators/flir'
make[1]: Leaving directory '/home/nate/builds/hamlib-4.6~git/rotators/flir'
```

This warning is seen on both W32 and W64 builds:

``` make
  CC       rotctld-rotctld.o
rotctl.c:112:18: warning: ‘have_rl’ defined but not used [-Wunused-const-variable=]
  112 | static const int have_rl = 0;
      |                  ^~~~~~~
```

mdblack98, 2024-08-27:
Is this working now?

N0NB, 2024-08-27:
* On 2024 27 Aug 12:15 -0500, Michael Black wrote:
> Is this working now?

As far as I know it is.  I got the email from my cron job which means
the VM shut down and that has always meant success.

Until next time...

73, Nate

-- 
"The optimist proclaims that we live in the best of all
possible worlds.  The pessimist fears this is true."
Web: https://www.n0nb.us
Projects: https://github.com/N0NB
GPG fingerprint: 82D6 4F6B 0E67 CD41 F689 BBA6 FB2C 5130 D55A 8819

closed 2024-08-27 by mdblack98