Hamlib/Hamlib #1604: Retries/timeouts not being respected when accidentally using wrong port for Icom radios

tmiw, 2024-08-31:
## Steps to reproduce

Attempt connection to an IC-705 (or other modern Icom radio) using the incorrect serial port. (For example, on my MacBook Pro, `/dev/cu.usbmodem143201` is the CI-V port and `/dev/cu.usbmodem143203` is for GPS; I used the latter for this test.) Additionally, set timeouts as follows:

```
rig_set_conf(rig_, rig_token_lookup(rig_, "timeout"), "1000");
rig_set_conf(rig_, rig_token_lookup(rig_, "retry"), "1");
rig_set_conf(rig_, rig_token_lookup(rig_, "timeout_retry"), "1");
```

## Expected behavior

After 1 second, `rig_open()` returns `RIG_ETIMEOUT`.

## Actual behavior

`rig_open()` does not return (at least after ~30s or so when I pushed Ctrl-C).

## Notes

I see a lot of `icom_one_transaction: unknown async?  read again` messages in the terminal window, which seems to be printed by code in `rigs/icom/frame.c`:

```c
    if (sendbuf[3] != buf[2])
    {
        rig_debug(RIG_DEBUG_VERBOSE, "%s: unknown async?  read again\n", __func__);
        hl_usleep(100);
        rig_flush(rp);
        goto collision_retry;
    }
```

`collision_retry` itself doesn't seem to track the number of retries except in the case when `COL` gets returned by a device on the CI-V bus, so we never get bounced back to the higher level timeout/retry checking logic (and thus the settings set by `rig_set_conf` don't seem to be getting used).

mdblack98, 2024-09-01:
OK...should be better now.

Mike W9MDB

tmiw, 2024-09-02:
I tested f64727c1e094a89970e82ff6fea4dbb4e0e6c151 with the GPS port again and it still seems to be stuck printing `icom_one_transaction: unknown async?  read again`. This is despite using `rig_set_conf` above to set timeouts and retries. Just to make sure, am I setting the number of retries and timeout properly? (I was trying to set timeout in particular to 1000ms / 1 second.)

mdblack98, 2024-09-02:
Debug log please?

tmiw, 2024-09-03:
See below:

```
Mooneers-16-MacBook-Pro-16158:hamlib mooneer$ bin/rigctl -m 3085 -Z -vvvv -C timeout=1 -C retry=1 -C timeout_retry=1 -r /dev/cu.usbmodem141203 
2024-09-02T22:08:39.150794-0800: rigctl.c(531) Startup: bin/rigctl -m 3085 -Z -vvvv -C timeout=1 -C retry=1 -C timeout_retry=1 -r /dev/cu.usbmodem141203
2024-09-02T22:08:39.150820-0800: rigctl Hamlib 4.6~git 2024-09-01T03:25:58Z SHA=f64727 64-bit
2024-09-02T22:08:39.150825-0800: Report bugs to <hamlib-developer@lists.sourceforge.net>

2024-09-02T22:08:39.150835-0800: rig_init: 2038 time test passed
2024-09-02T22:08:39.150853-0800: initrigs4_icom: _init called
2024-09-02T22:08:39.150869-0800: rig_init: rig_model=Icom IC-705 20240831.11
2024-09-02T22:08:39.150903-0800: rig_init: rig has VFO_A
2024-09-02T22:08:39.150908-0800: rig_init: rig has VFO_B
2024-09-02T22:08:39.150912-0800: rig_init: rig has VFO_MEM
2024-09-02T22:08:39.150931-0800: **2:icom.c(703):icom_init entered
2024-09-02T22:08:39.150950-0800: **2:icom.c(775):icom_init returning(0) 
2024-09-02T22:08:39.150964-0800: rig_token_lookup called for timeout
2024-09-02T22:08:39.150969-0800: rig_confparam_lookup called for timeout
2024-09-02T22:08:39.150975-0800: rig_token_lookup called for timeout
2024-09-02T22:08:39.150979-0800: rig_confparam_lookup called for timeout
2024-09-02T22:08:39.150984-0800: rig_set_conf called
2024-09-02T22:08:39.150989-0800: rig_confparam_lookup called for 1073741838
2024-09-02T22:08:39.150994-0800: rig_set_conf: timeout='1'
2024-09-02T22:08:39.151001-0800: rig_token_lookup called for retry
2024-09-02T22:08:39.151005-0800: rig_confparam_lookup called for retry
2024-09-02T22:08:39.151010-0800: rig_token_lookup called for retry
2024-09-02T22:08:39.151014-0800: rig_confparam_lookup called for retry
2024-09-02T22:08:39.151019-0800: rig_set_conf called
2024-09-02T22:08:39.151023-0800: rig_confparam_lookup called for 1073741839
2024-09-02T22:08:39.151053-0800: rig_set_conf: retry='1'
2024-09-02T22:08:39.151063-0800: rig_token_lookup called for timeout_retry
2024-09-02T22:08:39.151068-0800: rig_confparam_lookup called for timeout_retry
2024-09-02T22:08:39.151073-0800: rig_token_lookup called for timeout_retry
2024-09-02T22:08:39.151078-0800: rig_confparam_lookup called for timeout_retry
2024-09-02T22:08:39.151082-0800: rig_set_conf called
2024-09-02T22:08:39.151087-0800: rig_confparam_lookup called for 1073741863
2024-09-02T22:08:39.151092-0800: rig_set_conf: timeout_retry='1'
2024-09-02T22:08:39.151097-0800: rig_set_conf called
2024-09-02T22:08:39.151101-0800: rig_confparam_lookup called for 1073741834
2024-09-02T22:08:39.151106-0800: rig_set_conf: rig_pathname='/dev/cu.usbmodem141203'
2024-09-02T22:08:39.151115-0800: rig.c(932):rig_open entered
2024-09-02T22:08:39.151197-0800: rig_open: cwd=/Users/mooneer/devel/freedv-gui/hamlib
2024-09-02T22:08:39.151214-0800: rig_open: /Users/mooneer/devel/freedv-gui/hamlib/hamlib_settings No such file or directory
2024-09-02T22:08:39.151232-0800: serial_open: /dev/cu.usbmodem141203
2024-09-02T22:08:39.208961-0800: **2:rig.c(8367):async_data_handler_start entered
2024-09-02T22:08:39.208981-0800: **2:rig.c(8374):async_data_handler_start returning(0) 
2024-09-02T22:08:39.208988-0800: rig_open: 0x7ff73000f43c rs->comm_state==1?=1
2024-09-02T22:08:39.313650-0800: **2:icom.c(1152):icom_rig_open entered
2024-09-02T22:08:39.313666-0800: icom_rig_open: IC-705 v20240831.11
2024-09-02T22:08:39.313672-0800: ***3:icom.c(823):icom_get_usb_echo_off entered
2024-09-02T22:08:39.313685-0800: ****4:frame.c(507):icom_transaction entered
2024-09-02T22:08:39.313691-0800: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2024-09-02T22:08:39.313697-0800: *****5:frame.c(145):icom_one_transaction entered
2024-09-02T22:08:39.952525-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:39.974984-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:40.946355-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:40.964460-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:41.938263-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:41.958936-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:41.978843-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:42.946899-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:42.965155-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:43.938492-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:43.958925-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:43.978181-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:44.957975-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:44.976171-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:45.938632-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:45.958922-0800: icom_one_transaction: unknown async?  read again
2024-09-02T22:08:45.978268-0800: icom_one_transaction: unknown async?  read again
[... 184 lines cut ...]
2024-09-02T22:09:41.238168-0800: network.c(1596): multicast publisher address=224.0.0.1, port=4532
2024-09-02T22:09:41.238243-0800: network.c(1663) MULTICAST_TRANSCEIVE enabled
2024-09-02T22:09:41.238252-0800: network.c(1669) MULTICAST_SPECTRUM enabled
2024-09-02T22:09:41.238296-0800: **2:network.c(1720):network_multicast_publisher_start returning(0) 
[... cut ...]
2024-09-02T22:09:41.238553-0800: is_networked: Can use 127.0.0.1
2024-09-02T22:09:41.238574-0800: rig.c(1625):rig_open returning2(0) 
2024-09-02T22:09:41.238597-0800: is_networked: Will use 10.0.1.163
2024-09-02T22:09:41.238611-0800: is_networked: Will use 192.168.20.1
2024-09-02T22:09:41.238620-0800: is_networked: Will use 192.168.4.1
Opened rig model 3085, 'IC-705'
2024-09-02T22:09:41.238645-0800: Backend version: 20240831.11, Status: Stable
2024-09-02T22:09:41.238666-0800: multicast_receiver: multicast binding to INADDR_ANY

Rig command: 
```

pvrose, 2025-04-06:
Tested - now getting expected timeout on IC-705. 

hamlib version Hamlib 4.7~git 2025-01-21T05:05:50Z SHA=bb4eb7 64-bit

Phil

`philip@chinook:~/run$ rigctl -m 3085 -Z -vvvv -C timeout=1 -C retry=1 -C timeout_retry=1 -r /dev/ttyACM1
2025-04-06T14:52:31.098328+0100: rigctl.c(532) Startup: rigctl -m 3085 -Z -vvvv -C timeout=1 -C retry=1 -C timeout_retry=1 -r /dev/ttyACM1
2025-04-06T14:52:31.098503+0100: rigctl Hamlib 4.7~git 2025-01-21T05:05:50Z SHA=bb4eb7 64-bit
2025-04-06T14:52:31.098537+0100: Report bugs to <hamlib-developer@lists.sourceforge.net>

2025-04-06T14:52:31.098653+0100: rig_init: 2038 time test passed
2025-04-06T14:52:31.098782+0100: initrigs4_icom: _init called
2025-04-06T14:52:31.098864+0100: rig_init: rig_model=Icom IC-705 20241030.11
2025-04-06T14:52:31.099028+0100: rig_init: rig has VFO_A
2025-04-06T14:52:31.099057+0100: rig_init: rig has VFO_B
2025-04-06T14:52:31.099077+0100: rig_init: rig has VFO_MEM
2025-04-06T14:52:31.099140+0100: **2:icom.c(705):icom_init entered
2025-04-06T14:52:31.099183+0100: **2:icom.c(777):icom_init returning(0) 
2025-04-06T14:52:31.099243+0100: rig_token_lookup called for timeout
2025-04-06T14:52:31.099282+0100: rig_confparam_lookup called for timeout
2025-04-06T14:52:31.099316+0100: rig_token_lookup called for timeout
2025-04-06T14:52:31.099344+0100: rig_confparam_lookup called for timeout
2025-04-06T14:52:31.099395+0100: rig_set_conf called
2025-04-06T14:52:31.099430+0100: rig_confparam_lookup called for timeout
2025-04-06T14:52:31.099466+0100: rig_set_conf: timeout='1'
2025-04-06T14:52:31.099505+0100: rig_token_lookup called for retry
2025-04-06T14:52:31.099534+0100: rig_confparam_lookup called for retry
2025-04-06T14:52:31.099560+0100: rig_token_lookup called for retry
2025-04-06T14:52:31.099587+0100: rig_confparam_lookup called for retry
2025-04-06T14:52:31.099613+0100: rig_set_conf called
2025-04-06T14:52:31.099641+0100: rig_confparam_lookup called for retry
2025-04-06T14:52:31.099667+0100: rig_set_conf: retry='1'
2025-04-06T14:52:31.099732+0100: rig_token_lookup called for timeout_retry
2025-04-06T14:52:31.099764+0100: rig_confparam_lookup called for timeout_retry
2025-04-06T14:52:31.099792+0100: rig_token_lookup called for timeout_retry
2025-04-06T14:52:31.099825+0100: rig_confparam_lookup called for timeout_retry
2025-04-06T14:52:31.099857+0100: rig_set_conf called
2025-04-06T14:52:31.099884+0100: rig_confparam_lookup called for timeout_retry
2025-04-06T14:52:31.099910+0100: rig_set_conf: timeout_retry='1'
2025-04-06T14:52:31.099931+0100: rig_set_conf called
2025-04-06T14:52:31.099950+0100: rig_confparam_lookup called for rig_pathname
2025-04-06T14:52:31.099972+0100: rig_set_conf: rig_pathname='/dev/ttyACM1'
2025-04-06T14:52:31.100000+0100: rig.c(945):rig_open entered
2025-04-06T14:52:31.100198+0100: serial_open: /dev/ttyACM1
2025-04-06T14:52:31.151022+0100: **2:rig.c(8399):async_data_handler_start entered
2025-04-06T14:52:31.151098+0100: **2:rig.c(8406):async_data_handler_start returning(0) 
2025-04-06T14:52:31.151131+0100: rig_open: 0x5643057471d4 rs->comm_state==1?=1
2025-04-06T14:52:31.251350+0100: **2:icom.c(1158):icom_rig_open entered
2025-04-06T14:52:31.251410+0100: icom_rig_open: IC-705 v20241030.11
2025-04-06T14:52:31.251456+0100: ***3:icom.c(825):icom_get_usb_echo_off entered
2025-04-06T14:52:31.251488+0100: ****4:frame.c(525):icom_transaction entered
2025-04-06T14:52:31.251524+0100: icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
2025-04-06T14:52:31.251561+0100: *****5:frame.c(145):icom_one_transaction entered
2025-04-06T14:52:33.263268+0100: *****5:frame.c(330):icom_one_transaction returning(-5) Communication timed out

2025-04-06T14:52:33.263342+0100: icom_transaction: retry=0: serial_setup: tcsetattr TCSANOW
read_string_generic called, rxmax=4095 direct=1, expected_len=1
**2:rig.c(8399):async_data_handler_start entered
async_data_handler_start: async data support disabled since async_data_enabled=0
**2:rig.c(8406):async_data_handler_start returning(0) 
rig_open: 0x5643057471d4 rs->comm_state==1?=1
**2:icom.c(1158):icom_rig_open entered
icom_rig_open: IC-705 v20241030.11
***3:icom.c(825):icom_get_usb_echo_off entered
****4:frame.c(525):icom_transaction entered
icom_transaction: cmd=0x03, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
write_block(): TX 6 bytes
read_string_generic called, rxmax=200 direct=1, expected_len=1
[... 17 lines cut ...]
write_block(): TX 6 bytes
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(1372): retrying read timeout 1/1 timeout=1000
read_string_generic(): Timed out 2.011 seconds after 0 chars, direct=1
*****5:frame.c(330):icom_one_transaction returning(-5) Communication timed out

Communication timed out

Communication timed out
Communication timed out

2025-04-06T14:52:33.363611+0100: ****4:frame.c(556):icom_transaction returning(-5) Communication timed out

2025-04-06T14:52:33.363644+0100: ***3:icom.c(833):icom_get_usb_echo_off returning(-5) Communication timed out

2025-04-06T14:52:33.363675+0100: icom_rig_open: Unable to determine Icom echo status -- is rig on and connected?
2025-04-06T14:52:33.363760+0100: **2:icom.c(1193):icom_rig_open returning(-5) Communication timed out

2025-04-06T14:52:33.363806+0100: **2:rig.c(8484):async_data_handler_stop entered
2025-04-06T14:52:33.363827+0100: **2:rig.c(8515):async_data_handler_stop returning(0) 
2025-04-06T14:52:33.363859+0100: **2:rig.c(8525):morse_data_handler_stop entered
2025-04-06T14:52:33.564028+0100: **2:rig.c(8571):morse_data_handler_stop returning(0) 
2025-04-06T14:52:33.564090+0100: ser_close: restoring options
2025-04-06T14:52:33.566424+0100: rig.c(1476):rig_open returning2(-5) Communication timed out
`

N0NB, 2025-04-06:
That is a rather old version of the 4.7~git branch, Phil.  Can you try the current snapshot?

https://n0nb.users.sourceforge.net

Thanks!

pvrose, 2025-04-06:
Nate,Will do, it'll be tomorrow at the earliest.Phil.Get Outlook for AndroidFrom: Nate Bargmann ***@***.***>Sent: Sunday, April 6, 2025 6:21:31 PMTo: Hamlib/Hamlib ***@***.***>Cc: pvrose ***@***.***>; Comment ***@***.***>Subject: Re: [Hamlib/Hamlib] Retries/timeouts not being respected when
 accidentally using wrong port for Icom radios (Issue #1604)

N0NB left a comment (Hamlib/Hamlib#1604)That is a rather old version of the 4.7~git branch, Phil.  Can you try the current snapshot?https://n0nb.users.sourceforge.netThanks!

pvrose, 2025-04-06:
I manged to find some time:

It works with hamlib version Hamlib 4.7~git 2025-03-31T01:34:43Z SHA=a9ce0b 64-bit which was the latest I downloaded a few minutes ago.

I tried both rigctl and my app's API use of hamlib.

Phil