Hamlib/Hamlib #1606: h e l p : Problem to generate the Python bindings on windows 11.

ofracaro, 2024-09-02:
Hello,

I'm currently attempting to compile Hamlib on a Cygwin environment in Windows 11, specifically aiming to generate the Python bindings. Here's a summary of the steps I've taken so far:

I installed the necessary tools on Cygwin, including gcc, g++, make, autoconf, automake, libtool, swig, and python.
I cloned the Hamlib repository from GitHub.
I tried generating the configure script using autoreconf -i and subsequently running it with:

"./configure --with-python-binding PYTHON=/cygdrive/c/python/python.exe"

However, I encountered an issue with Python version validation that prevents the configuration from completing. The specific error message is:
"This version of the AC_PYTHON_DEVEL macro doesn't work properly with versions of Python before 2.1.0."

I've been looking through the configure.ac file to find a way to disable this Python version validation, but I'm not entirely sure how to proceed without negatively impacting the configuration process.

**My Question:**

_Do you recommend any changes in the configure.ac file to skip the Python version check?
_Is there any instruction or guide available for compiling the Python bindings on Windows?
_Alternatively, is there a precompiled library for Python on Windows available from any repository?

For additional context, I was able to compile the Python bindings on Linux without any issues, but I need it too in windows 11.
I would greatly appreciate any guidance you can provide. 

Thank you very much for your time and for all the work on Hamlib.

Luis Omar Fracaro
LU9EOE 
lu9eoe@gmail.com

mdblack98, 2024-09-02:
The full message should help
This version of the AC@&t@_PYTHON_DEVEL macro
doesn't work properly with versions of Python before
2.1.0. You may need to re-run configure, setting the
variables PYTHON_CPPFLAGS, PYTHON_LIBS, PYTHON_SITE_PKG,
PYTHON_EXTRA_LIBS and PYTHON_EXTRA_LDFLAGS by hand.
Moreover, to disable this check, set PYTHON_NOVERSIONCHECK
to something else than an empty string.

ofracaro, 2024-09-02:
Muchas gracias Michael, estare haciendo esa prueba y vemos como resulta.