Hamlib/Hamlib #1616: Connecting to CI-V over WiFi

luarvique, 2024-09-24:
Sorry for bringing this up as an issue, but I do not seem to be able to connect to an IC-705 over WiFi using rigcontrol. Using the following command line:
`
rigctl -m 3085 -c 0xA4 -r 192.168.1.254:50001 -v f
`
The above command results in connection timeout. The IP address and port are correct, as can be verified by using wfView.

It is important to note that IC-705 also requires username and password for connection. I have not found how to supply them to rigctl.

mdblack98, 2024-09-24:
please run this and send the log.txt file.

rigctl -m 3085 -r 192.168.1.254:50001 -v f >log.txt 2>&1

luarvique, 2024-09-24:
**log.txt:**
```
Opened rig model 3085, 'IC-705'
get_freq: error = Communication timed out
rig_get_freq(2171): freqMainB=0, modeMainB=, widthMainB=0
rig.c(2173):rig_get_freq return(-5)
rigctl_parse.c(2013):rigctl_get_freq return(-5)
```

[log.txt](https://github.com/user-attachments/files/17116659/log.txt)

mdblack98, 2024-09-24:
I found some indirect information on the network protocol.  Icom maintains it as proprietary information and there's no documentation availabe.
It's been reverse engineered by some so I can put this in my queue to implement -- likely to be quite a while (like version 4.7 development) as it's not trivial to implement.

Mike W9MDB

Here it is.
log.txt

luarvique, 2024-09-24:
Acknowledged. On my end, I am looking to use Rigctl to integrate OpenWebRX+ with HAM transceivers. OWRX+ will use its own SDR devices to receive signals, but will also mirror its frequency/modulation settings to a connected transceiver.

luarvique, 2024-10-05:
Just letting you know that Hamlib/RigCtl integration is now live in OpenWebRX+ 1.2.70 and some people are even using it:

https://m.youtube.com/watch?v=fqhPZgwqxnU

BE999, 2025-01-19:
Hello, I‘m also an owner of the IC 705 and I really would appreciate a solution for milestone 4.7 and I hope the integration in OpenwebRX+ will work for me! 
Thank you very much in advance!
Bernd
DN9HBE