Hamlib/Hamlib #1617: Icom IC-707 CI-V Bugs:  memory clear, memory ranges and tuning steps

MiM-TMassey, 2024-10-01:
Version:  rigctld Hamlib 4.6~git 2024-09-29
OS:  Windows 10 x64
Radio Icom IC-707 via XGGComms USB-Digimode-4 on COM4.

### Very brief tl;dr:
There seem to be a couple of mistakes in the IC-707 rig code.  This rig does **not** support memory clear (CI-V command 0x0b), and it **does** support tuning steps (CI-V command 0x10 sub 0x00 and 0x01).  Also, the memory ranges are incorrect:  it supports 1-25 single memories, 26-30 split (dual) memories and 31-32 scan edge memories.

Also, a couple of small questions:  what does "E" mean in capabilities, and it seems that `--show-conf` now exits the daemon instead of letting it run?  Finally, if there's any other testing I can do for the IC-707, please let me know.

### Details:
I just acquired a new-in-box Icom IC-707 radio -- yes, it sat on a prepper's shelf for 30 years!  I bought this radio with almost no knowledge of this radio.  I saw that it's missing a *lot* of buttons you'd normally have...  But I **did** check to see that this radio was supported by Hamlib.  It was, so I figured I was good to go:  I can use rig control to make up for the lack of buttons, right?

Now that I own it, I've now found that this is a bit of a rare model.  There is *very* little documentation for this radio outside of the basic user's manual.  Most importantly, there is no CI-V information specifically for this radio.  In addition, this is my first older radio:  I had no idea how limited early rig control was in these older devices...

The first goal was to find out what radio this was closest to.  After much frustrating testing, it seems that this radio is **very** close to the same CI-V features as an IC-737.  I was able to make all documented IC-737 CI-V commands work except the antenna select command (0x12) because the IC-707 only has a single antenna port, and some of the tuning steps (0x10):  the IC-737 supports a bunch, and the IC-707 supports less.  It's also very similar to the IC-IC-725/726/728/729, except the IC-707 does not support the memory clear command (0x0b), and adds the above-mentioned tuning step command.

However, when comparing the features I've tested to those available in Hamlib, I've found a few differences.  I thought I would document them here in the hopes that someone might be able to make the changes to correct the IC-707 model. I've cut out the relevant items from the output of --dump-caps and added notes from my testing:

```
VFO Ops: CPY FROM_VFO TO_VFO MCL
```
Does "MCL" mean Memory clear?  708 does not have memory clear from my actual testing:  CI-V 0x0b command returns ... 0xfa ... .  That seems to be verified by checking the --dump-caps from the IC-737, which does *NOT* list "MCL" as a VFO Ops.  Assming "MCL" is CI-V command 0x0b, this should be removed from the IC-707.

```
Memories:2024-10-01T10:52:55.448178-0500: rig_strmtype called

        1..26:          MEM
          Mem caps: FREQ MODE WIDTH 2024-10-01T10:52:55.448178-0500: rig_strmtype called

        27..30:         EDGE
          Mem caps: FREQ MODE WIDTH 2024-10-01T10:52:55.448178-0500: rig_strmtype called

        31..35:         SAT
          Mem caps: FREQ MODE WIDTH
```
This differs from the actual memory configuration of the IC-707:  1-25 Normal single-frequency; 26-30 Split dual-frequency; 31-32 Scan edge channels (high and low limits for VFO scan).  Page 25 of IC-707 manual (Page 28 of PDF).

```
Tuning steps:
        10.0 Hz:        AM CW USB LSB FM
Tuning steps status:    OK (0)
```
The radio physically supports three tuning steps:  10Hz, 1kHz and 1MHz.  Using CI-V command 0x10 I can only get two of them to work:  10Hz (subcommand 0x00) and 1kHz (subcommand 0x01).  I could not find a subcommand for 1MHz.  The IC-737 does not have a 1MHz tuning step option:  my guess is this was added to the IC-707 to overcome the lack of band buttons and 10-digit number pad and wasn't added to the CI-V interface.  However, if anyone has any ideas to find the 1MHz tuning step I'm happy to test!

```
Can set Tuning Step:    N
Can get Tuning Step:    N
```
See above regarding tuning steps.  (Though I **think** "get Tuning Step" is not supported.)

```
Can set Transceive:     E
Can set Channel:        E
Can get Channel:        E
```
What is "E"?  (At least it matches the IC-737...)

There's one other detail I've found updating to the lastest daily build:  Does --show-conf now exit on use?  I really liked being able to use --show-conf in my normal use just to ensure that all of my settings were correct -- I tweak things frequently and it's handy to have all of the settings listed in my logs when I start the daemon...  If that was intentional, I would suggesting updating the manpage to add that detail -- just like it does for --dump-caps.

Finally, are there any settings for the IC-707 that should be better tested?  I know that until recently the IC-707 rig was in Alpha state, and might have ony been moved to Beta because no one complained for a long time -- because maybe no one is using it?  :)  But if there's anything that I can do to further test the rig code I'm more than happy to do so!

Thank you very much for all of your hard work on Hamlib.  It has **greatly** benefitted **everyone** who connects their computer to their radio.  I hope that I can help make the IC-707 a little bit more complete and correct.

mdblack98, 2024-10-01:
Looks like 1MHz tuning step could be 10,11, or 12 if it is supported -- give it a try and let me know if it works.

MiM-TMassey, 2024-10-01:
I tried the following command with a subcommand from 0x00 to 0x20:

 0xfe 0xfe 0x3e 0xe0 0x10 **0x20** 0xfd

All returned 0xFA except 0x00 and 0x01. :(

MiM-TMassey, 2024-10-03:
Hello!

I see that the code has been updated, so I updated to the latest daily build:   rigctld Hamlib 4.6~git 2024-09-29.

The output of "G ?" is now "CPY FROM_VFO TO_VFO" which seems correct.  Excellent!  I've tested each of these functions and AFAICT from my limited experience these are all working correctly.

Now, for tuning steps:  I'm having trouble making this work.  The rigctl manpage says that the tuning step command is "N", with a parameter of step size in Hz.  The new ic707.c code seems to show steps of 10 and 1000 Hz, if I'm interpreting it correctly -- which also matches the radio manual.  Here's a quote of the code from https://github.com/Hamlib/Hamlib/blob/master/rigs/icom/ic707.c

```
 .tuning_steps =     {
        {IC707_ALL_RX_MODES, 10}, 
        {IC707_ALL_RX_MODES, 1000}, 
        RIG_TS_END,
    },
```

However, when I try to use the "N" command, I get a "RPRT -11".  It doesn't matter if I use "N 10" or "N 1000" (or any other number), I get the same error.  I've included the debug output from those command below, but the relevant part seems to be:  `**2:rig.c(6305):rig_set_ts returning(-11) Feature not available` .  This is different than if I were to use an "X" instead of a number:  in that case, I get a "RPRT -1" (Invalid parameter).

I just reconfirmed:  If I manually send the CI-V codes, they work.  If I send "... 0x10 0x01 ...", an arrow appears above the 1kHz place to signify 1kHz tuning.  If I then send "... 0x10 0x01 ..." the arrow disappears above the 1kHz place to signify 10Hz tuning.

I'm not sure if there's still a problem with the tuning steps code in the ic707.c back end, or if I'm not using the N command correctly.  I would greatly appreciate your thoughts.

Thank you **very** much for your fast response above!  I was surprised to see an update so quickly and I'm happy to test any changes you make.  I believe tuning steps and scan functions (described in a different GitHub issue) are the only CI-V commands I haven't been able to reproduce in Hamlib successfully.  However, **please** let me know if there are **any** other items you would like me to test on this radio:  I'm all set up and highly motivated!  :)

MiM-TMassey, 2024-10-06:
Upgraded to:  hamlib-w32-4.6~git-20241006-53a3d8635

"N 10" and "N 1000" now work successfully!  Thank you very much!

One question:  is there a better way for clients of rigctld to get which tuning steps are available than to do a "1" and parse the output?  (I'm also doing that for bandwidths as well.)

I do believe that this implements the features requested in this issue and can be closed.  I will just use this last opportunity to outline the other questions mentioned here if you have a minute:

- --show-conf seems to now exit.  It did not exit in previous versions.  If this change was intentional, I would suggest updating the manpage to state this (as it currently does for --dump-caps)
- What does the "E" mean in the list of capabilities?  It's in places where you might expect a "Y".
- Is there anything else I can test for the IC-707 back end?

Thank you again for your work:  I'm amazed how fast these were updated!  I appreciate your help.

mdblack98, 2024-10-07:
show-conf is supposed to exit if no -r is given -- if you are giving it -r do you see it exit?

I added the ability to see the tuning steps by using "N ?" from rigctl.

It will produce a comma separated list of tuning steps.

e.g.
Rig command: N ?
10, 1000

MiM-TMassey, 2024-10-08:
> show-conf is supposed to exit if no -r is given -- if you are giving it -r do you see it exit?

That is **exactly** what I'm seeing.  And as I said before, this is a regression (or at least a change in behavior) since "rigctl Hamlib 4.6~git Jan 12 17:23:11Z 2023 SHA=9c48ec".  That happens to be the version of rigctld included with Log4OM2, which is the version of rigctld I was using before I started playing with this new radio (and started using the dailies so that I could accurately test and report).

For completeness, a full copy of the output of rigctld is included below, including the full command line I used to launch it.  You mentioned that this should not happen with the "-r" parameter.  I tend to use the full parameters for things with lots of parameters that I don't frequently use.  In the output below I've used the "--rig-path=" parameter, but I've tested it both ways with the same result.

Remove the "--show-conf" and it continues running as expected.

I can't test the new "N ?" setup yet; I need to wait until the dailies are rebuilt tonight/tomorow.  I will test then.

```
C:\Users\tmassey>"s:\software\AmateurRadio\RadioSoftware\Hamlib\hamlib-w32-4.6~git-20241006-53a3d8635\hamlib-w32-4.6~git\bin\rigctld.exe" --model=3012 --rig-file=COM4 --serial-speed=1200 --set-conf=stop_bits=1 --ptt-file=COM4 --ptt-type=RTS --port=4532 --show-conf
rig_pathname: "Path name to the device file of the rig"
        Default: /dev/rig, Value: COM4
        String.
write_delay: "Delay in ms between each byte sent out"
        Default: 0, Value: 0
        Range: 0..1000, step 1.0
post_write_delay: "Delay in ms between each command sent out"
        Default: 0, Value: 0
        Range: 0..1000, step 1.0
post_ptt_delay: "Delay in ms after PTT is asserted"
        Default: 0, Value: 0
        Range: 0..2000, step 1.0
timeout: "Timeout in ms"
        Default: 0, Value: 500
        Range: 0..10000, step 1.0
retry: "Max number of retry"
        Default: 0, Value: 3
        Range: 0..10, step 1.0
timeout_retry: "Set the # of retries for read timeouts that may occur with some serial interfaces"
        Default: 1, Value: 1
        Range: 0..100, step 1.0
Selected range list: "The tx/rx range list in use"
        Default: 0, Value:
        Range: 1..5, step 1.0
Selected range list: "The tx/rx range list name"
        Default: Default, Value:
        String.
device_id: "User-specified device ID for multicast state data and commands"
        Default: , Value:
        String.
vfo_comp: "VFO compensation in ppm"
        Default: 0, Value: 0.000000
        Range: 0..1000, step 0.0
poll_interval: "Polling interval in ms for transceive emulation, defaults to 1000, value of 0 disables polling"
        Default: 1000, Value: 1000
        Range: 0..1e+06, step 1.0
ptt_type: "Push-To-Talk interface type override"
        Default: RIG, Value: RTS
        Combo: RIG, RIGMICDATA, DTR, RTS, Parallel, CM108, GPIO, GPION, None
ptt_pathname: "Path to the device of the Push-To-Talk"
        Default: /dev/rig, Value: COM4
        String.
ptt_bitnum: "Push-To-Talk GPIO bit number"
        Default: 2, Value: 0
        Range: 0..7, step 1.0
dcd_type: "Data Carrier Detect (or squelch) interface type override"
        Default: RIG, Value: None
        Combo: RIG, DSR, CTS, CD, Parallel, CM108, GPIO, GPION, None
dcd_pathname: "Path to the device of the Data Carrier Detect (or squelch)"
        Default: /dev/rig, Value:
        String.
lo_freq: "Frequency to add to the VFO frequency for use with a transverter"
        Default: 0, Value: 0
        Range: 0..1e+09, step 0.1
cache_timeout: "Cache timeout, value of 0 disables caching"
        Default: 500, Value: 500
        Range: 0..5000, step 1.0
auto_power_on: "True enables compatible rigs to be powered up on open"
[... 76 lines cut ...]
        Range: 0..255, step 1.0
mode731: "CI-V operating frequency data length, needed for IC731 and IC735"
        Default: 0, Value: 0
        Check button.
no_xchg: "Don't Use VFO XCHG to set other VFO mode and Frequency"
        Default: 0, Value: 0
        Check button.
tone_enable: "Overcome a bug in IC-705 to enable tone after frequency change"
        Default: 0, Value:
        Check button.
filter_usbd: "Filter to use for USBD/LSBD when setting mode"
        Default: 0, Value:
        Range: 0..3, step 1.0
filter_usb: "Filter to use when for USB/LSB setting mode"
        Default: 0, Value:
        Range: 0..3, step 1.0
filter_cw: "Filter to use for CW/CWR when setting mode"
        Default: 0, Value:
        Range: 0..3, step 1.0
filter_fm: "Filter to use for FM/PKTFM when setting mode"
        Default: 0, Value:
        Range: 0..3, step 1.0

C:\Users\tmassey>
```

MiM-TMassey, 2024-10-09:
Upgraded to:  rigctl Hamlib 4.6~git 2024-10-07T17:19:03Z SHA=882d22 32-bit

Tested "N ?" successfully.  With my IC-707, I received the following output:

```
10, 1000
RPRT 0
```

--show-conf still exits, even with valid -r / --rig-file parameters.

MiM-TMassey, 2024-10-09:
Update on Tuning Steps:  get tuning steps ("n") does not work as expected.

According to Hamlib dump_state, the IC-707 radio has "Can get Tuning Step:    Y".  However, looking at the CI-V documentation, the only tuning steps command I can find is to *set* a tuning step.  I don't see a way to *get* a tuning step.  And when I try to use the Hamlib protocol command "n", I get a "RPRT -9".  So it seems that "Can get Tuning Step:" should be **N**, not Y.

Aso, seems that https://github.com/Hamlib/Hamlib/commit/e096426ddc8210f298571ad943159f2b3844ddef may fix the --show-conf issue.  I'll test the changes to --show-conf tomorrow after the daily builds are updated.

MiM-TMassey, 2024-10-10:
Upgraded to:  rigctld Hamlib 4.6~git 2024-10-09T13:53:37Z SHA=e09642 32-bit

### Still Incorrect
- Icom IC-707 does not allow **getting** current tuning step, yet dump_state shows:  `Can get Tuning Step:    Y`.

### Corrected
- `--show-conf`:  The program no longer exits when you use this parameter with a valid `-r`.  This is the behavior I expected, and the same as the version from 2023 I was using.  Thank you!

### Other Questions
Here are my other outstanding questions.  If you don't follow up here, which is fine, I will send an e-mail to the hamlib-developer@sourceforge.net e-mail list once the dust settles here.  But if you want to answer I would be grateful!  :)

- Is there a way to determine from Hamlib which memory ranges allow split and which ones do not?  The code in ic707.c that defines the channels **does** differentiate (correctly) between split-capable channels and not, but from what I can tell of the `dump-caps` output there's no way to tell.  It would be nice if, say, "SPLIT" were one of the "Mem caps"... :)
- What does "E" mean in certain of the "Can set/Can get" lines?  In particular:
```
Can set Transceive:     E
Can set Channel:        E
Can get Channel:        E
```

Thank you very much for all of your help!  I really appreciate it.  The IC-707 back-end is working **so** much better now!

mdblack98, 2024-10-10:
What does "not allow" mean?  Can you provide debug?

E means it's emulated so it's not really the rig doing it -- all done in software

Here are my other outstanding questions. If you don't follow up here, which is fine, I will send an e-mail to the ***@***.*** e-mail list once the dust settles here. But if you want to answer I would be grateful! :)

MiM-TMassey, 2024-10-10:
> What does "not allow" mean?  Can you provide debug?

"Not allow" means that there is zero C-IV commands that ask the radio to provide what the current tuning step is.  I'm sure there *is* such a C-IV command, but the IC-707 does not support it.  When you do an `n`, you get an `RPRT -9`.

Here's the debug output from `n`:
```
2024-10-10T19:01:18.915052-0500: mutex_rigctld: client lock engaged
2024-10-10T19:01:18.915052-0500: rigctl_parse.c(3217):rigctl_get_ts entered
2024-10-10T19:01:18.915052-0500: **2:rig.c(6371):rig_get_ts entered
2024-10-10T19:01:18.915052-0500: ***3:icom.c(6867):icom_get_ts entered
2024-10-10T19:01:18.915052-0500: ****4:frame.c(507):icom_transaction entered
2024-10-10T19:01:18.915052-0500: icom_transaction: cmd=0x10, subcmd=0xffffffff, payload_len=0
2024-10-10T19:01:18.915052-0500: *****5:frame.c(145):icom_one_transaction entered
2024-10-10T19:01:19.031197-0500: *****5:frame.c(434):icom_one_transaction returning(-9) Command rejected by the rig

2024-10-10T19:01:19.031197-0500: icom_transaction: failed: rigctl_parse: cmd=n(6e) handle=3
mutex_rigctld: client lock engaged
rigctl(d): n 'currVFO' '' '' ''
rigctl_parse: vfo_opt=0
rigctl_parse.c(3217):rigctl_get_ts entered
**2:rig.c(6371):rig_get_ts entered
**rig.c(6388) trace
***3:icom.c(6867):icom_get_ts entered
****4:frame.c(507):icom_transaction entered
icom_transaction: cmd=0x10, subcmd=0xffffffff, payload_len=0
*****5:frame.c(145):icom_one_transaction entered
write_block(): TX 6 bytes
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 6 characters, direct=1
read_string_generic called, rxmax=200 direct=1, expected_len=1
read_string_generic(): RX 6 characters, direct=1
*****5:frame.c(434):icom_one_transaction returning(-9) Command rejected by the rig

Command rejected by the rig
Command rejected by the rig

2024-10-10T19:01:19.031197-0500: ****4:frame.c(538):icom_transaction returning(-9) Command rejected by the rig

2024-10-10T19:01:19.031197-0500: ***3:icom.c(6874):icom_get_ts returning(-9) Command rejected by the rig

2024-10-10T19:01:19.031197-0500: **2:rig.c(6390):rig_get_ts returning(-9) Command rejected by the rig

2024-10-10T19:01:19.031197-0500: rigctl_parse.c(3223):rigctl_get_ts returning2(-9) Command rejected by the rig

2024-10-10T19:01:19.031197-0500: mutex_rigctld: client lock disengaged
2024-10-10T19:01:19.031197-0500: handle_socket: rigctl_parse retcode=-9
2024-10-10T19:01:19.031197-0500: mutex_rigctld: client lock engaged
2024-10-10T19:01:19.031197-0500: mutex_rigctld: client lock disengaged
2024-10-10T19:01:19.031197-0500: mutex_rigctld: client lock engaged
2024-10-10T19:01:19.031197-0500: mutex_rigctld: client lock disengaged
```

For a comparison, here's the debug output from `n 10`:
```
2024-10-10T19:00:26.342503-0500: mutex_rigctld: client lock engaged
2024-10-10T19:00:26.342503-0500: rigctl_parse.c(3196):rigctl_set_ts entered
2024-10-10T19:00:26.342503-0500: **2:rig.c(6299):rig_set_ts entered
2024-10-10T19:00:26.342503-0500: ***3:icom.c(6824):icom_set_ts entered
2024-10-10T19:00:26.342503-0500: ****4:frame.c(507):icom_transaction entered
2024-10-10T19:00:26.342503-0500: icom_transaction: cmd=0x10, subcmd=0x00, payload_len=0
2024-10-10T19:00:26.342503-0500: *****5:frame.c(145):icom_one_transaction entered
2024-10-10T19:00:26.474016-0500: *****5:frame.c(485):icom_one_transaction returning(0)
2024-10-10T19:00:26.474016-0500: ****4:frame.c(538):icom_transaction returning(0)
2024-10-10T19:00:26.474016-0500: ***3:icom.c(6853):icom_set_ts returning(0)
2024-10-10T19:00:26.474016-0500: **2:rig.c(6313):rig_set_ts returning(0)
2024-10-10T19:00:26.474016-0500: rigctl_parse.c(3207):rigctl_set_ts returning2(0)
2024-10-10T19:00:26.474016-0500: mutex_rigctld: client lock disengaged
2024-10-10T19:00:26.474016-0500: mutex_rigctld: client lock engaged
2024-10-10T19:00:26.474016-0500: mutex_rigctld: client lock disengaged
2024-10-10T19:00:26.474016-0500: mutex_rigctld: client lock engaged
2024-10-10T19:00:26.474016-0500: mutex_rigctld: client lock disengaged
```

And thank you for letting me know what the "E" means!  I appreciate it.

MiM-TMassey, 2024-10-11:
Updated to:  rigctld Hamlib 4.6~git 2024-10-11T03:58:12Z SHA=327517 32-bit

For Get Step, dump_state now reports the following: `Can get Tuning Step:    N`.  Perfect!  And doing a 'n' now returns `RPRT -11` as expected.

My only last question is:  is there a way to query Hamlib to see the difference between memory slots 1-25 and 26-30?  1-25 are marked as "MEM", 31-32 are marked as "EDGE", but 26-30 aren't marked with anything.  For all of them, the Mem caps are the same: "FREQ MODE WIDTH".  Is there the option to add "SPLIT" to the mem caps or the missing field after the numbers?  In short, how can I programatically determine that memory slots 1-25 are single-freq slots and 26-30 are split/dual-freq slots?

I will go ahead and close this ticket at this time.  Please feel free to answer that question in the ticket if you'd like:  I'll still keep an eye on this for a reply! :)  But if the question is not answered here, I will follow up on the dev mailing list.

Once again, thank you very much for all of your hard work!  Everything is working really well -- at least, as well as the radio was designed to work 30 years ago... :)

MiM-TMassey, 2024-10-11:
I did some more digging about the memory types.

In ic707.c, the memories are defined as:
``
.chan_list =  {
        {   1,  25, RIG_MTYPE_MEM, IC_MIN_MEM_CAP },
        {  26,  30, RIG_MTYPE_SPLIT, IC_MIN_MEM_CAP },    /* split ? */
        {  31,  32, RIG_MTYPE_EDGE, IC_MIN_MEM_CAP },
        RIG_CHAN_END,
    },
```

There's only two places in the codebase where "RIG_MTYPE_SPLIT" is used:  ic707.c above and rig.h:
```
typedef enum {
    RIG_MTYPE_NONE =    0,  /*!< None */
    RIG_MTYPE_MEM,          /*!< Regular */
    RIG_MTYPE_EDGE,         /*!< Scan edge */
    RIG_MTYPE_CALL,         /*!< Call channel */
    RIG_MTYPE_MEMOPAD,      /*!< Memory pad */
    RIG_MTYPE_SAT,          /*!< Satellite */
    RIG_MTYPE_BAND,         /*!< VFO/Band channel */
    RIG_MTYPE_PRIO,         /*!< Priority channel */
	RIG_MTYPE_VOICE,		/*!< Stored Voice Message */
	RIG_MTYPE_MORSE,		/*!< Morse Message */
	RIG_MTYPE_SPLIT			/*!< Split operations */
} chan_type_t;
```

Interesting that only the IC-707 code uses the SPLIT mem cap:  I would have thought that this would have been more common?  And this code all predates my interest in the IC-707, so I'm not sure what motivated this to be added only for it.

In any case, with more digging, there's another area it seems where memory channel capabilities need to be defined.  /src/misc.c:
```
static const struct
{
    chan_type_t mtype;
    const char *str;
} mtype_str[] =
{
    { RIG_MTYPE_MEM, "MEM" },
    { RIG_MTYPE_EDGE, "EDGE" },
    { RIG_MTYPE_CALL, "CALL" },
    { RIG_MTYPE_MEMOPAD, "MEMOPAD" },
    { RIG_MTYPE_SAT, "SAT" },
    { RIG_MTYPE_BAND, "BAND" },
    { RIG_MTYPE_PRIO, "PRIO" },
    { RIG_MTYPE_NONE, "" },
};
```

It seems that RIG_MTYPE_SPLIT isn't in that list, and that's why there isn't a more specific tag for channels 26-30.

I thought it was odd that only the IC-707 has memory channels marked as SPLIT.  For example, a quick check of the IC-737 shows that it **too** has single-freq and split/dual-freq memories (PDF page 40 of the manual).  So does the IC-725/726/728/729.  I checked the IC-706 manual:  it seems by then that the rigs were upgraded so all memories were split memories.

Given that there are a few (but only a few) rigs that have these different types of memory, maybe it would make sense to mark the **single-freq memories** instead of the split memories?  If most memories are split, and it's a given that all current and new rigs will have all-split memories, why not tag the old and limited memory slots?

So that would be my proposal:  change memory caps in ic707.c:
```
.chan_list =  {
        {   1,  25, RIG_MTYPE_SINGLE, IC_MIN_MEM_CAP }, /* Single-freq memories */
        {  26,  30, RIG_MTYPE_MEM, IC_MIN_MEM_CAP },
        {  31,  32, RIG_MTYPE_EDGE, IC_MIN_MEM_CAP },
        RIG_CHAN_END,
    },
```

Change the SPLIT type to SINGLE in the enum of memory types in rig.h:
```
typedef enum {
    RIG_MTYPE_NONE =    0,  /*!< None */
    RIG_MTYPE_MEM,          /*!< Regular */
    RIG_MTYPE_EDGE,         /*!< Scan edge */
    RIG_MTYPE_CALL,         /*!< Call channel */
    RIG_MTYPE_MEMOPAD,      /*!< Memory pad */
    RIG_MTYPE_SAT,          /*!< Satellite */
    RIG_MTYPE_BAND,         /*!< VFO/Band channel */
    RIG_MTYPE_PRIO,         /*!< Priority channel */
	RIG_MTYPE_VOICE,		/*!< Stored Voice Message */
	RIG_MTYPE_MORSE,		/*!< Morse Message */
	RIG_MTYPE_SINGLE			/*!< Single-freq operations */
} chan_type_t;
```

And add SINGLE to the mtype_str[] struct:
```
static const struct
{
    chan_type_t mtype;
    const char *str;
} mtype_str[] =
{
    { RIG_MTYPE_MEM, "MEM" },
    { RIG_MTYPE_EDGE, "EDGE" },
    { RIG_MTYPE_CALL, "CALL" },
    { RIG_MTYPE_MEMOPAD, "MEMOPAD" },
    { RIG_MTYPE_SAT, "SAT" },
    { RIG_MTYPE_BAND, "BAND" },
    { RIG_MTYPE_PRIO, "PRIO" },
    { RIG_MTYPE_SINGLE, "SINGLE" },
    { RIG_MTYPE_NONE, "" },
};
```

I'm lazy, and I haven't forked Hamlib; but if you want an actual, formal PR just ask and I'm happy to do so.  But you'll probably want to make the changes your own way, anyway.

In case you want to update the models that I found the single/dual memory divisions:  IC-737:  memories 1-89 are single, 90-99 are split, and "P1, P2" are edge channels; IC-725/726/728/729:  0-22 singe, 23-24 dual, 25-26 edge.  Oddly enough, the IC-271/471 support 32 memories and all support duplex, with channels 1-2 as edge channels.

I'm going to re-open this issue for further visibility and look forward to your thoughts.

Thank you very much for all of your help!

MiM-TMassey, 2024-12-23:
Given the recent push for 4.6 I thought I would reply to this to see if there’s anything I can do to help this to be addressed before 4.6 is released?

mdblack98, 2024-12-23:
I don't like the idea of getting rid of SPLIT.  SINGLE isn't very descriptive.
The idea would be for generic set/get channel calls that would work on all manufacturer's models if possible which may be true now for Icom.  The MTYPE_SPLIT is really only needed when there's a difference in the rig as MTYPE_MEM is generic and already would cover split...or in the IC707 case MTYPE_MEM is the single channel with MTYPE_SPLIT the other.
You'll see that set/get channel is only implemented on the IC746 and ICR75 for Icom and many more for Yaesu/Kenwood.
I don't see this as important for 4.6 (it's an enhancement and not a bug) but if you want to fork Hamlib and make a patch either for just the 707 or perhaps in icom.c for more generic use go ahead.
I'm moving this to the 4.7 development.

MiM-TMassey, 2024-12-23:
It’s your code. I defer to your thoughts.

In the end, all I as a programmer would like is a way of being able to query Hamlib and get the difference between channels that can record a single frequency and channels that can record two frequencies. Right now, that is impossible, because the SPLIT attribute does not work.  How that’s resolved is up to you!

In my reply above, I also outlined what seems to be missing from making the current SPLIT notation work. I believe  /src/misc.c needs to add SPLIT to the definition of mtype_str[].  It seems that when SPLIT was added elsewhere, the process wasn’t completed here.

This seems like a bug to me, not a new feature.  Having said that, it seems from what I can tell that the IC – 707 is the only radio that (still?) uses SPLIT, and I seem to be the only person who cares about that radio at the moment.

mdblack98, 2024-12-23:
It does return that info -- I did fix misc.c

rigctl -m 3012 -u

Memories:
    1..25:       MEM
      Mem caps: FREQ MODE WIDTH 
    26..30:       SPLIT
      Mem caps: FREQ MODE WIDTH 
    31..32:       EDGE
      Mem caps: FREQ MODE WIDTH

MiM-TMassey, 2024-12-23:
So you did!  Awesome!

That takes care of all of the issues here.  I really appreciate it.  Closing the ticket.

closed 2024-12-23 by MiM-TMassey